HASwitchPlate/openHASP #577: Suggestion : Allow for topic attribute for objects

analystcmyk, 2023-11-13:
Running Version 0.7.0-rc6

I have an object defined as : 

`{"page":0,"id":5,"obj":"bar","x":0,"y":30,"h":30,"w":100, "min":0, "max":100,"val":22}`

I can change the text by publishing raw value `88` via MQTT to topic `hasp/plate/command/p0b5.val`

My sensor publishes every 5 seconds a raw temperature value to `sensor1/temp`

### Suggest

Allow for topic to be added directly to the object.

`{"page":0,"id":5,"obj":"bar","x":0,"y":30,"h":30,"w":100, "min":0, "max":100,"val":{"srctopic":"sensor1/temp"}}`

The `"val":{"srctopic":"sensor1/temp"}` would mean that published value in the `srctopic` would set the value. 
 
This would save a lot of workarounds like mqtt-mqtt republishing, and in the case of multiple devices would ease the effort of configuring.

 A lot of smart devices eg the ones flashed with OpenBeken  https://github.com/openshwprojects/OpenBK7231T_App publish directly any value there. 

### Questions

- topic should be within same broker? If not it would mean a lot of extra work, or 
complex syntax like `user:passw@mqtt://topicpath/:1883`
- limit to read published values only?  
Or allow actions to publish to custom topics as well?

Any thoughts would be appreciated.

smcgann99, 2023-11-14:
The main issue I can see is why should val be treated differently to other objects?
What I mean is that it would also be nice to apply this function to a text property or a switch toggle state etc etc ....
I haven't used OpenBK but in the case of Tasmota for example you could easily get the device to publish the value to the hasp topic directly.

analystcmyk, 2023-11-14:
I'll try with OpenBEKEN. 
Espurna afaik does not support specific values to be published in a different topic.

analystcmyk, 2023-11-14:
OpenBEKEN can handle it with this command.

`addEventHandler OnChannelChange 2 publish hasp/plate/command/p0b5.val $CH2 1`

Nevertheless you are not always in control of the topic (third party devices etc), so it still might be something to consider.

smcgann99, 2023-12-21:
> The `"val":{"srctopic":"sensor1/temp"}` would mean that published value in the `srctopic` would set the value.

It's not just the topic but also the payload property to use, so not quite that simple.

fvanroie, 2024-04-27:
This is typically the function of an Home Automation System, to keep tabs on all devices and convert topic/payload/protocol where needed. This should not be the functionality of a display-controller.