HASwitchPlate/openHASP #583: Unzip - Compression is not supported

jazztunes, 2023-11-30:
## Issue
Not sure if this is really a bug or just an FYI.

It appears that the unzip command only supports files stored without compression. Most zip utilities use at least some level of compression by default.  If your zip file uses compression, you'll see an error such as this:

`FILE: Compression is not supported 8`

## Workaround
In Linux, use compression level "0". For example.
`$ zip -0 -r images.zip *.png`

In windows, you'd need to download a zip utility or use power shell. Something that allows you to specify a compression level.

nagyrobi, 2023-11-30:
It's like that by design:

https://openhasp.haswitchplate.com/0.6.3/commands/#unzip

jazztunes, 2023-12-01:
Appreciate the quick reply - love the project!

closed 2023-12-01 by jazztunes