HASwitchPlate/openHASP #590: Bar doesn't display anything if value is under 1.0 and radius not effecting bar

zagi988, 2023-12-07:
### Describe the bug

### To Reproduce
```
{"page":1,"id":72,"obj":"bar","x":15,"y":430,"w":185,"h":30,"min":0,"max":5,"radius":5}
```

### Expected behavior
Bar to display the proper value even for values under 1.0.

### Screenshots or video
[image]

zagi988, 2023-12-09:
Ok, just figured that I need to set "radius10" for radius. At least that isn't a bug.

fvanroie, 2023-12-09:
`min`, `max`, `val` and `start_value` are all `int16` so you can't use floating point numbers.

fvanroie, 2023-12-09:
You shouldn't use floating point in any case, the result is undefined. Use Round to make the number an integer so it is interpreted correctly.

If you want more then 5 steps for the Amperage, you can multiply by a fixed number (e.g. 10 or 20) frist before rounding the result. Also adjust the max in that case by the same factor.

zagi988, 2023-12-09:
So wait, if I got it right, what you are telling me is that I shouldn't have any .something values?
Like, I can't have any bar display something that is let's say 23.4, I must round it to 23?

fvanroie, 2023-12-09:
23.4 is not an integer, 23 is... LVGL only support integer values.
If you want to show 23.4, then multiply by 10 before rounding to get 234. But you must multiply your `max` value also by 10.

zagi988, 2023-12-09:
aaaaa, I understand now. Thanks!

fvanroie, 2023-12-09:
No problem, you're welcome.

zagi988, 2024-05-18:
We still have the problem of radius not affecting the bar in bar. Only bar background gets changed.

fvanroie, 2024-05-18:
Show your jsonl and screenshot please. Did you use `radius10` as indicated above?

fvanroie, 2024-08-24:
Is this still an issue? Please let me know.