HASwitchPlate/openHASP #598: MQTT client fails to connect on MacOS

alvarolobato, 2023-12-22:
### Perform all steps below and tick them with [x]
- [x ] Check the related part of the [Documentation](https://www.openhasp.com/)
- [ x] Update openHASP to the latest version
- [ x] Reproduce the issue and describe all steps

### Describe the bug
Connection to MQTT fails on master with "client error", it works well in 0.6.3
```
[10:42:01.763][252504/252504  0] MQTT: src/mqtt/hasp_mqtt_paho_single.cpp 363
[10:42:01.763][252504/252504  0] MQTT: src/mqtt/hasp_mqtt_paho_single.cpp 369
[10:42:01.763][252504/252504  0] MQTT: src/mqtt/hasp_mqtt_paho_single.cpp 375
[10:42:01.763][252504/252504  0] MQTT: src/mqtt/hasp_mqtt_paho_single.cpp 379
[10:42:01.763][252504/252504  0] MQTT: src/mqtt/hasp_mqtt_paho_single.cpp 294
[10:42:01.763][252504/252504  0] MQTT: src/mqtt/hasp_mqtt_paho_single.cpp 300
[10:42:01.763][252504/252504  0] MQTT: src/mqtt/hasp_mqtt_paho_single.cpp 315
[10:42:01.763][252504/252504  0] MQTT: src/mqtt/hasp_mqtt_paho_single.cpp 333
[10:42:01.967][252504/252504  0] MQTT: Failed to connect, return code -1

```
I tested with old versions of Paho and it doesn't seem related to changes in the library, something seems to have been changed in how the connection is performed in MacOS from OpenHasp.

### To Reproduce
- Build from master on MacOS
- Either configure you MQTT server in the config file or use command line parameters
- Connection will always fail

### Expected behavior
I expect it to connect to the MQTT server
### Screenshots or video
Logs with Paho tracing enabled
```

Command-line arguments:
  argv[0]   ./program
  argv[1]   --mqtthost
  argv[2]   localhost
  argv[1]   --mqtthost
{"mqtt":{"host":"localhost"}}
[10:47:39.285][    0/    0  0] MQTT: host => localhost

        open____ _____ _____ _____
          |  |  |  _  |   __|  _  |
          |     |     |__   |   __|
          |__|__|__|__|_____|__|
        Home Automation Switch Plate
        Open Hardware edition v0.7.0-rc9

[10:47:39.285][    0/    0  0] MAIN: resolution 240 x 320
[10:47:39.285][    0/    0  0] MAIN: pre setup
[10:47:39.285][258680/259296  1] MAIN: Sysname:  Darwin
[10:47:39.285][258680/259296  1] MAIN: Nodename: Alvaros-MBP.local
[10:47:39.285][258680/259296  1] MAIN: Release:  23.1.0
[10:47:39.285][258680/259296  1] MAIN: Version:  Darwin Kernel Version 23.1.0: Mon Oct  9 21:27:24 PDT 2023; root:xnu-10002.41.9~6/RELEASE_ARM64_T6000
[10:47:39.285][258680/259296  1] MAIN: Machine:  arm64
[10:47:39.285][258680/259296  1] MAIN: Processor  : unknown
[10:47:39.285][258680/259296  1] MAIN: CPU freq.  : 0 MHz
[10:47:39.285][258680/259296  1] TFT : Starting...
[10:47:39.395][258680/259240  1] TFT : Driver     : SDL2
[10:47:39.395][258680/259240  1] TFT : SDL Version: v2.28.3
[10:47:39.395][258680/259240  1] TFT : DMA        : Disabled
[10:47:39.395][258680/259240  1] TFT : Started
[10:47:39.395][258680/259240  1] MAIN: Sysname:  Darwin
[10:47:39.395][258680/259240  1] MAIN: Nodename: Alvaros-MBP.local
[10:47:39.395][258680/259240  1] MAIN: Release:  23.1.0
[10:47:39.395][258680/259240  1] MAIN: Version:  Darwin Kernel Version 23.1.0: Mon Oct  9 21:27:24 PDT 2023; root:xnu-10002.41.9~6/RELEASE_ARM64_T6000
[10:47:39.395][258680/259240  1] MAIN: Machine:  arm64
[10:47:39.395][258680/259240  1] MAIN: Processor  : unknown
[10:47:39.395][258680/259240  1] MAIN: CPU freq.  : 0 MHz
[10:47:39.395][258680/259240  1] LVGL: Starting...
[10:47:39.395][258680/259240  1] LVGL: Version    : 7.11.0
[... 254 lines cut ...]
20231222 104739.498 (7987781312)  (1)< MQTTClient_connectURI:1702 (-1)
20231222 104739.599 (7987781312) (0)< MQTTClient_connectAll:1910 (-1)
[10:47:39.599][252504/252504  0] MQTT: Failed to connect, return code -1
[10:47:39.599][252504/252504  0] MQTT: src/mqtt/hasp_mqtt_paho_single.cpp 346
[10:47:39.599][252504/252504  0] MAIN: src/main_sdl2.cpp 137
[10:47:39.599][252504/252504  0] GPIO: Saving Pin config #0 pin 3 - type 11 - group 0 - func 255
[10:47:39.599][252504/252504  0] GPIO: Saving Pin config #1 pin 4 - type 10 - group 0 - func 255
[10:47:39.599][252504/252504  0] GPIO: Saving Pin config #2 pin 13 - type 2 - group 0 - func 255
[10:47:39.599][252504/252504  0] GPIO: Saving Pin config #3 pin 14 - type 34 - group 0 - func 255
[10:47:39.599][252504/252504  0] GPIO: Saving Pin config #4 pin 5 - type 172 - group 0 - func 255
[10:47:39.599][252504/252504  0] MAIN: src/main_sdl2.cpp 146
[10:47:39.599][252504/252504  0] MAIN: loop started
20231222 104739.599 (7987781312) (0)> MQTTClient_receive:2777
20231222 104739.599 (7987781312) (0)< MQTTClient_receive:2820 (-3)
20231222 104739.599 (7987781312) (0)> MQTTClient_receive:2777
20231222 104739.599 (7987781312) (0)< MQTTClient_receive:2820 (-3)
20231222 104739.599 (7987781312) (0)> MQTTClient_receive:2777
20231222 104739.599 (7987781312) (0)< MQTTClient_receive:2820 (-3)
20231222 104739.599 (7987781312) (0)> MQTTClient_receive:2777
20231222 104739.599 (7987781312) (0)< MQTTClient_receive:2820 (-3)
20231222 104739.599 (7987781312) (0)> MQTTClient_receive:2777
20231222 104739.599 (7987781312) (0)< MQTTClient_receive:2820 (-3)
20231222 104739.599 (7987781312) (0)> MQTTClient_receive:2777
20231222 104739.599 (7987781312) (0)< MQTTClient_receive:2820 (-3)
```

alvarolobato, 2023-12-22:
Taking a look at the changes between the two versions it looks like this change is what broke it:
https://github.com/HASwitchPlate/openHASP/commit/56ae45dae251566e38fc72e42395282d62e12a13#diff-979b76318c39babb49890eba446d0fc0e22c2cace106834c8eb54dd873c026b8L307

Commenting back again the method to add the callback made it work again.

fvanroie, 2024-02-02:
Can you create a PR with the fix, please?

fvanroie, 2024-02-23:
I believe this has been fixed inPR #631 and #640, can you confirm this is working for you now?

closed 2024-03-11 by fvanroie