HASwitchPlate/openHASP #601: idle does not seem to work

stefan242, 2023-12-26:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

"When the display has not been touched for the Short Idle amount of seconds, an idle = short event will be sent out."

Short Idle and Long idle are set to 10 and 20 seconds, but no event `hasp/platex/state/idle` is started.

I can inject `idle short` or `idle long` manually to `hasp/platex/command` and with `idle long` the display blanks. If I touch the display when off, there is also no event, but the screen is turned on.

openHASP 0.7.0-rc9 with a gs-t3e_16MB

### To Reproduce

In Node-RED:

```
[{"id":"a6c6126a172abf3a","type":"mqtt in","z":"63025264bf39ecf8","name":"","topic":"hasp/platex/state/idle","qos":"2","datatype":"auto-detect","broker":"7f211795.773398","nl":false,"rap":true,"rh":0,"inputs":0,"x":140,"y":1920,"wires":[["3fdf4c6d0b34a095","5c8b31c2ded406c9"]]},{"id":"5c8b31c2ded406c9","type":"debug","z":"63025264bf39ecf8","name":"debug xxx","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":510,"y":1840,"wires":[]},{"id":"7f211795.773398","type":"mqtt-broker","name":"Mosquitto","broker":"localhost","port":"1883","clientid":"","usetls":false,"compatmode":false,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","closeTopic":"","closeQos":"0","closePayload":"","willTopic":"","willQos":"0","willPayload":""}]
```

### Expected behavior

The debug node should produce:
`[node: debug xxx]
hasp/platex/state/idle : msg.payload : string[4/5/3]
"long" or "short" or "off"
`

pepe59, 2023-12-27:
Same problem here. In the Home assistant, the automation set according to the documentation is supposed to turn off the display, but it doesn't work.
Sunton ESP32-8048S050C
OTA fw 0.7.0-rc9 33d21e0
It worked in the previous version.

smcgann99, 2024-01-04:
Can confirm NOT working in 0.7.0-rc9 894370b

stefan242, 2024-01-16:
@fvanroie any chance to get / test a new rc, which sends the idle events via MQTT again?

fvanroie, 2024-01-16:
Check the latest build run here: https://github.com/HASwitchPlate/openHASP/actions/runs/7524851274

stefan242, 2024-01-17:
That does not work either:

This is my plate1 with 0.6.3 with a Lanbon L8 and short idle 10, long idle 20. I inject a "idle off" (in Node-RED) to hasp/plate1/command and get back via MQQT in-node hasp/plate1/state/idle:

[image]

This works as expected.

This is my plate3 with 0.7.0-rc9 8d5ae79 with a gs-t3e_16MB and short idle 10, long idle 20. I inject a "idle off" to hasp/plate3/command and other than the idle off via hasp/plate3/state/idle nothing else happens.

[image]

The same, when I inject "idle short". No switch to "idle long" like with 0.6.3

openHASP 0.7.0-rc9 8d5ae79
Build DateTime Jan 15 2024 05:44:40 UTC
Environment gs-t3e_16MB

fvanroie, 2024-01-17:
I reverted back a previous PR in https://github.com/HASwitchPlate/openHASP/pull/605
Please test the latest build a57029c from the Actions tab.

stefan242, 2024-01-17:
YES!! thank you!

closed 2024-01-17 by stefan242