HASwitchPlate/openHASP #603: Any chance we can it work for another esp32-s3 GT911 touch display (ESP32-4848S040)

mswaleh, 2024-01-02:
I found this cheap display at Aliexpress (https://www.aliexpress.us/item/1005006334976806.html) which appears to be new(er) in the market.

Board: ESP32-S3-N16R8
Display: ST7701, Touch: GT911

This is very close to "GS-T3E v2.3" in specs, which is one from the nightly list (https://nightly.openhasp.com/).

I am able to install it, it even connect to WiFi AP and then to my WiFi via the webUI. I can also see screenshot in the web UI but nothing comes on screen. I was trying to look at the source code but quite new to all this. Anybody can point me right direction or help with this, would be greatly appreciated.

tbxMb, 2024-01-02:
I also have one of these and i managed to setup the pin configuration for the Display.
I got the pin mappings out of the `1-Demo/Demo_Arduino/1_1_86switch_onoff/86switch_onoff/86switch_onoff.ino` sketch provided with the units documentation.

Everything seems to work except the colors. 
[image]
[image]

[image]
[image]

mswaleh, 2024-01-02:
wow, that was fast. thanks. I saw and copied your changes. I am also trying to figure out to see if there is anything I can do to make it work. I will share here. I like this display. Couple of binaries they have for demo, work really well. hoping to get it to work like it does in the demos.

cleverhomeKK, 2024-01-02:
Hi, I have the same problem. One more thing, if it helps with the search: It will be red at: #00FFFF
instead of #FF0000 . Seems somehow negated.

tbxMb, 2024-01-02:
Here are some tests. The image indeed gets displayed with inverted colors. When inverting the photo of the display in Gimp everything looks Ok.
[image]

I tried to enable the Invert option in the display settings but that does not work.

mswaleh, 2024-01-02:
@tbxMb great job.
It appears, inverted colors are closer to the original. by the way, can you share the env entries you made to make it work. I am trying to build it on another repo (https://github.com/mswaleh/msopenHASP) but my job is failing because of "
UnknownEnvNamesError: Unknown environment names 'esp32-s3-4848s050'".

tbxMb, 2024-01-02:
I think you got a typo there its `esp32-s3-4848s040`

mswaleh, 2024-01-02:
ha. literally saw seconds before your comment :-D. Much appreciated

tbxMb, 2024-01-02:
I manually enabled the hardware invert function via a spi write and managed to get the primary colors looking alright. But gradients still look weird. 
[image]
Im not sure what is causing this but i think i have seen something like once.

tbxMb, 2024-01-03:
### ITS WORKING 🥳

[image]

I had to change one byte in the setup operations from 0x08 to 0x00, i also merged the display invert command into the setup operations.

Can someone validate that it is also working on their device ?

mswaleh, 2024-01-03:
@tbxMb can you link the artifacts file?

tbxMb, 2024-01-03:
@mswaleh i cant link you one, but here is the commit that you can build yourself :) 

[7af48cf](https://github.com/tbxMb/openHASP/commit/7af48cff375fc0c73ffa10230f0dc45a7c1a2354)

mswaleh, 2024-01-03:
I can confirm that it is working smoothly. saweeet.. great job @tbxMb

smcgann99, 2024-01-10:
This looks really nice for the price. 
Can you confirm the spec, web site is a bit lacking, it seems to have battery input 3 x relays I assume a 240V PSU built in etc ....

cleverhomeKK, 2024-01-10:
Hi,
The device has a 230V power supply (and USB-C on the side) and 1-3 relay outputs (230V).
These can of course also be controlled by OpenHasp.
However, the back part :-) does not fit into a round EU flush-mounted box.

smcgann99, 2024-01-11:
> Hi, The device has a 230V power supply (and USB-C on the side) and 1-3 relay outputs (230V). These can of course also be controlled by OpenHasp. However, the back part :-) does not fit into a round EU flush-mounted box.

Any chance you could post some pictures of the back of the case ? 
Its very similar to the GS-T3E v2.3, but MUCH cheaper

mswaleh, 2024-01-11:
It works with 110V too. 3 ways you can power it, USB-C, GPIO pins (5v and gnd), and AC. I think it can take 110-250VAC.

fduch2k, 2024-01-11:
works like a charm on my esp32-4848S040 device. No glitches, no color issues. Thank you

smcgann99, 2024-01-11:
Just checked the download link on product page an its actually a TFT screen not IPS ;-(

RednazPublic, 2024-01-14:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

RednazPublic, 2024-01-14:
I hope it can help. The board description image has yellow circles and these are zero ohm resistors, the connection between the boards (where I left "?") are L1, L2 and L3. I made a mistake, the correct digital audio is NS4168 and the 4 inch display relay and I2S audio are shared IO ports, IO1, IO2 and IO40. If audio is required, 0 ohm resistors R25, R26 and R27 must be moved to R21, R22 and R23.

smcgann99, 2024-01-15:
> I hope it can help. The board description image has yellow circles and these are zero ohm resistors, the connection between the boards (where I left "?") are L1, L2 and L3. I made a mistake, the [...]

Thanks very much for this. What is the picture quality / viewing angle like with it being a TFT ?
Have noticed any ghosting (screen burn etc), that has been reported on cheaper displays ?

tbxMb, 2024-01-15:
Id say the Image quality is really good, and the viewing angles too you can look at it from almost 90° it gets a bit darker but is still legible. As for burn in i haven't had time to use it longer than a few hours but i did not really notice any Burn In yet.

RednazPublic, 2024-01-15:
> Thanks very much for this. What is the picture quality / viewing angle like with it being a TFT ? Have noticed any ghosting (screen burn etc), that has been reported on cheaper displays ?

I'm sorry, it arrived last Friday and I was just checking the board and connection, I didn't really use it for over an hour. I'm actually going to use it for another project and saw your request for photos.

teoconnect, 2024-02-10:
Very good device, all works for me too, the display is also very good
thank's for your hard work

it's possible to include the device in the nightly build?
https://nightly.openhasp.com/
[image]

fvanroie, 2024-02-10:
> it's possible to include the device in the nightly build?

Please provide a Pull request with the ini file for this device.

teoconnect, 2024-02-10:
I meant the portal https://nightly.openhasp.com/
where currently the entry for esp32-s3-4848s040 is missing,
in the .ini file configuration is already present, just select the 2 entries below
     user_setups/esp32s3/*.ini
     esp32-s3-4848S040
with this selection the device work propertly

kipwittchen, 2024-02-19:
I recently purchased this device and am anticipating its arrival! Pardon my ignorance, where would the pull request be made to get this board added to the nightly build portal? I see the html file that lists the boards: https://github.com/fvanroie/install/blob/main/html/index.html but that doesn't seem to link to the platformio_override.ini file to uncomment the two lines necessary?

Aleborg, 2024-02-26:
> Very good device, all works for me too, the display is also very good
> thank's for your hard work
> 
> it's possible to include the device in the nightly build?
> https://nightly.openhasp.com/
> [image]
> 
> below the link of the build that i have installed
> https://www.dropbox.com/scl/fi/emfuptraro12qfbw6albs/esp32-s3-4848S040_full_16MB_v0.7.0-rc9_2dce5da.bin?rlkey=xkk4w8g3gxujltxq4mzptnfgo&dl=0

Awesome, thanks, works perfect for me!
Does anyone know if it's possible to use the built in bluetooth as a proxy fof HA?

fvanroie, 2024-02-26:
> Does anyone know if it's possible to use the built in bluetooth as a proxy fof HA?

openHASP does not support bluetooth  as a proxy. You have to use another firmware or another device for that.

Aleborg, 2024-02-27:
@fvanroie thanks for the quick reply!
A pity but understandable given that it can interfere with the wifi!

fvanroie, 2024-03-02:
> I meant the portal https://nightly.openhasp.com/ where currently the entry for esp32-s3-4848s040 is missing, in the .ini file configuration is already present,

The device has been added to nightly, please test if you have the chance.

flexiti, 2024-03-03:
I confirm that the anESP32-4848S040 compilation works, the custom option also works fine. **Thanks for this.** 
Does anyone have an electrical diagram for this model?

flexiti, 2024-03-03:
According to me, here is NS4168 (not NS4188 as commented in description photo above) - 2.5W audio I2S, how can we use it?

flexiti, 2024-03-09:
Switch SW1 is connected to pin 5 IP5306, It disconnects the power (actually acts as a reset) if you hold it for 10 seconds

flexiti, 2024-03-09:
Yellow resistor on the right, its on/off power supply from USB, yellow on the right - disconnect backlight from the port IO38

flexiti, 2024-03-09:
Sensor DS18B20,AHT20,BMP280 are working

[image]

xiasi0, 2024-03-10:
> ESP32-4848S040 本地连接传感器 - DS18B20、AHT20、BMP280 正在工作
> 
> ![ESP32_view1](https://private-user-images.githubusercontent.com/11538048/311454964-721b5a04-8a78-4e35-b028-503a707a480c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAwNzQ1NzMsIm5iZiI6MTcxMDA3NDI3MywicGF0aCI6Ii8xMTUzODA0OC8zMTE0NTQ5NjQtNzIxYjVhMDQtOGE3OC00ZTM1LWIwMjgtNTAzYTcwN2E0ODBjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzEwVDEyMzc1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE2NjJjNGU2NDg2MGU5ZmMxNWNkYzNjYjJiNTAzYzRiYmI1OGRkZjY4ZTllMWQwZmNiYmZjOGRiYjUwYzNkMGUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.AWWmQRmSRnqEQPLipT6HChwrwyafVvMJQtIrKR30QBY)

Is the backlight brightness of ESP32-4848S040 unable to be adjusted? I tried adjusting the brightness and it would directly turn off the backlight

fvanroie, 2024-03-10:
> Is the backlight brightness of ESP32-4848S040 unable to be adjusted? I tried adjusting the brightness and it would directly turn off the backlight

Please try https://github.com/HASwitchPlate/openHASP/actions/runs/8222316763

xiasi0, 2024-03-10:
> > ESP32-4848S040 的背光亮度是无法调节的吗？我尝试调整亮度，它会直接关闭背光
> 
> 请尝试 https://github.com/HASwitchPlate/openHASP/actions/runs/8222316763

This firmware esp32-s3-4848s040'full_16MB_v0.7.0-rc11'f27744b still turns off the backlight completely when the backlight brightness is around 80%.

xiasi0, 2024-03-10:
> > > ESP32-4848S040 的背光亮度是无法调节的吗？我尝试调整亮度，它会直接关闭背光
> > 
> > 
> > 请尝试 https://github.com/HASwitchPlate/openHASP/actions/runs/8222316763
> 
> 当背光亮度在 80% 左右时，该固件 esp32-s3-4848s040'full_16MB_v0.7.0-rc11'f27744b 仍会完全关闭背光。

IO1, IO2, IO40 correspond to three relays. IO1 and IO2 are working normally. IO40 has no 3.3V output and no corresponding output. Please review

fvanroie, 2024-03-10:
> This firmware esp32-s3-4848s040'full_16MB_v0.7.0-rc11'f27744b still turns off the backlight completely when the backlight brightness is around 80%.

Can you try adding     -D BACKLIGHT_FREQUENCY=2000  to the config?
The default is 20000, maybe that's too much for those boards

kipwittchen, 2024-03-10:
> Sensor connected locally to ESP32-4848S040 - DS18B20,AHT20,BMP280 are working
> 
> ![ESP32_view1](https://private-user-images.githubusercontent.com/11538048/311454964-721b5a04-8a78-4e35-b028-503a707a480c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAwOTE2NTIsIm5iZiI6MTcxMDA5MTM1MiwicGF0aCI6Ii8xMTUzODA0OC8zMTE0NTQ5NjQtNzIxYjVhMDQtOGE3OC00ZTM1LWIwMjgtNTAzYTcwN2E0ODBjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzEwVDE3MjIzMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgyNjkwYTdjNWZkNGY1NDYzM2NmMjg2ZTJhN2RmNTMzM2JjZGQ5MjgxMGEzZjE5ZmE2ODllMWZiZDZjMTRlNWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.nrWYusR2kYs0RVcFIqlHeKcsMhpLD8lvCJ9MaQJpbrY)

What GPIO settings or config did you use to access these sensors?

flexiti, 2024-03-10:
https://github.com/HASwitchPlate/openHASP/discussions/666
I2C 
https://github.com/HASwitchPlate/openHASP/discussions/664
Port 1

flexiti, 2024-03-10:
> Can you try adding -D BACKLIGHT_FREQUENCY=2000 to the config?
> The default is 20000, maybe that's too much for those boards
@fvanroie
I changed it and tried - PWM backlight working on this board with these settings (2000)

kipwittchen, 2024-03-10:
> > Can you try adding -D BACKLIGHT_FREQUENCY=2000 to the config?
> > The default is 20000, maybe that's too much for those boards
> > @fvanroie
> > I changed it and tried - PWM backlight working on this board with these settings (2000)

Also can confirm this works! Though anything below 25% is too dark to see.

kipwittchen, 2024-03-10:
> @kipwittchen #666 I2C #664 Port 1

Thanks! For some reason I can't seem to get it to read values in but I'll keep plugging away. I'm trying to get it to publish the values via mqtt so may be my modifications causing the issue

xiasi0, 2024-03-11:
> > This firmware esp32-s3-4848s040'full_16MB_v0.7.0-rc11'f27744b still turns off the backlight completely when the backlight brightness is around 80%.
> 
> Can you try adding -D BACKLIGHT_FREQUENCY=2000 to the config? The default is 20000, maybe that's too much for those boards

> > This firmware esp32-s3-4848s040'full_16MB_v0.7.0-rc11'f27744b still turns off the backlight completely when the backlight brightness is around 80%.
> 
> Can you try adding -D BACKLIGHT_FREQUENCY=2000 to the config? The default is 20000, maybe that's too much for those boards

It's normal now, thank you

xiasi0, 2024-03-11:
[image]

I haven't set up GPIO, what are these entities? Can I not compile this part of the entity?

xiasi0, 2024-03-11:
> > > @fvanroie
> > > I changed it and tried - PWM backlight working on this board with these settings (2000)
> 
> i'm new to openhasp, but where do i add that? (-D BACKLIGHT_FREQUENCY=2000) thank

[image]

Find it here, but be careful, I'll lower the frequency. Will cause abnormal noise in the MOS tube. Similar to electric current sound. This is inevitable to reduce frequency.

kipwittchen, 2024-03-11:
> @kipwittchen #666 I2C #664 Port 1

Hope this isn't a dumb question, are these sensors actually on the board or did you wire them yourself to the board?

RealDeco, 2024-03-11:
> Find it here, but be careful, I'll lower the frequency. Will cause abnormal noise in the MOS tube. Similar to electric current sound. This is inevitable to reduce frequency.

ok i had hoped for a more userfriendly place, i'm not a developer so i'll just sit quiet and pray that someone hopefully fix that in the future :)) 

thanks for the answer anyway

fvanroie, 2024-03-11:
No need to sit back and wait, try out the [latest build from Actions](https://github.com/HASwitchPlate/openHASP/actions/runs/8234155942) and test that. If it works fine please report and I'll update the nightly installer.

RealDeco, 2024-03-11:
> No need to sit back and wait, try out the [latest build from Actions](https://github.com/HASwitchPlate/openHASP/actions/runs/8234155942) and test that. If it works fine please report and I'll update the nightly installer.

wow nice! that fixed the backlight, but the "short idle" in config settings sets it to 10% where it's not visible, 50% would be better, is there a place to edit that?

thanks for quick reply 👍

xiasi0, 2024-03-11:
\

> No need to sit back and wait, try out the [latest build from Actions](https://github.com/HASwitchPlate/openHASP/actions/runs/8234155942) and test that. If it works fine please report and I'll update the nightly installer.

> No need to sit back and wait, try out the [latest build from Actions](https://github.com/HASwitchPlate/openHASP/actions/runs/8234155942) and test that. If it works fine please report and I'll update the nightly installer.

The GPIO40 setting relay is not available. May I ask how to cancel compilation for useless entities

kipwittchen, 2024-03-11:
> > No need to sit back and wait, try out the [latest build from Actions](https://github.com/HASwitchPlate/openHASP/actions/runs/8234155942) and test that. If it works fine please report and I'll update the nightly installer.
> 
> wow nice! that fixed the backlight, but the "short idle" in config settings sets it to 10% where it's not visible, 50% would be better, is there a place to edit that?
> 
> thanks for quick reply 👍
> 
> edit: it works i can now change brightness smooth, but under 35 it's like off.

Are you using the home assistant integration? It the backlight % at idle can be adjusted under settings there.

RealDeco, 2024-03-11:
> Are you using the home assistant integration? It the backlight % at idle can be adjusted under settings there.

yeah, it's from home assistant i can see that under 35% the display is like off, so the "short idle" settings at 10% is same as off, but i can now adjust backlight more smoothly, before it was line on/off at 85%

RealDeco, 2024-03-12:
i just noticed something i didn't notice before, when backlight is anything ELSE than 100% the display makes a  constant ringing tone, but not when at 100%, could it be the freq at 2000 causing this?

RealDeco, 2024-03-13:
> hello, does anyone have a bin file available with some background image

you can make one here: https://lvgl.io/tools/imageconverter

tiimsvk, 2024-03-13:
I tested and the problem will be in esphome code

xiasi0, 2024-03-14:
> i just noticed something i didn't notice before, when backlight is anything ELSE than 100% the display makes a constant ringing tone, but not when at 100%, could it be the freq at 2000 causing this?

Yes, a frequency of 2000 is a sound that can be heard by the ears. Adjust the frequency to 20000, and if the frequency is not within the range of ear hearing, you will not be able to hear.

[21 more replies]

closed 2024-03-14 by fvanroie