HASwitchPlate/openHASP #606: support for esp32-2432s022c board

cosmos1978, 2024-01-04:
I recently bought some of these screens (ESP32-2432s022c)

I tried configuring it based on your github but I only am able to get some distorted images on the display. ( snow pattern )
It uses ST7789 driver and the same touch driver as your board. The difference is that the tft display uses an 8 bit parallel bus.
It also needs to use lovyanGFX since it uses pin's above 31

I have some sample code that runs perfect on the board. And I tried to map the pins like in that code but I am stuck now on this distorted image

Perhaps you have some ideas on how to get it working ?

below is the working arduino code and the platformio.ini env I created based on https://github.com/elik745i/ESP32-2432s024c since he was able to include support for the touch chipset 

public:
LGFX(void)
{
{
auto cfg = _bus_instance.config();
cfg.freq_write = 25000000;
cfg.pin_wr = 4;
cfg.pin_rd = 2;
cfg.pin_rs = 16;

        cfg.pin_d0 = 15;
        cfg.pin_d1 = 13;
        cfg.pin_d2 = 12;
        cfg.pin_d3 = 14;
        cfg.pin_d4 = 27;
        cfg.pin_d5 = 25;
        cfg.pin_d6 = 33;
        cfg.pin_d7 = 32;

        _bus_instance.config(cfg);
        _panel_instance.setBus(&_bus_instance);
    }

    {
        auto cfg = _panel_instance.config();

        cfg.pin_cs = 17;
        cfg.pin_rst = -1;
        cfg.pin_busy = -1;

        cfg.panel_width = 240;
        cfg.panel_height = 320;
        cfg.offset_x = 0;
        cfg.offset_y = 0;
        cfg.offset_rotation = 0;
        // cfg.dummy_read_pixel = 8;
        // cfg.dummy_read_bits = 1;
        cfg.readable = false;
        cfg.invert = false;
        cfg.rgb_order = false;
        cfg.dlen_16bit = false;
        cfg.bus_shared = true;

        _panel_instance.config(cfg);
    }

    setPanel(&_panel_instance);
}
};

[env:esp32-2432s022c_4MB]
extends = esp32-2432s022, flash_4mb
board = esp32dev
build_flags =
${esp32-2432s022.build_flags}
${esp32.no_ps_ram}

${esp32.hspi}
-D LGFX_USE_V1=1
-D ILI9341_DRIVER=1
-D ESP32_PARALLEL=1
-D TOUCH_DRIVER=0x0820
-D HASP_USE_LGFX_TOUCH=1
-D TFT_ROTATION=0
-D TFT_WIDTH=240
-D TFT_HEIGHT=320
-D TFT_BCKL=5  ;None, configurable via web UI (e.g. 2 for D4)
-D TFT_CS=17  ; Chip select control pin
-D TFT_RST=-1 ; Reset pin
-D TFT_DC=16	
-D TFT_WR=4   ; Write strobe control pin - must use a pin in the range 0-31
-D TFT_RD=2
-D TFT_D0=15  ; Must use pins in the range 0-31 for the data bus
-D TFT_D1=13  ; so a single register write sets/clears all bits
-D TFT_D2=12
-D TFT_D3=14
-D TFT_D4=27
-D TFT_D5=25
-D TFT_D6=33
-D TFT_D7=32
-D TOUCH_SDA=21
-D TOUCH_SCL=22
;-D TOUCH_IRQ=34   ; use 34-39 as these are input only pins
-D TOUCH_RST=-1   ; not used, connected to 3.3V
-D SUPPORT_TRANSACTIONS
-D TOUCH_OFFSET_ROTATION=0
;-D I2C_TOUCH_FREQUENCY=400000
;-D I2C_TOUCH_PORT=1
-D I2C_TOUCH_ADDRESS=0x15
-D SPI_FREQUENCY=65000000
-D SPI_TOUCH_FREQUENCY=2500000
-D SPI_READ_FREQUENCY=20000000
lib_deps =
${esp32-2432s022.lib_deps}
${lovyangfx.lib_deps}

gekkie, 2024-03-20:
I second this request

cafalchio, 2024-07-09:
same issue, not sure which are the correct pins

Panaceabeachbum, 2024-07-29:
Same here, any help appreciated

Panaceabeachbum, 2024-07-30:
Is there a pinout available for this particular board? I have no problem finding a get started pdf for a number of their other boards but not this one

bennydiamond, 2024-08-30:
Somebody seems to have a platformio config for this very board.
https://github.com/rzeldent/platformio-espressif32-sunton/blob/main/esp32-2432S022C.json

Probably this could be used in some fashion in this project.

bennydiamond, 2024-12-22:
I'm trying to make a config for this board. 
So far here's where I'm at
```
[env:esp32-2432s022c]
extends = arduino_esp32_v2, flash_4mb
board = esp32dev
upload_speed = 921600

build_flags =
    ${arduino_esp32_v2.build_flags}
    ${esp32.no_ps_ram}

;region -- TFT_eSPI build options ------------------------
    -D LGFX_USE_V1=1
;    -D USER_SETUP_LOADED=1
    -D ST7789_DRIVER=1
;    -D CGRAM_OFFSET=
;    -DSUPPORT_TRANSACTIONS
    -D TFT_INVERSION_OFF
    -D TFT_PARALLEL_8_BIT
    -D TFT_ROTATION=0 ; 0=0, 1=90, 2=180 or 3=270 degree
    -D TFT_WIDTH=240
    -D TFT_HEIGHT=320
    -D TFT_CS=17       ; Chip select control pin=library pulls permanently low
    -D TFT_DC=16       ; Data Command control pin
    -D TFT_RST=-1     ; Reset pin, toggles on startup
    -D TFT_WR=4       ; Write strobe control pin
    -D TFT_RD=2      ; Read strobe control pin
    -D TFT_D0=15      ; Must use pins in the range 0-31 or alternatively 32-48
    -D TFT_D1=13      ; so a single register write sets/clears all bits.
    -D TFT_D2=12      ; Pins can be randomly assigned, this does not affect
    -D TFT_D3=14      ; TFT screen update performance.
    -D TFT_D4=27
    -D TFT_D5=25
    -D TFT_D6=33
    -D TFT_D7=32
    -D TFT_BCKL=0    ; LED back-light

    ;CST816S driver
    -D TOUCH_DRIVER=0x816 
    -D HASP_USE_LGFX_TOUCH=1
    -D TOUCH_SDA=21
    -D TOUCH_SCL=22
    -D TOUCH_IRQ=-1
    -D TOUCH_RST=-1
    -D I2C_TOUCH_FREQUENCY=400000
    -D I2C_TOUCH_PORT=0
    -D I2C_TOUCH_ADDRESS=0x15

    -D SD_MISO=19
    -D SD_MOSI=23
    -D SD_SCLK=18
    -D SD_CS=5

;    -D PWR_EN=10
;    -D PWR_ON=14
;    -D BAT_ADC=5
;    -D BUTTON1=0
;    -D BUTTON2=21

; -- Debugging options -----------------------------
;   -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG
;endregion

lib_deps =
    ${arduino_esp32_v2.lib_deps}
    ${lovyangfx.lib_deps}
    ;${tft_espi.lib_deps}

```

Here's the result:
[image]

So not all that great but it's a start.
I'm seeing comments in other places that LCD pins should be kept on a same IO bank (pins 0 - 32 or 32 - 42) to make it work good. It's not the case on this board and I'm hoping it's not a deal breaker. I wouldn't mind a lower refresh rate if I can just have something displayed on it!

bennydiamond, 2024-12-22:
Success!
[image]

cosmos1978, 2024-12-22:
Wow that is amazing.

bennydiamond, 2024-12-23:
I guess you can close the issue now!

cosmos1978, 2024-12-23:
closing issue. Many thanks Bennydiamond

closed 2024-12-23 by cosmos1978

ottoklein, 2025-04-19:
//For Lovyan GFX

//https://github.com/HASwitchPlate/openHASP/issues/606

#define LGFX_USE_V1
#include <LovyanGFX.hpp>

class LGFX : public lgfx::LGFX_Device {
  lgfx::Panel_ST7789 _panel_instance;  // ST7796UI
  lgfx::Bus_Parallel8 _bus_instance;   // MCU8080 8B
  //lgfx::Light_PWM _light_instance;// same like Boot!
  lgfx::Touch_CST816S _touch_instance;

public:
  LGFX(void) {
    {
      auto cfg = _bus_instance.config();
      cfg.freq_write = 25000000;

      cfg.pin_wr = 4;
      cfg.pin_rd = 2;
      cfg.pin_rs = 16;

      // LCD data interface, 8bit MCU (8080)
      cfg.pin_d0 = 15;  
      cfg.pin_d1 = 13;  
      cfg.pin_d2 = 12;  
      cfg.pin_d3 = 14;  
      cfg.pin_d4 = 27;  
      cfg.pin_d5 = 25;  
      cfg.pin_d6 = 33;  
      cfg.pin_d7 = 32;  

      _bus_instance.config(cfg);
      _panel_instance.setBus(&_bus_instance);
    }
    {
      auto cfg = _panel_instance.config();

      cfg.pin_cs = 17;   
      cfg.pin_rst = -1; 
      cfg.pin_busy = -1;

      cfg.panel_width = 240;
      cfg.panel_height = 320;
      cfg.offset_x = 0;
      cfg.offset_y = 0;
      cfg.offset_rotation = 0;
       //cfg.dummy_read_pixel = 8;//
       //cfg.dummy_read_bits = 1;//
      cfg.readable = false;
      cfg.invert = false;
      cfg.rgb_order = false;
      cfg.dlen_16bit = false;
      cfg.bus_shared = true;

      _panel_instance.config(cfg);
    }
    {
     /*   
      auto cfg = _light_instance.config();

      cfg.pin_bl = 0;  
      cfg.invert = false;
      cfg.freq = 44100;
      cfg.pwm_channel = 7;

      _light_instance.config(cfg);
      _panel_instance.setLight(&_light_instance);
      */
    }
    {
      auto cfg = _touch_instance.config();

      cfg.x_min = 0;            // タッチスクリーンから得られる最小のX値(生の値)
      cfg.x_max = 239;          // タッチスクリーンから得られる最大のX値(生の値)
      cfg.y_min = 0;            // タッチスクリーンから得られる最小のY値(生の値)
      cfg.y_max = 319;          // タッチスクリーンから得られる最大のY値(生の値)
      cfg.pin_int = -1;         //  // INTが接続されているピン番号
      cfg.bus_shared = true;    // 画面と共通のバスを使用している場合 trueを設定
      cfg.offset_rotation = 0;  // 表示とタッチの向きのが一致しない場合の調整 0~7の値で設定           //6;// 6 ist angepasst!   = bei callibration

      // I2C接続の場合 Touch is ok
      cfg.i2c_port = 1;     // 使用するI2Cを選択 (0 or 1)
      cfg.i2c_addr = 0x15;  // I2Cデバイスアドレス番号
      cfg.pin_sda = 21;     //   // SDAが接続されているピン番号
      cfg.pin_scl = 22;     //     // SCLが接続されているピン番号
      cfg.freq = 400000;    // I2Cクロックを設定
      cfg.pin_int = 34 ;// use 34-39 as these are input only pins    
      cfg.pin_rst = -1 ;//not used, connected to 3.3V

    
      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);  // タッチスクリーンをパネルにセットします。
    }
    setPanel(&_panel_instance);
  }
};

cpertel, 2025-06-11:
> Success! [image]
> 
> Touchscreen also works.

Hi, sorry, but could you please be so kind and provide (a link to) the binary to upload to the device in order to use openHASP?
would be really great.

Thanks in advance and sorry if off topic

romka1977, 2025-08-19:
Hi,
I have an esp32-2432s022c variant with CST820 capacitive controller and with current esp32-2432s022c.ini capacitive touch doesn't work

In order to make it work i made these modifications:

    -D TOUCH_DRIVER=0x0820
    -D HASP_USE_LGFX_TOUCH=1

I am not a huge expert on how we can make this variety into the repo and if it is feasible at all

fvanroie, 2025-08-20:
Thank you. I updated the ini file, please test the [new build from here](https://github.com/HASwitchPlate/openHASP/actions/runs/17098392542/job/48488234612).
I think this should work for all 2.2" with CST820 capacitive controller now. If not, please report back here.

NellowTCS, 2025-09-14:
> Here's the result: [image]

I have a similar error. How did you fix it?