HASwitchPlate/openHASP #617: No response to page= command

hmorkemo, 2024-01-19:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

romainrossi, 2024-02-10:
In another issue, we are discussing Wifi instabilities on the Sunton board : https://github.com/HASwitchPlate/openHASP/issues/592#issue-2034340266
Maybe your issue is related, and your Sunton board is just loosing wifi connection.
On my board, it is very easy to spot by using the `ping` command from another computer to the Sunton.
Hope this helps.

hmorkemo, 2024-02-12:
@romainrossi Great! I think a piece of copper tape solved it!