HASwitchPlate/openHASP #618: idle_* scripts no longer seem to be called

hb020, 2024-01-21:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

with today's git HEAD, the following has arisen:
 
The ```idle_*.cmd``` scripts no longer seem to be called in the latest commit (8d5ae79). Very likely due to [this commit](https://github.com/HASwitchPlate/openHASP/commit/9864ae7a87a7cbd1d09b0c4c685763c038336f98)

Right now, I have 
```MQTT PUB: idle => long``` in the logs, but no ```MSGR: Loading L:/idle_long.cmd```

### To Reproduce

* get the latest commit (8d5ae79)
* create a number of pages (one having number 11 in my example)
* have the following in ```idle_long.cmd```
```
json ['page 11']
```
* when sending the command "page 11", the page shows.
* go to another page
* upon idle_long, the script will not be called and the page will not be called.
* when reverting to commit 0ebfe7d, it works again.

### Expected behavior

upon idle_long, the idle_long.cmd script should be called. (and idle_short, and idle_off,...)

### Screenshots or video

yeyeto2788, 2024-01-26:
Hey @oklona aren't the filenames incorrect? Based on the docs those are supposed to have `-` instead of `_`.

Even if the names are different, did that fully work?

oklona, 2024-01-26:
Yeah, I just copied it from the "dispatch_idle" function below in the source file, which I don't see being called from anywhere. I didn't research much, but it seems to me that "dispatch_idle" was previously used, but was replaced by "dispatch_idle_state". Therefore I assumed I had simply misunderstood the docs. So I did change my file names to use underscore instead of dash. -But apart from that, it worked 100%. Obviously, changing the underscore to dash in the source code would do the same, but during debug I see that underscore is used also for mqtt_on.cmd, mqtt_off.cmd and such. So to me it seems like at some point developers have decided to use underscore.

yeyeto2788, 2024-01-26:
Oh yeah!!! It works 😄. Thanks for that, I guess it would be great to create a PR to the repo.

hb020, 2024-01-26:
It could be wise to check with the reason behind the [revert commit mentioned above](https://github.com/HASwitchPlate/openHASP/commit/9864ae7a87a7cbd1d09b0c4c685763c038336f98). It moved the script calls to ```dispatch_idle()```, and right now the only places that function gets called is during init phases. It might just be a typo, maybe there is a mix up between ```dispatch_idle()``` and ```dispatch_idle_state()```, but anyway, there is reason behind that revert commit. I was hoping for @fvanroie to chime in, he did the commit.

oklona, 2024-01-26:
Agreed. This looks like it has been forgotten somewhere (I cannot find anywhere in the code actually calling `dispatch_idle_state()` currently). So, since I am very new to this, I only chime in with what could possibly fix this, and leave up to others to decide whether my solution is what they want :-)

hb020, 2024-01-27:
Created pull request

hb020, 2024-01-28:
PR accepted, tested, closing the issue.

closed 2024-01-28 by hb020