HASwitchPlate/openHASP #638: Screen unstable on ESP32-8048S048 when there is activity

FredericG-BE, 2024-02-15:
### Perform all steps below and tick them with [x]
- [ x ] Check the related part of the [Documentation](https://www.openhasp.com/)
- [ x ] Update openHASP to the latest version
- [ x ] Reproduce the issue and describe all steps

### Describe the bug
When there is looping text or when flashing new FW, the screen jumps up and down.
At first I thought it was related to screen updates but probably it is related to CPU load. I just discovered this also happens when saving a file in the file editor

### To Reproduce
I used this sjon for the attached video
{
    "page": 1,
    "id": 1,
    "x": 0,
    "y": 0,
    "w": 300,
    "h": 100,
    "obj": "label",
    "text": "This is looping text",
    "text_font": 80,
    "mode": "loop"
}
{
    "page": 1,
    "id": 2,
    "x": 0,
    "y": 150,
    "w": 300,
    "h": 100,
    "obj": "label",
    "text": "This is looping text",
    "text_font": 80,
    "mode": "loop"
}
{
    "page": 1,
    "id": 3,
    "x": 0,
    "y": 300,
    "w": 300,
    "h": 100,
    "obj": "label",
    "text": "This is looping text",
    "text_font": 80,
    "mode": "loop"
}

### Expected behavior
Stable screen

### Screenshots or video

https://github.com/HASwitchPlate/openHASP/assets/11998085/ff970092-cf9a-4e9f-aa27-327447711120

FredericG-BE, 2024-02-16:
I see. 
Out of interest, it seems the stock/demo code these boards are shipped with does not show this issue. Would they do special tricks? 

I propose we mention it in the documentation. Will all Sunton devices with PSram suffer from this?

fvanroie, 2024-02-16:
It depends what the demo software does... I highly doubt they will be using TrueType fonts and scrolling animations.
If you load all graphics objects into RAM then it won't be an issue, but if you start using the PSram or flash at high rates, it affects the display.

closed 2024-03-10 by fvanroie