HASwitchPlate/openHASP #663: esp32-2432s024c support

Psytoshgen, 2024-03-08:
Hi,
Can you add esp32-2432s024c support?

https://www.whizz.ae/product/19714260/diymalls-2-4-inch-esp32-display-ili9341-tft-lcd-module-esp32-2432s024c-capacitive-touch-screen-240x320-esp-wroom-32-development-board-for-arduino-ide/

cosmos1978, 2024-03-16:
here are the board definitions for all of the CYD displays
https://github.com/rzeldent/platformio-espressif32-sunton?tab=readme-ov-file
https://github.com/rzeldent/esp32-smartdisplay

I would be very happy if this can work on those displays

fvanroie, 2024-04-29:
Are you interested in creating and testing a config for this device?
I don't own the particular screen, so your help would be appreciated.

cosmos1978, 2024-04-29:
Sure I can help. I have a few of these boards here. esp32-2432s022c 
Its a very interesting board since it has a battery management circuit onboard.

I will need some help to get me started.

Psytoshgen, 2024-05-01:
I’m not good in coding, but I will try

tinloaf, 2024-05-16:
I also have two esp32-2431S024 displays lying around and would be willing to help testing.

Rudo1111, 2025-02-04:
hello, does anyone have the esp32 2432s024c (C-USB) board working via esphome in Home Assistant? please yaml. thank you

[image]

[image]

[image]

Rudo1111, 2025-02-04:
`esphome:
  name: esp32-wroom-28-ili9341-c1
  friendly_name: ESP32 Wroom 2.8" ILI9341 č.1
  on_boot:
    priority: 600
    then:
      - delay: 10s

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  - platform: esphome
    password: "xxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  output_power: 8.5dB  # Skús znížiť napríklad na 5dB

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Wroom-28-Ili9341-C1"
    password: "h3j91yL94PEs"

captive_portal:
web_server:
  port: 80

i2c:
  sda: GPIO27
  scl: GPIO22
  scan: true
  id: bus_a

spi:
  - id: lcd
    clk_pin: GPIO14
    mosi_pin: GPIO13
    miso_pin: GPIO12
  - id: touch
    clk_pin: GPIO25
    mosi_pin: GPIO32
    miso_pin: GPIO39

display:
  - platform: ili9xxx
    model: ILI9341
    spi_id: lcd
    cs_pin: 15
    dc_pin: 2
    rotation: 90
    invert_colors: false
    update_interval: 5s  
    pages:
      - id: page1
        lambda: |-
          std::map<std::string, std::string> days = {
          {"Monday", "Pondelok"},
          {"Tuesday", "Utorok"},
[... 298 lines cut ...]
  - file: "fonts/Roboto-Regular.ttf"
    id: roboto
    size: 25

image:
  - file: "pictures/power-off-button.png"   # Cesta k obrázku
    id: on_off_button
#    type: RGB565      # Použiť farebný obrázok tj. pre lambda bide obrazok z bielim pozadim: it.image(50, 50, id(on_off_button));
  - file: "pictures/radio-station.png"   # Cesta k obrázku
    id: on_off_radio
#    type: RGB565      # Použiť farebný obrázok
  - file: "pictures/lamp-outline-custom.png"   # Cesta k obrázku
    id: on_off_lamp
#    type: RGB565      # Použiť farebný obrázok

# Time Configuration
time:
  - platform: homeassistant
    id: current_time

globals:
  - id: brightness_level
    type: float
    restore_value: yes
    initial_value: '0.5'  # Štartovací jas na 50 %`

Rudo1111, 2025-02-04:
[image]
[image]

Rudo1111, 2025-02-04:
So I managed to do it and created a working yaml for esp32 2432s024c (C-USB), so if anyone wants to they can use it 😊. `i2c:
  sda: GPIO21  # Orig gpio 27 for 2.8" LCD but for 2.4" gpio 21
  scl: GPIO22   # Orig gpio 22 for 2.8" LCD but for 2.4" gpio 22
  scan: true
#  id: bus_a

spi:
  - id: lcd
    clk_pin: GPIO14     # Orig gpio 14 for 2.8" LCD but for 2.4" gpio 14
    mosi_pin: GPIO13  # Orig gpio 13 for 2.8" LCD but for 2.4" gpio 13
    miso_pin: GPIO12  # Orig gpio 12 for 2.8" LCD but for 2.4" gpio 12
#  - id: touch
#    clk_pin: GPIO25     # Orig gpio 25 for 2.8" LCD but for 2.4" no
#    mosi_pin: GPIO32  # Orig gpio 32 for 2.8" LCD but for 2.4" no
#    miso_pin: GPIO39  # Orig gpio 39 for 2.8" LCD but for 2.4" no

display:
  - platform: ili9xxx
    model: ILI9341
    spi_id: lcd
    cs_pin: 15       # Orig gpio 15 for 2.8" LCD but for 2.4" gpio 15
    dc_pin: 2         # Orig gpio 2 for 2.8" LCD but for 2.4" gpio 15
#    reset_pin: 3
    rotation: 180
    invert_colors: false
    update_interval: 5s  
    dimensions:
       height: 240
       width: 320

touchscreen:
  platform: xpt2046
  spi_id: touch
  cs_pin: 33                # Orig gpio 33 for 2.8" LCD but for 2.4" gpio 33
#  interrupt_pin: 36      # Orig gpio 36 for 2.8" LCD but for 2.4" gpio no
  update_interval: 150ms  
  threshold: 400
  calibration:              # Setting about for 2.4 " ???
    x_min: 200
    x_max: 3900
    y_min: 200
    y_max: 3900
  transform:
    mirror_x: true
    mirror_y: false
    swap_xy: false

output:
  - platform: ledc
    pin: GPIO27            #  Orig gpio 21 for 2.8" LCD but for 2.4" is ok 27 
    id: former_led_pin
  - platform: ledc
    id: output_red
    pin: GPIO4             # Orig gpio 33 for 2.8" LCD but for 2.4" gpio 4 is ok
    inverted: true
  - platform: ledc
    id: output_green
    pin: GPIO16         # Orig gpio 16 for 2.8" LCD but for 2.4" gpio 16 is ok
    inverted: true
  - platform: ledc
    id: output_blue
    pin: GPIO17        # Orig gpio 17 for 2.8" LCD but for 2.4" gpio 17 is ok
    inverted: true

light:
  - platform: monochromatic
    output: former_led_pin
    name: "Display Backlight"
    id: back_light
    restore_mode: ALWAYS_ON
  - platform: rgb
    name: LED
    id: led
    red: output_red
    green: output_green
    blue: output_blue
    restore_mode: ALWAYS_OFF

[image]

`

fvanroie, 2025-02-05:
Please post ESPhome yaml files in the appropriate repository, thanks.

Rudo1111, 2025-02-05:
What pins are used for uart on the P1 bus for RX and TX? And how to set in yaml for Home Assistant Audio - speaker on GPIO 26. Thanks

fvanroie, 2025-02-05:
This is an openHASP discussion thread, not for ESPhome related questions.

ervnaz, 2025-10-02:
I also have a few esp32-2432s024c  and will be happy to try and make openhasp work on these

kjmerlinz, 2025-10-17:
I have one of these esp32-2432s024c running openhasp. Found files here https://github.com/elik745i/ESP32-2432s024c but they havent been updated for a long time. Would be great to have this properly supported.