HASwitchPlate/openHASP #667: Black Screen only on Sunton esp32_3248035C (capacitive)

wildblue, 2024-03-10:
### Perform all steps below and tick them with [x]
- [ ] Check the related part of the [Documentation](https://www.openhasp.com/)
- [ X] Update openHASP to the latest version
- [ X] Reproduce the issue and describe all steps

### Describe the bug
After installing the binary, the Screen stays black. I tried the Version from nightly webinstaller (full) and from GitHub actions after the enty about the blackscreen.

### To Reproduce
Install and see nothing is displayed.

### Expected behavior
At least the QR-Code or anything is displayed 

### Screenshots or video
No Image, Just try to imagine a black screen.

wildblue, 2024-03-11:
Changed Backlight Pin from 21 to 27 and Display works. you have to change that Default value for the capacitative Version of the Display to Pin 27.

fvanroie, 2024-03-11:
Pin 27 is already the [default backlight pin](https://github.com/HASwitchPlate/openHASP/blob/916cae2b28cda07827c02b4d8bd04355702099d5/user_setups/esp32/esp32-3248s035.ini#L77) for this board.

closed 2024-03-11 by fvanroie