HASwitchPlate/openHASP #672: No Ampersand sign for entering wifi passwords

NateChurch, 2024-03-14:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

When performing a manual entry of a wifi password using the touch screen keyboard, there is no ampersand(&) sign on the symbols page. 

### To Reproduce

1. Click the *Factory Reset* from the openHASP menu.
2. touch the screen and attempt to configure the wifi using the on-screen keyboard

### Expected behavior
There is an ampersand on the on-screen keyboard. This looks like a limitation to the LVGL library.

fvanroie, 2024-03-14:
This character was missing from the virtual keyboard upstream. This has been added and should be available in [the latest build](https://github.com/HASwitchPlate/openHASP/actions/runs/8283081859).

fvanroie, 2024-04-16:
The ampersand is now available in the symbols:

[image]

closed 2024-04-16 by fvanroie