HASwitchPlate/openHASP #676: Improve lvgl memory

1achy, 2024-03-15:
<html>
<body>

Device Memory
--
Free Heap | 65.80 KiB
Free Block | 55.98 KiB
Fragmentation | 14%
PSRam Free | 7.18 MiB
PSRam Size | 7.99 MiB

Total | 64.00 KiB
Free | 5.61 KiB
Fragmentation | 14%

</body>
</html>

1achy, 2024-03-15:
it s block if add rows

1achy, 2024-03-15:
`
{"comment":"---- Config ---- ","page_size":"small_portrait","project_name":"openhasp"}

{"page":0,"id":10,"obj":"label","x":660,"y":10,"h":30,"w":120,"text":"00.0°C","text_font":18,"align":"right","bg_color":"#46B2ED","text_color":"#FFFFFF"}
{"page":0,"id":11,"obj":"label","x":10,"y":10,"h":30,"w":120,"text":"00:00","text_font":18,"align":"left","bg_color":"#46B2ED","text_color":"#FFFFFF"}
{"comment":" ----------- icone------------"}
{"page":0,"id":6,"obj":"btn","x":660,"y":0,"w":30,"h":40,"text_font":"2","text":"\uE5A9","text_color":"gray","bg_opa":0,"border_width":0}
{"page":0,"id":7,"obj":"btn","x":620,"y":0,"w":30,"h":40,"text_font":"2","text":"\uEAD7","text_color":"gray","bg_opa":0,"border_width":0}

{"comment":"----------- page 1 menu------- "}

{"page":1, "obj":"btn","id":100,"x":0,"y":0,"w":800,"h":40,"text":"HOME","text_font":18,"bg_color":"#000000","text_color":"#FFFFFF","radius":0,"border_side":0}

{"page":1,"id":1,"obj":"obj","x":10,"y":50,"w":240,"h":150,"border_color":"#000000","radius":0,"bg_color":"#42a5f6"}
{"page":1,"id":101,"obj":"arc","x":60,"y":20,"w":130,"h":130,"parentid":1,"border_width":0,"radius":0,"bg_opa":0,"type":0,"max":6900, "line_width": 1, "line_width1": 5, "line_color": "#ffffff", "line_color1": "#ffffff", "value_str": "0000 W", "value_color": "#ffffff"}
{"page":1,"id":102,"obj":"label", "parentid":1, "x": 5, "y": 130, "w":120, "text_color":"#FFFFFF", "text_font":12, "text":"Energia"}

{"page":1,"id":2,"obj":"btn","x":260,"y":50,"w":75,"h":75,"border_color":"#000000","radius":0,"bg_color":"#4cb050", "text_color":"#FFFFFF", "text_font": 32,"text":"\uE438", "value_str": "Termosif.","action": "p3", "value_font": 12, "value_color": "#ffffff", "value_ofs_y": "22", "value_ofs_x": "0"}

{"page":1,"id":3,"obj":"btn","x":335,"y":50,"w":75,"h":75,"border_color":"#000000","radius":0,"bg_color":"#42a5f6", "text_color":"#FFFFFF", "text_font": 32,"text":"\uE502", "value_str": "TV", "value_font": 12, "value_color": "#ffffff", "value_ofs_y": "22", "value_ofs_x": "-20"}

{"page":1,"id":4,"obj":"btn","x":260,"y":125,"w":75,"h":75,"border_color":"#000000","radius":0,"bg_color":"#ffb74e", "text_color":"#FFFFFF", "text_font": 32,"text":"\uE004", "value_str": "Helena", "value_font": 12, "value_color": "#ffffff", "value_ofs_y": "22", "value_ofs_x": "-10"}

{"page":1,"id":5,"obj":"btn","x":335,"y":125,"w":75,"h":75,"border_color":"#000000","radius":0,"bg_color":"#4cb050", "text_color":"#FFFFFF", "text_font": 32,"text":"\uE68A", "value_str": "Alarme", "value_font": 12, "value_color": "#ffffff", "value_ofs_y": "22", "value_ofs_x": "-10"}

{"page":1,"id":8,"obj":"obj","x":420,"y":50,"w":150,"h":150,"border_color":"#000000","radius":0,"bg_color":"#42a5f6"}
{"page":1,"id":81,"obj":"arc","x":20,"y":20,"w":100,"h":100,"parentid":8,"border_width":0,"radius":0,"bg_opa":0,"type":0,"max":100, "line_width": 1, "line_width1": 5, "line_color": "#ffffff", "line_color1": "#ffffff", "value_str": "00°C", "value_color": "#ffffff"}
{"page":1,"id":82,"obj":"label", "parentid":8, "x": 5, "y": 100, "w":120, "text_color":"#FFFFFF", "text_font":12, "text":"AQS"}

{"page":1,"id":6,"obj":"btn","x":580,"y":50,"w":75,"h":75,"border_color":"#000000","radius":0,"bg_color":"#fc1a2e", "text_color":"#FFFFFF", "text_font": 32,"text":"\uF1DB", "value_str": "Janelas", "action": "p5", "value_font": 12, "value_color": "#ffffff", "value_ofs_y": "22", "value_ofs_x": "-5"}

{"page":1,"id":7,"obj":"btn","x":655,"y":50,"w":75,"h":75,"border_color":"#000000","radius":0,"bg_color":"#42a5f6", "text_color":"#FFFFFF", "text_font": 32,"text":"\uE335", "value_str": "Luci", "action": "p2", "value_font": 12, "value_color": "#ffffff", "value_ofs_y": "22", "value_ofs_x": "-10"}

{"page":1,"id":9,"obj":"btn","x":580,"y":125,"w":75,"h":75,"border_color":"#000000","radius":0,"bg_color":"#42a5f6","text_color":"#FFFFFF", "text_font": 32,"text":"\uE0AC", "value_str": "Sala", "value_font": 12, "value_color": "#ffffff", "value_ofs_y": "22", "value_ofs_x": "-5"}

{"page":1,"id":10,"obj":"btn","x":655,"y":125,"w":75,"h":75,"border_color":"#000000","radius":0,"bg_color":"#42a5f6","text_color":"#FFFFFF", "text_font": 32,"text":"\uE0AC", "value_str": "Tapparelle", "value_font": 12, "value_color": "#ffffff", "value_ofs_y": "22", "value_ofs_x": "-5"}

{"page":1,"id":11,"obj":"obj","x":10,"y":210,"w":180,"h":75,"border_color":"#000000","radius":0,"bg_color":"#5d35b0", "text_color":"#FFFFFF"}
{"page":1,"id":111,"obj":"label", "parentid":11, "x": 5, "y": 8, "w":60,"h":60,"text_color":"#FFFFFF",  "text_font": 50, "action": "p6", "text":"\uE156"}
{"page":1,"id":112,"obj":"label", "parentid":11, "x": 65, "y": 10, "w":135,"h":60,"text_color":"#FFFFFF",  "mode": "break", "text_font": 16, "action": "p7", "text":"Connection..."}
{"page":1,"id":113,"obj":"label", "parentid":11, "x": 65, "y": 30, "w":135,"h":60,"text_color":"#FFFFFF",  "mode": "break", "text_font": 12, "action": "p7", "text":"Connection..."}

{"page":1,"id":12,"obj":"btn","x":190,"y":210,"w":75,"h":75,"border_color":"#000000","radius":0,"bg_color":"#5d35b0", "text_color":"#FFFFFF", "text_font": 32, "text":"\uE5F1", "value_str": "EVSE", "value_font": 12, "value_color": "#ffffff", "value_ofs_y": "22", "value_ofs_x": "-15"}

{"comment":"----------- page 1 impianto solare------- "}

{"page":1,"id":200,"obj":"obj","x":310,"y":210,"w":330,"h":240,"border_color":"#000000","radius":0,"bg_color":"#5d35b0", "text_color":"#FFFFFF"}

[... cut ...]
{ "page": 7, "id": 103, "obj": "obj", "x": 120, "y": 34, "w": 59, "h": 73, "opacity": 0, "action": {"up": "page 6"}}
{ "page": 7, "id": 104, "obj": "obj", "x": 180, "y": 34, "w": 59, "h": 73, "opacity": 0, "action": {"up": "page 7"}}

{"page":7,"id":150,"obj":"btn","x":680,"y":360,"w":60,"h":60,"border_color":"#000000","radius":0,"bg_color":"#5d35b0", "text_color":"#FFFFFF", "text_font": 32, "text":"\uE6A1", "value_str": "back", "action": "back", "value_font": 12, "value_color": "#ffffff", "value_ofs_y": "22", "value_ofs_x": "-15"}
`

fvanroie, 2024-03-16:
Try increasing ` #  define LV_MEM_SIZE    (64 * 1024U)  // 64Kb on ESP32 with PSram`
or set `#define LV_MEM_CUSTOM      1`

1achy, 2024-03-16:
> Try increasing ` # define LV_MEM_SIZE (64 * 1024U) // 64Kb on ESP32 with PSram` or set `#define LV_MEM_CUSTOM 1`

for make this do u need compile?

1achy, 2024-03-17:
I ever used https://nightly.openhasp.com/   precompiled 0.7 FOR 
[image]

1) The file include\user_config_override-template.h 

2) CAN I USE : Compiling online with Gitpod   (WHAT IS IT LINK FOR 0.7 version)

`/* MIT License - Copyright (c) 2019-2024 Francis Van Roie
   For full license information read the LICENSE file in the project folder */

/***************************************************
// This file contains the default settings that are
// burned into the compiled firmware.
//
// These default settings can be changed at runtime
//
// To use: Save a copy as user_config_override.h
***************************************************/
#ifndef HASP_USER_CONFIG_OVERRIDE_H
#define HASP_USER_CONFIG_OVERRIDE_H

#define SERIAL_SPEED 115200

/***************************************************
          WiFi Settings
 **************************************************/
#define WIFI_SSID ""
#define WIFI_PASSWORD ""

/***************************************************
          Http Server Settings
 **************************************************/
#define HTTP_USERNAME ""
#define HTTP_PASSWORD ""

/***************************************************
          MQTT Client Settings
 **************************************************/
#define MQTT_HOSTNAME ""
#define MQTT_PORT 1883
#define MQTT_USERNAME ""
#define MQTT_PASSWORD ""
#define MQTT_PREFIX "hasp"
#define MQTT_NODENAME "plate01"
#define MQTT_GROUPNAME "plates"

#define MQTT_TELEPERIOD 60000
#define MQTT_STATEPERIOD 300000

/***************************************************
 *        OTA Settings
 **************************************************/
//#define HASP_USE_ARDUINOOTA 1                       // Enable the Arduino OTA service
#define ARDUINOOTA_PORT 3232
#define ARDUINOOTA_PASSWORD ""
#define OTA_URL ""

/***************************************************
 *        Syslog Settings
 **************************************************/
#define HASP_USE_SYSLOG 0
#define SYSLOG_SERVER ""
#define SYSLOG_PORT 514
#define APP_NAME "HASP"

/***************************************************
 *        Timezone Settings
 **************************************************/
#define NTPSERVER1 "pool.ntp.org"
#define NTPSERVER2 "time.nist.gov"
[... 74 lines cut ...]
// #define LV_FONT_DEFAULT &HASP_FONT_1

/***************************************************
 *        GPIO Settings
 **************************************************/
//#define HASP_GPIO_TEMPLATE "[197658,263456,329249,655628,655886,656155,0,0]"  // Lanbon L8 3-gang GPIO config
//#define HASP_GPIO_TEMPLATE "[3214348,197658,263456,329249,94699520,0,0,0]" // Lanbon L8 Dimmer GPIO config

/***************************************************
 *        Other Settings
 **************************************************/
//#define HASP_USE_MDNS 0                             // Disable MDNS
//#define HASP_USE_CUSTOM 1                           // Enable compilation of custom code from /src/custom
//#define HASP_USE_HA                                 // Enable Home Assistant auto-discovery
//#define HASP_START_CONSOLE 0                        // Disable starting of serial console at boot
//#define HASP_START_TELNET 0                         // Disable starting of telnet service at boot
//#define HASP_START_HTTP 0                           // Disable starting of web interface at boot
//#define HASP_START_FTP 0                            // Disable starting of ftp server at boot
//#define LV_MEM_SIZE (64 * 1024U)                    // 64KiB of lvgl memory (default 48)
//#define LV_VDB_SIZE (32 * 1024U)                    // 32KiB of lvgl draw buffer (default 32)
//#define HASP_DEBUG_OBJ_TREE                         // Output all objects to the log on page changes
//#define HASP_LOG_LEVEL LOG_LEVEL_VERBOSE            // LOG_LEVEL_* can be DEBUG, VERBOSE, TRACE, INFO, WARNING, ERROR, CRITICAL, ALERT, FATAL, SILENT
//#define HASP_LOG_TASKS                              // Also log the Taskname and watermark of ESP32 tasks

#endif // HASP_USER_CONFIG_OVERRIDE_H`

1achy, 2024-03-20:
Where do i found source 0.7?

fvanroie, 2024-03-20:
> Where do i found source 0.7?

It is here: https://github.com/HASwitchPlate/openHASP

1achy, 2024-03-23:
> > Where do i found source 0.7?
> 
> It is here: https://github.com/HASwitchPlate/openHASP
> 
> For Gitpod use https://gitpod.io/new/#https://github.com/HASwitchPlate/openHASP/tree/master

i cant found 0.7, i need for sutun 7 inch

[image]

1achy, 2024-03-23:
sunton esp 7 inch CORRECT  + Try increasing  

#  define LV_MEM_SIZE    (64 * 1024U)  // 64Kb on ESP32 with PSram or set #define LV_MEM_CUSTOM      1

**platform_overdrive :**

; USAGE: Copy this file and rename it to platformio_override.ini
;
; The platformio_override.ini file is not overwritten or monitored by git
; ONLY edit platformio_override.ini to make local changes to the parameters

[platformio]
extra_configs =
    ; Uncomment or edit the lines to show more User Setups in the PIO sidebar
    ; user_setups/darwin/*.ini
    ; user_setups/esp32/*.ini
    ; user_setups/esp32s2/*.ini
     user_setups/esp32s3/*.ini
    ; user_setups/linux/*.ini
    ; user_setups/stm32f4xx/*.ini
    ; user_setups/win32/*.ini

[override]
; -- Hasp config options --------------------------------------
build_flags =
; -- Uncomment the next line to use the file include/user_config_override.h settings
;    -DUSE_CONFIG_OVERRIDE

;region -- Default Build Environments : Used when Build All ---
extra_default_envs =
    ; Uncomment specific environments or create extra:
    ; az-touch-mod-esp32_ili9341_4MB
    ; az-touch-mod-esp32_ili9341_8MB
    ; d1-mini-esp32_ili9341
    ; d1-r32-unoshield_ili9341_adc
    ; d1-r32-unoshield_ili9486_adc
    ; d1-r32-waveshare_ili9486
    ; esp32s2-featherwing-24-v2
    ; esp32s2-featherwing-35-v2
    ; esp32-2432s028r_4MB
    ; esp32-3248s035c_4MB
    ; esp32-3248s035c_4MB_lovyan
    ; esp32-3248s035r_4MB
    ; esp32-one_ili9486
    ; esp32-one_st7796
    ; esp32-s3-4827s043c_16MB
    ; esp32-s3-4827s043r_16MB
    ; esp32-s3-4848S040_16MB
    ; esp32-s3-8048s043c_16MB
    ; esp32-touchdown
    ; freetouchdeck_4MB
    ; freetouchdeck_8MB
    ; freetouchdeck-s3_8MB
    ; gs-t3e_16MB
    ; huzzah32-featherwing-24
    ; huzzah32-featherwing-24-v2
    ; huzzah32-v2-featherwing-24-v2
    ; huzzah32-featherwing-35
    ; huzzah32-featherwing-35-v2
    ; huzzah32-v2-featherwing-35-v2
    ; lanbon_l8
    ; lanbon_l8_eth
    ; lilygo-lily-pi_ili9481
    ; lilygo-lily-pi_st7796
    ; lolin-d32-pro_ili9341
    ; m5stack-core2
    ; m5stack-tough
    ; makerfabs-s3-tft35-spi
    ; makerfabs-tft35-cap_16MB
    ; makerfabs-tft35-cap_4MB
[... 204 lines cut ...]

[tft_espi]
lib_deps =
    bodmer/TFT_eSPI@2.5.43
    ;https://github.com/Bodmer/TFT_eSPI.git#master

[goodix]
lib_deps =
    git+https://github.com/netwizeBE/arduino-goodix.git

[ft6336]
lib_deps =
    git+https://github.com/aselectroworks/Arduino-FT6336U.git

[tsc2007]
lib_deps =
    git+https://github.com/adafruit/Adafruit_TSC2007.git

[gsl1680]
lib_deps =
    git+https://github.com/arovak/GSL2038.git

[stmpe610]
lib_deps =
    adafruit/Adafruit STMPE610@^1.1.6

1achy, 2024-03-24:
What do u choose for sunton 7 inch :  

extra_default_envs =
; Uncomment specific environments or create extra:
; az-touch-mod-esp32_ili9341_4MB
; az-touch-mod-esp32_ili9341_8MB
; d1-mini-esp32_ili9341
; d1-r32-unoshield_ili9341_adc

And what file define 👍

LV_MEM_CUSTOM.???

1achy, 2024-03-28:
I use :

[platformio]
extra_configs =
    ; Uncomment or edit the lines to show more User Setups in the PIO sidebar
    ; user_setups/darwin/*.ini
    ; user_setups/esp32/*.ini
    ; user_setups/esp32s2/*.ini
     user_setups/esp32s3/*.ini
    ; user_setups/linux/*.ini
    ; user_setups/stm32f4xx/*.ini
    ; user_setups/win32/*.ini

[override]
; -- Hasp config options --------------------------------------
build_flags =
; -- Uncomment the next line to use the file include/user_config_override.h settings
    -DUSE_CONFIG_OVERRIDE

;region -- Default Build Environments : Used when Build All ---
extra_default_envs =
    ; Uncomment specific environments or create extra:
    ; az-touch-mod-esp32_ili9341_4MB
    ; az-touch-mod-esp32_ili9341_8MB
    ; d1-mini-esp32_ili9341
    ; d1-r32-unoshield_ili9341_adc
    ; d1-r32-unoshield_ili9486_adc
    ; d1-r32-waveshare_ili9486
    ; esp32s2-featherwing-24-v2
    ; esp32s2-featherwing-35-v2
    ; esp32-2432s028r_4MB
    ; esp32-3248s035c_4MB
    ; esp32-3248s035c_4MB_lovyan
    ; esp32-3248s035r_4MB
    ; esp32-one_ili9486
    ; esp32-one_st7796
    ; esp32-s3-4827s043c_16MB
    ; esp32-s3-4827s043r_16MB
    ; esp32-s3-4848S040_16MB
    ; esp32-s3-8048s043c_16MB
     sunton-8048s070c_16MB
    ; esp32-touchdown
    ; freetouchdeck_4MB
    ; freetouchdeck_8MB
    ; freetouchdeck-s3_8MB

I have this error 👍 
Compiling .pio\build\sunton-8048s070c_16MB\src\drv\tft\tft_driver_posix_fbdev.cpp.o
Compiling .pio\build\sunton-8048s070c_16MB\src\drv\tft\tft_driver_sdl2.cpp.o
Compiling .pio\build\sunton-8048s070c_16MB\src\drv\tft\tft_driver_tftespi.cpp.o
Compiling .pio\build\sunton-8048s070c_16MB\src\drv\tft\tft_driver_win32drv.cpp.o
In file included from src/custom/my_custom_template.cpp:7:
src/hasplib.h:10:10: fatal error: user_config_override.h: No such file or directory

******************************************************************************
* Looking for user_config_override.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:user_config_override.h"
* Web  > https://registry.platformio.org/search?q=header:user_config_override.h
*
******************************************************************************

 #include "user_config_override.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from src/custom/my_custom_fan_template.cpp:14:
src/hasplib.h:10:10: fatal error: user_config_override.h: No such file or directory

******************************************************************************
* Looking for user_config_override.h dependency? Check our library registry!
[... 110 lines cut ...]
*** [.pio\build\sunton-8048s070c_16MB\src\drv\tft\Arduino_PCA9535SWSPI.cpp.o] Error 1
In file included from src/drv/touch/touch_driver_gt911.cpp:7:
include/hasp_conf.h:12:10: fatal error: user_config_override.h: No such file or directory

******************************************************************************
* Looking for user_config_override.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:user_config_override.h"
* Web  > https://registry.platformio.org/search?q=header:user_config_override.h
*
******************************************************************************

 #include "user_config_override.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\sunton-8048s070c_16MB\src\drv\touch\touch_driver_gt911.cpp.o] Error 1
==================================================================== [FAILED] Took 20.16 seconds ====================================================================

Environment            Status    Duration
---------------------  --------  ------------
sunton-8048s070c_16MB  FAILED    00:00:20.155
=============================================================== 1 failed, 0 succeeded in 00:00:20.155 ===============================================================

 *  The terminal process "C:\Users\Achille\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

1achy, 2024-03-29:
I follow this: 

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:w="urn:schemas-microsoft-com:office:word"
xmlns:m="http://schemas.microsoft.com/office/2004/12/omml"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Word.Document>
<meta name=Generator content="Microsoft Word 15">
<meta name=Originator content="Microsoft Word 15">
<link rel=File-List
href="file:///D:/users/U124567/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

<link rel=themeData
href="file:///D:/users/U124567/AppData/Local/Temp/msohtmlclip1/01/clip_themedata.thmx">
<link rel=colorSchemeMapping
href="file:///D:/users/U124567/AppData/Local/Temp/msohtmlclip1/01/clip_colorschememapping.xml">

<style>

</style>

</head>

<body lang=IT link=blue vlink="#954F72" style='tab-interval:35.4pt;word-wrap:
break-word'>

<p class=MsoNormal style='margin-top:19.2pt;margin-right:0cm;margin-bottom:
7.7pt;margin-left:0cm;line-height:normal;mso-outline-level:2;background:white'><span
lang=EN-US style='font-size:18.5pt;font-family:Roboto;mso-fareast-font-family:
"Times New Roman";mso-bidi-font-family:"Times New Roman";color:black;
mso-color-alt:windowtext;letter-spacing:-.1pt;mso-font-kerning:0pt;mso-ligatures:
none;mso-ansi-language:EN-US;mso-fareast-language:IT'>COMPILING</span><span
lang=EN-US style='font-size:18.5pt;font-family:Roboto;mso-fareast-font-family:
"Times New Roman";mso-bidi-font-family:"Times New Roman";letter-spacing:-.1pt;
mso-font-kerning:0pt;mso-ligatures:none;mso-ansi-language:EN-US;mso-fareast-language:
IT'><o:p></o:p></span></p>

<p class=MsoNormal style='mso-margin-top-alt:auto;mso-margin-bottom-alt:auto;
line-height:normal;background:white'><span lang=EN-US style='font-size:12.0pt;
font-family:Roboto;mso-fareast-font-family:"Times New Roman";mso-bidi-font-family:
"Times New Roman";color:black;mso-color-alt:windowtext;mso-font-kerning:0pt;
mso-ligatures:none;mso-ansi-language:EN-US;mso-fareast-language:IT'>A copy
of&nbsp;</span><span lang=EN-US style='font-size:10.0pt;font-family:"var(--md-code-font-family)",serif;
mso-fareast-font-family:"Times New Roman";mso-bidi-font-family:"Courier New";
color:black;mso-color-alt:windowtext;mso-font-kerning:0pt;mso-ligatures:none;
mso-ansi-language:EN-US;mso-fareast-language:IT'>platformio_override-template.ini</span><span
lang=EN-US style='font-size:12.0pt;font-family:Roboto;mso-fareast-font-family:
"Times New Roman";mso-bidi-font-family:"Times New Roman";color:black;
mso-color-alt:windowtext;mso-font-kerning:0pt;mso-ligatures:none;mso-ansi-language:
EN-US;mso-fareast-language:IT'>&nbsp;to&nbsp;</span><span lang=EN-US
style='font-size:10.0pt;font-family:"var(--md-code-font-family)",serif;
mso-fareast-font-family:"Times New Roman";mso-bidi-font-family:"Courier New";
color:black;mso-color-alt:windowtext;mso-font-kerning:0pt;mso-ligatures:none;
mso-ansi-language:EN-US;mso-fareast-language:IT'>platformio_override.ini</span><span
lang=EN-US style='font-size:12.0pt;font-family:Roboto;mso-fareast-font-family:
"Times New Roman";mso-bidi-font-family:"Times New Roman";color:black;
mso-color-alt:windowtext;mso-font-kerning:0pt;mso-ligatures:none;mso-ansi-language:
EN-US;mso-fareast-language:IT'>&nbsp;will be readily prepared for you.<br>
Click&nbsp;</span><span lang=EN-US style='font-size:10.0pt;font-family:"var(--md-code-font-family)",serif;
mso-fareast-font-family:"Times New Roman";mso-bidi-font-family:"Courier New";
color:black;mso-color-alt:windowtext;mso-font-kerning:0pt;mso-ligatures:none;
mso-ansi-language:EN-US;mso-fareast-language:IT'>platformio_override.ini</span><span
lang=EN-US style='font-size:12.0pt;font-family:Roboto;mso-fareast-font-family:
"Times New Roman";mso-bidi-font-family:"Times New Roman";color:black;
mso-color-alt:windowtext;mso-font-kerning:0pt;mso-ligatures:none;mso-ansi-language:
EN-US;mso-fareast-language:IT'>&nbsp;to open it, and to select your firmware
flavour for your hardware uncomment in the&nbsp;</span><span lang=EN-US
[... 194 lines cut ...]
color:black;mso-color-alt:windowtext;mso-ansi-language:EN-US'>include\user_config_override.h</span></code><span
lang=EN-US style='font-family:Roboto;color:black;mso-color-alt:windowtext;
mso-ansi-language:EN-US'>&nbsp;and change the settins per your needs.<br
style='box-sizing: inherit'>
Uncomment&nbsp;</span><code style='box-sizing: inherit;color:var(--md-code-fg-color);
font-feature-settings: "kern";background-color:var(--md-code-bg-color);
border-radius: 0.1rem;-webkit-box-decoration-break: clone;word-break:break-word;
-webkit-tap-highlight-color: transparent;outline: none'><span lang=EN-US
style='font-size:10.0pt;font-family:"var(--md-code-font-family)",serif;
color:black;mso-color-alt:windowtext;mso-ansi-language:EN-US'>-DUSE_CONFIG_OVERRIDE</span></code><span
lang=EN-US style='font-family:Roboto;color:black;mso-color-alt:windowtext;
mso-ansi-language:EN-US'>&nbsp;in&nbsp;</span><code style='box-sizing: inherit;
color:var(--md-code-fg-color);font-feature-settings: "kern";background-color:
var(--md-code-bg-color);border-radius: 0.1rem;-webkit-box-decoration-break: clone;
word-break:break-word;-webkit-tap-highlight-color: transparent;outline: none'><span
lang=EN-US style='font-size:10.0pt;font-family:"var(--md-code-font-family)",serif;
color:black;mso-color-alt:windowtext;mso-ansi-language:EN-US'>platformio_override.ini</span></code><span
lang=EN-US style='font-family:Roboto;color:black;mso-color-alt:windowtext;
mso-ansi-language:EN-US'>&nbsp;to ensure that the config overrides are taken
into account during the compilation process.</span><span lang=EN-US
style='font-family:Roboto;mso-ansi-language:EN-US'><o:p></o:p></span></p>

</body>

</html>

1achy, 2024-03-30:
The problen is rename: include\user_config_override-template.h to include\user_config_override.h

NOW where can found : 

 STEP 5 Parameters lvgl

??????????????????????
define LV_MEM_SIZE (64 * 1024U) // 64Kb on ESP32 with PSram
define LV_MEM_CUSTOM 1
??????????????????????

1achy, 2024-03-30:
> i use [Sunton ESP32-S3 7" 800*480](https://www.makerfabs.com/sunton-esp32-s3-7-inch-tn-display-with-touch.html)
> 
> ## Device Memory
> Free Heap | 65.80 KiB Free Block | 55.98 KiB Fragmentation | 14% PSRam Free | 7.18 MiB PSRam Size | 7.99 MiB
> 
> Total | 64.00 KiB **Free | 5.61 KiB** Fragmentation | 14%
> 
> out of memory ..... block screen...
> 
> solution reflash firmware ...

Now i have :   

<html>
<body>

Device Memory
--
Free Heap | 60.83 KiB
Free Block | 49.98 KiB
Fragmentation | 17%
PSRam Free | 7.19 MiB
PSRam Size | 7.99 MiB

Total | 64.00 KiB
Free | 10.86 KiB
Fragmentation | 18%

</body>
</html>

1achy, 2024-03-30:
is it better use define LV_MEM_CUSTOM 1 ?

marsman7, 2024-04-01:
> is it better use define LV_MEM_CUSTOM 1 ?

I don't think that's a good idea for you, you would have to programme your own memory management.

fvanroie, 2024-04-01:
`LV_MEM_CUSTOM 1` just means to use the default heap memory pool instead of the reserved LVGL block. It uses the standard C memory management of malloc/free/etc. It has been used and should work fine... the only downside I can think of is a higher fragmentation, which should be compensated by the larger pool.

When you start customizing the firmware, it is up to you to test it and tweak where needed.

1achy, 2024-04-01:
> `LV_MEM_CUSTOM 1` just means to use the default heap memory pool instead of the reserved LVGL block. It uses the standard C memory management of malloc/free/etc. It has been used and should work [...]

Di u have same guide for improve this mem allocaror?

fvanroie, 2024-04-02:
Just change `#define LV_MEM_CUSTOM      0` to `#define LV_MEM_CUSTOM      1` in `include\lv_conf_v7.h` and then recompile.

1achy, 2024-04-06:
i try :
[image]

lvgl memory dont change :

[image]

1achy, 2024-04-06:
> Just change `#define LV_MEM_CUSTOM 0` to `#define LV_MEM_CUSTOM 1` in `include\lv_conf_v7.h` and then recompile.

so i  do

hb020, 2024-04-11:
Just created a PR that will allow avoiding to have to modify `lv_conf_v7.h` or `lv_conf_v8.h`, and will allow control from `user_config_override.h`, as is expected.

fvanroie, 2024-04-16:
You'll need to keep an eye on the memory fragmentation on the info page.
I doubt the WiFi signal issue has anything to do with the LVGL memory option, but please let me know if you find anything conclusive.

MrEcosse, 2024-04-23:
I've used this method too to overcome an issue with both the Sunton and Guition screen crashing due to lack of available LVGL memory.  Forgive my ignorance here (my first program was input via paper tape on an acoustic coupler telephone line) but with these higher resolution screens would it be possible for the default LVGL allocation to be higher?  I am not able to create as many screens or use as many objects as I do on my Lanbon screens without having to do a custom compile.

fvanroie, 2024-04-29:
What settings did you change? Were there any adverse effects?
If there needs to be a different limit, please propose a change or PR.

I can't know what number of pages and objects are being used by everybody, so your feedback is needed to come up with a reasonable default value. The current default have been in place for quite a while and are quite stable. We can increase the LVGL memory, but the change needs to be justified and stable. What limit do you suggest?

fvanroie, 2024-05-13:
No suggestions?

MrEcosse, 2024-05-13:
Sorry been travelling again.  I changed LV_MEM_CUSTOM 1 as noted in a previous thread.  So far no noticeable problems in the past 3 weeks.  Fragmentation had increased to 70%.  I just rebooted one of the panels and fragmentation now 45%.  I'll add notes if anything changes/goes wrong!

closed 2024-07-21 by fvanroie