HASwitchPlate/openHASP #684: CrowPanel 7 Inch support.

WarC0zes, 2024-03-21:
Hi,
Can you add support for it's ESP WZ8048C070 ?

https://www.elecrow.com/esp32-display-7-inch-hmi-display-rgb-tft-lcd-touch-screen-support-lvgl.html

Specification:

Model: 7.0 inch -CrowPanel ESP32 display
Main Processor: ESP32-S3-WROOM-1-N4R8
Resolution: 800*480
Flash: 4MB
PSRAM: 8MB

I use these settings and the screen works fine with 0.7.0rc8 version. I haven't tried the latest 0.7.0rc11. 

settings:
https://github.com/HASwitchPlate/openHASP/discussions/559
https://github.com/HASwitchPlate/openHASP/discussions/559#discussioncomment-7922944

fvanroie, 2024-04-12:
This board has now been added to https://nightly.openhasp.com/

WarC0zes, 2024-04-12:
Hi,
I updated via ÔTA. During the update, the screen was shaking but it's good.
I'm going to do more in-depth testing, but for now it's working well.

[image]
[image]

Thank you very much for the addition, I'm so happy.

WarC0zes, 2024-04-12:
I don't see the option for Wireguard. I don't use Wireguard, it doesn't bother me.

Crowpanel 7inch 0.7.0-rc12
[image]

Sunton 7inch 0.7.0-rc11
[image]

fvanroie, 2024-04-12:
It's a compile time option that's disabled now by default as it hasn't been well tested.
If you need WireGuard you can enable it when compiling.

WarC0zes, 2024-04-13:
> It's a compile time option that's disabled now by default as it hasn't been well tested. If you need WireGuard you can enable it when compiling.

I thought it was a bug. No worries, I don't use wireguard ;-)

smcgann99, 2024-04-26:
@fvanroie Have you tested the 7" screen yet, interested to see if its better than the sunton ?

WarC0zes, 2024-04-26:
@smcgann99,
I have both, a 7 inch sunton and 7 inch elecrow. The Elecrow only has 4MB of flash compared to 16MB on the Sunton. Afterwards they both work as well as the other for me.

The sunton I tinkered with to stabilize the wifi, I don't know if there has been a fix since.

closed 2024-05-24 by fvanroie

enjoy-development, 2024-11-20:
Hello
does some have an image (bin) for the elecrow Crowpanel 7 Version3 ?@WarC0zes @fvanroie 
i tried the nightbuild and made my own via gitpod, but both won't work with touch . 
With ESPHOME (same Crowpanel) does the Touch work. very strange  

Touchsettings in crowpanel-hmi.ini are similar to ESP Home Settings

see https://github.com/HASwitchPlate/openHASP/blob/master/user_setups/esp32s3/crowpanel-hmi.ini

But I have the exact values as in ESPHome where the same touchpanel works
; Touch Settings
    -D HASP_USE_LGFX_TOUCH=1
    -D TOUCH_WIDTH=800
    -D TOUCH_HEIGHT=480
    -D TOUCH_DRIVER=0x911
    -D TOUCH_SCL=20
    -D TOUCH_SDA=19
    -D TOUCH_IRQ=-1
    -D TOUCH_RST=-1
    -D I2C_TOUCH_ADDRESS=0x5D 
    -D I2C_TOUCH_FREQUENCY=400000

are similar to ESPHome Settings
# for Touch          
i2c:
  sda: 19
  scl: 20
touchscreen:
  platform: gt911
  id: LCD_touchscreen
  display: main_display
  update_interval: 50ms
  address: 0x5D

if i run mqtt publish to the device 
hasp/crowpanel_7/state/statusupdate
the rsponse is
..
  "tftDriver": "Other",
...

Any ideas ? 
regard 
Oliver

ThomasCr, 2026-04-06:
Hi.. I also use a `ESP32-S3 CrowPanel Basic 7" 800x480 HMI IPS Touchscreen Display with 240MHz ESP32-S3`  - on the backside it has `SKU: DIS08070H`

I cant get the touchpad to work with `elecrow-s3-8048c070_full_4MB_v0.7.0-rc13_042fe05.bin`. On ESPhome it was running, after fiddeling around. Thats my config...

```
i2c:
  - id: i2c_bus
    sda: 19
    scl: 20
    scan: true

# PCA9554 GPIO extender for display reset
pca9554:
  - id: p_c_a
    address: 0x20
    i2c_id: i2c_bus

output:
  - platform: ledc
    pin: 2
    id: gpio_backlight_pwm
    frequency: 1220

light:
  - platform: monochromatic
    output: gpio_backlight_pwm
    name: "Display Backlight"
    id: display_backlight
    restore_mode: ALWAYS_ON
    default_transition_length: 0s

display:
  - platform: rpi_dpi_rgb
    id: main_display
    invert_colors: true
    dimensions:
      width: 800
      height: 480
    de_pin: 41
    hsync_pin: 39
    vsync_pin: 40
    pclk_pin: 0
    pclk_frequency: 20MHz
    pclk_inverted: true
    hsync_front_porch: 40
    hsync_pulse_width: 48
    hsync_back_porch: 13
    vsync_front_porch: 1
    vsync_pulse_width: 31
    vsync_back_porch: 13
    data_pins:
      red: [14, 21, 47, 48, 45]
      green: [9, 46, 3, 8, 16, 1]
      blue: [15, 7, 6, 5, 4]
    color_order: RGB
    update_interval: 16ms
    auto_clear_enabled: false # takes 2.8 seconds to clear the display

touchscreen:
  - platform: gt911
    id: main_touchscreen
    i2c_id: i2c_bus
    #address: 0x5D
    display: main_display
    update_interval: 50ms

lvgl:
  displays:
    - main_display
  touchscreens:
    - main_touchscreen
  color_depth: 16
  bg_color: 0x000000
  border_width: 0
```