HASwitchPlate/openHASP #685: Use MQTT topic as an image source

hmorkemo, 2024-03-22:
Showing images in a panel is quite static, and the image files have to be locally stored.

I want to be able to send images from HA to the screen.

One solution to this is in PR #643, but it appears to have limitations due to memory usage and possible loss of OTA upgrades on older screens.
A better solution, I think, would be to have the image src parameter listen to an mqtt topic. (Ref #643, which is a good idea in itself).
Then I could say { id: "p2b10", obj: "img", ... , src: "mqtt:weatherforecast1" }

In HA I would then publish to mqtt { topic: "hasp/panel1/weatherforecast1", payload: theimage }.
The image format should be png or whatever OpenHASP can handle, encoded in a standard way.

pepe59, 2024-03-23:
I don't have a better way. I am using sending a weather image using automation
```
service: openhasp.push_image
target:
  entity_id: openhasp.ovladac3
data:
  image: http://1xx.1x.x.xx:8123/local/pocasi/Blesky.png
  obj: p1b2
  width: 120
  height: 120
  fitscreen: true
```

FreeBear-nc, 2024-05-09:
> In HA I would then publish to mqtt { topic: "hasp/panel1/weatherforecast1", payload: theimage }. The image format should be png or whatever OpenHASP can handle, encoded in a standard way.

Not sure you can have a binary file as a payload image with HA. The mqtt docs seem to suggest that payloads are converted to strings before being published.

dosipod, 2024-05-13:
@fvanroie I only started using  openHASP recently because i got Sunton ESP32-2432S028 on offer , i really like what you did , amazing work . I was able to use bin images loaded internally but only few of them .

Being able to use external images i think would be a great addition, not really sure using HA is ideal  ,what do you think the best was to do so even if we  need to buy  better devices  but would also be nice that we could use what we have as the 4MB ESP32  is still very common and we have a ton of them and external display is very cheap .

dosipod, 2024-05-13:
Thank you yes i think that makes sense and ESP32-2432S028 i got on offer was only $2  ,we will try to work on the image conversion if something could be done but I did also face somewhat the same issues with Tasmota and lvgl so i hope that will not be just a dead end

dosipod, 2024-05-14:
I  looked into our stock and found few esp32 cam , will that do ?The aim is just to display images hosted on external server , nothing fancy and touch is not needed so any display is okay  . Did not see that mentioned with openHASP

smcgann99, 2024-05-21:
@fvanroie Sorry this is off topic but - 
_Hence the decoding from JPEG/PNG to BIN is done in Home Assistant as not to burden the ESP32 too much_

I don't use HA, but I do use a http image source pointing to a jpg on my node red server. From the above would it be more efficient to convert to bin and host the BIN file ?

fvanroie, 2024-05-21:
Yes indeed. PNG/JPEG can't be used directly in LVGL and have to be decoded to BIN first. This happens in memory on the ESP32.
You can save memory and CPU cycles when the decoding is done externally, at the expense of a little bit more network traffic.

closed 2024-05-24 by fvanroie