HASwitchPlate/openHASP #686: display does not use manual Node Topic

stefan242, 2024-03-23:
### Perform all steps below and tick them with [x]
- [X] Check the related part of the [Documentation](https://www.openhasp.com/)
- [X] Update openHASP to the latest version
- [X] Reproduce the issue and describe all steps

### Describe the bug
I have two gs-t3e_16MB displays, the one with the described problem just upgraded to Version 0.7.0-rc11 677e8cc.

One display has Hostname plate3, installed in the bath, the second one next to my PC named plate3a to help me install everything.
To get and send the MQTT messages on both of them, I left the original Node Topic on the plate3 display as "hasp/%hostname%/%topic%" but changed the one on my plate3a to "hasp/plate3/%topic%". (I left the default on all other Topic configuration entries.)

But MQTT messages from the plate3a display (in this case a slider update) are send to Node-RED with the topic "hasp/plate3a/state/p2b20"

MQTT updates send *to* the plate3a display are working though.

### To Reproduce
In Node-RED:
```
[{"id":"72c5bbd64d2d050a","type":"mqtt in","z":"63025264bf39ecf8","name":"","topic":"hasp/plate3/state/p2b20","qos":"2","datatype":"auto-detect","broker":"7f211795.773398","nl":false,"rap":true,"rh":0,"inputs":0,"x":210,"y":1200,"wires":[["d689b70b4daee530"]]},{"id":"d689b70b4daee530","type":"debug","z":"63025264bf39ecf8","name":"debug Heizung Bad 4","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":500,"y":1200,"wires":[]},{"id":"19aa16284aa1d027","type":"mqtt in","z":"63025264bf39ecf8","name":"","topic":"hasp/plate3a/state/p2b20","qos":"2","datatype":"auto-detect","broker":"7f211795.773398","nl":false,"rap":true,"rh":0,"inputs":0,"x":830,"y":1200,"wires":[["ce23aebc6fc6951f"]]},{"id":"ce23aebc6fc6951f","type":"debug","z":"63025264bf39ecf8","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":1070,"y":1200,"wires":[]},{"id":"7f211795.773398","type":"mqtt-broker","name":"Mosquitto","broker":"localhost","port":"1883","clientid":"","usetls":false,"compatmode":false,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","closeTopic":"","closeQos":"0","closePayload":"","willTopic":"","willQos":"0","willPayload":""}]
```

### Expected behavior

All events should send to the topic hasp/plate3/%topic%.

### Screenshots or video

smcgann99, 2024-04-07:
I wonder if this is related to https://github.com/HASwitchPlate/openHASP/issues/697 ?

stefan242, 2024-04-07:
I think so!
> MQTT updates send to the plate3a display are working though.

This seems to be important, since one direction is working!

stefan242, 2024-04-07:
And since I have another Openhasp based Lanbon L8 plate, which has another size and other display informations, I cannot not use hasp/plates/... either. (see the other issue #697 regarding group topics)

stefan242, 2024-04-08:
Sorry for asking so dumb, but I cannot find a new (compiled) firmware to test the fix with my GS-T3E. :-(

fvanroie, 2024-04-08:
You can download the latest firmware artifacts at the bottom of [this build](https://github.com/HASwitchPlate/openHASP/actions/runs/8592205266).

stefan242, 2024-04-08:
been there, done that, updated the firmware and voila, it works!
Hostname is "plate3a", Node Topic is "hasp/plate3/%topic%" and MQTT messages from the plate to Node-RED arrive as  
{"topic":"hasp/plate3/state/p2b20","payload":{"event":"up","val":27},"qos":0,"retain":false,"_topic":"hasp/plate3/state/p2b20","_msgid":"a548aa74c391f9e3"}
(for example)
Thank you!

closed 2024-04-08 by stefan242