HASwitchPlate/openHASP #691: Sunton ESP-32-8048S070 Backlight dimming is not working

DukemFIVE, 2024-03-29:
### Perform all steps below and tick them with [x]
- [ x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x ] Update openHASP to the latest version
- [x ] Reproduce the issue and describe all steps

### Describe the bug
If I try to dim the display using HomeAssistant (e.g. to 50% brightness) then the display goes completely black.
I have to set the brightness to over 85%, then the display will turn on again, but at 100%. If I choose a brightness below 85% then it becomes completely dark. The threshold value of 85% is not always consistent, sometimes it is 81 or 83.
I also tried to lower the startup brightness in the config, I have the same problem there.
I use the V07RC11 with the Sunton ESP-32-8048S070 (Touch)

### To Reproduce
Set the backlight to lower then 80%

### Expected behavior
The backlight dims to the set percentage

### Screenshots or video
x

WarC0zes, 2024-03-29:
Hi,
I've no problem with firmware 0.7.0-rc11 677e8cc and openHASP integration v0.7.2.

openHASP
--
Version | 0.7.0-rc11 677e8cc
Build DateTime | Feb 26 2024 21:04:04 UTC
Environment | sunton-8048s070c_16MB
Uptime | 8d 3h 10m 11s

[image]

DukemFIVE, 2024-03-30:
Hi,

I have the 0.7.2 installed (reinstalled it already)
[image]

[image]
[image]

How can I check if there is a hardware defect?

WarC0zes, 2024-03-30:
You have the HA core up to date ?
v2024.3.3 ?

DukemFIVE, 2024-04-01:
Yes v2024.3.3

marsman7, 2024-04-06:
which gpio pin is used to control the backlight on the sunton?

smcgann99, 2024-04-06:
BACKLIGHT_FREQUENCY needs to be around 100 for that screen

fvanroie, 2024-04-06:
Please test the l[atest Sunton builds](https://github.com/HASwitchPlate/openHASP/actions/runs/8584608794) from the Action tab.

smcgann99, 2024-04-07:
Checked and working OK on my sunton-8048s070c_16MB

DukemFIVE, 2024-04-07:
Works great ! Thank you !!

closed 2024-04-07 by DukemFIVE