HASwitchPlate/openHASP #697: Custom Group Topic name does not work

MrEcosse, 2024-04-05:
### Perform all steps below and tick them with [x]
- [x ] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x ] Update openHASP to the latest version
- [ x] Reproduce the issue and describe all steps

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

marsman7, 2024-04-06:
I can't understand that.
I have copied the jsonl line but deleted "parentid". I can send an MQTT group message to the label without any problems.

Please show more of your "pages.jsonl" and the complete text at Group Topic.

MrEcosse, 2024-04-06:
If you send a message with the default group name of pages it all works OK.  It is when you change the name of the group it no longer works.

marsman7, 2024-04-06:
it remains "hasp/plates/command/p1b1.text" despite group topic "hasp/mygroup/%topic%"
I show later in source code.

marsman7, 2024-04-07:
I have found the place where the problem lies but I have no explanation for the error.

In "src/mqtt/hasp_mqtt_esp.cpp" i have added 2 LOGs
```
String mqttGetTopic(Preferences preferences, String subtopic, String key, String value, bool add_slash)
{
    String topic = preferences.getString(key.c_str(), value);

    LOG_INFO(TAG_MQTT,"mqttGetTopic 1 key[%s] topic[%s] value[%s]", key.c_str(), topic.c_str(), value.c_str());
    LOG_INFO(TAG_MQTT, "mqttGetTopic 2 key[%s] topic[%s]", FP_CONFIG_GROUP_TOPIC, preferences.getString(FP_CONFIG_GROUP_TOPIC, "notfound").c_str());

    topic.replace(F("%hostname%"), haspDevice.get_hostname());
    topic.replace(F("%hwid%"), haspDevice.get_hardware_id());
    topic.replace(F("%topic%"), subtopic);
    topic.replace(F("%prefix%"), MQTT_PREFIX);

    if(add_slash && !topic.endsWith("/")) {
        topic += "/";
    }
    return topic;
}
```

This function is called twice in "void mqttStart()". The output of the second log line has exactly the same parameters in both calls but the value of Topic is different. Why?

```
MQTT: mqttGetTopic 1 key[node_t] topic[hasp/foo/%topic%] value[hasp/%hostname%/%topic%]
MQTT: mqttGetTopic 2 key[group_t] topic[hasp/mygroup/%topic%]
MQTT: mqttGetTopic 1 key[group_t] topic[hasp/plates/%topic%] value[hasp/plates/%topic%]
MQTT: mqttGetTopic 2 key[group_t] topic[notfound]
```

"preferences.getString" returns different results.

marsman7, 2024-04-07:
On the second call of "mqttGetTopic" "preferences.getString" returns the error INVALID_HANDLE

stefan242, 2024-04-07:
Here the backlink to my similar issue: [#686](https://github.com/HASwitchPlate/openHASP/issues/686)

marsman7, 2024-04-07:
The problem ist multible call of "mqttGetTopic" and in it the call of "preferences.getString".

In first call all ok, in all further calls no longer.

please review my fix.

MrEcosse, 2024-04-08:
Thanks all for looking into and fixing this

smcgann99, 2024-04-13:
@MrEcosse Is this working for you now ?
If so you can close the issue ?

marsman7, 2024-04-13:
For your information: The fix from above is a workaround. I have continued to search for the actual cause.

The actual problem is the parameter transfer from "preferences" to the function mqttGetTopic. The function works with a copy of "preferences". This can be fixed if only the pointer to "preferences" is passed to the function "mqttGetTopic".

`String mqttGetTopic(Preferences *preferences, String subtopic, String key, String value, bool add_slash)`

Only add the asterisk and call the function with 

`mqttNodeCommandTopic = mqttGetTopic(&preferences, subtopic, FP_CONFIG_NODE_TOPIC, MQTT_DEFAULT_NODE_TOPIC, false);`

Only add ampersand. This means that you are not working with a copy of "preferences" but with the same instance. Working with the copy only works with the first call of "mqttGetTopic" from the second call onwards.

closed 2024-04-23 by MrEcosse