HASwitchPlate/openHASP #704: parentid dont work

matsekberg, 2024-04-08:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug
I try to reduce the size of configuration by using parent objects but cant get it to work.
Only objects with explicit properties displays correct.

### To Reproduce

This is the config:

```
{"page":4,"id":99,"obj":"btn","x":0,"y":0,"h":0,"w":0,"toggle":1,"bg_color90":"#880000","bg_color91":"#005500","bg_color92":"#00FF00","bg_color93":"#FF0000","text_font":24, "radius30":7}
{"id":5,"obj":"btn","x":30,"y":260,"h":64,"w":120,"text":"KYLBOX","parentid":99}
{"id":6,"obj":"btn","x":170,"y":260,"h":64,"w":120,"text":"V.PUMP","parentid":99}
{"id":7,"obj":"btn","x":30,"y":340,"h":64,"w":120,"text":"STEG","parentid":99}
{"id":8,"obj":"btn","x":170,"y":340,"h":64,"w":120,"text":"EXTRA","toggle":1,"bg_color90":"#880000","bg_color91":"#005500","bg_color92":"#00FF00","bg_color93":"#FF0000","text_font":24, "radius30":7}
```
id 99 is supposed to hold stylings and other common properties.
id 5, 6 and 7 has a parentid pointing to the 99 object with common props.
id 8 uses explicit same properties.

### Expected behavior
The expected behaviour is four buttons with equal styling and size. 

### Screenshots or video

The image show that only id 8 (with explicit properties) s displayed.
[image]

This is the expected design:
[image]

marsman7, 2024-04-09:
Your parent object has a size of zero (w=0 and h=0). When you change your object to :

`{"page":4,"id":99,"obj":"btn","x":0,"y":0,"h"480,"w":320,"toggle":1,"bg_color90":"#880000","bg_color91":"#005500","bg_color92":"#00FF00","bg_color93":"#FF0000","text_font":24, "radius30":7}`

All buttons are visible. But I don't think that's what you want.

matsekberg, 2024-04-09:
My understanding from the documentation is that if the child has the same property as the parent, then the child will override.
X, y, h and w should be taken from the children.

I could try make the leftmost button a parent and the other children.

fvanroie, 2024-04-09:
**Some** properties (typically that are related to **text and opacity**) can be inherited, .i.e. not *all* properties are inheritable.
You need to set `toggle`, `bg_color90`,`bg_color91`,`bg_color92`,`bg_color93`,`radius30` on each button individually.
There is no way to group styling properties in the way you are trying to achieve...

You can create a parent `obj` object and set the `text_font`there once. All `x` and `y` coordinates will be relative to the parent.

matsekberg, 2024-04-09:
Ok so me trying to use parent as a template won’t work that well.
And you see no reason to implement it that way? 
In order to simplify the config file.

fvanroie, 2024-04-09:
Maybe in due time... but you can implement this too if you really needed it.
I have made the whole project open source so users can benefit and contribute alike...

closed 2024-04-09 by fvanroie