HASwitchPlate/openHASP #706: Sometimes display wakeup if get auto discovery.

vampywiz17, 2024-04-11:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

### To Reproduce

Install latest beta firmware 7.0 rc11 (Guition ESP32-S3-4848S040)

### Expected behavior

The display can not wake up

### Screenshots or video

Get this code on console, when the display are wake up:

```
8:39.881][98292/109668 10][25324/26112  4] MQTT PUB: discovery => {"node":"nappali_uj","mdl":"1-S3 4848S040","mf":"openHASP","hwid":"806599c88d40","pages":12,"sw":"0.7.0-rc11","node_t":"hasp/nappali_uj/","uri":"http://192.168.31.162","input":{},"power":[],"light":[],"dim":[]}
```

vampywiz17, 2024-04-11:
hmm it does not depend on discovery...

I get this, if the display are wakeing up.

```
#[22:50:34.205][98292/109668 10][25324/26056  3] HASP: First touch Disabled
#[22:50:34.219][94196/106476 11][25324/26056  3] MQTT PUB: backlight => {"state":"on","brightness":255}
#[22:50:34.231][90100/106488 15][25324/26056  3] MQTT PUB: idle => off
#[22:50:34.244][90100/104904 14][25324/26056  3] MSGR: File not found: L:/idle_off.cmd
#[22:50:34.272][98292/109556 10][25324/26056  3] MQTT RCV:  = backlight {"state": "on", "brightness": 255}
#[22:50:34.285][94196/107960 12][25324/26056  3] MSGR: backlight={"state": "on", "brightness": 255}
#[22:50:34.300][94196/107960 12][25324/26056  3] MQTT PUB: backlight => {"state":"on","brightness":255}
```

vampywiz17, 2024-04-11:
Strange thing... it only happen, if i power it my display via 230V... if i use USB type C port, the problem not happen more than one hour...

smcgann99, 2024-04-14:
I have that screen running on 240V and do not see this problem, how often does it wake up ?

vampywiz17, 2024-04-14:
@smcgann99 

I did more test and seems maybe not the 230V the problem... 

sometimes happen about 5-10 min, sometimes not happen hours... I dont understand... Seems happen more time, if for example the sun is shine directly to the screen... but it only just a idea...

smcgann99, 2024-04-14:
This doesn't really sound like a bug to me. 
Do you have the complete log prior to the screen idle off, perhaps something is being sent to the screen ?

vampywiz17, 2024-04-21:
@smcgann99 

Hello sorry the late answer.

Here:

```
#[20:08:00.000][102388/111876  8][25364/26012  3] MQTT RCV: p0b3.text = 20:08
#[20:08:59.990][102388/111876  8][25364/26012  3] MQTT RCV: p0b3.text = 20:09
#[20:09:38.479][98292/110356 10][25364/26012  3] MQTT PUB: statusupdate => {"node":"nappali_uj","idle":"long","version":"0.7.0-rc12","uptime":37576,"ssid":"iot_network","rssi":-63,"ip":"192.168.31.162","mac":"80:65:99:C8:8D:40","heapFree":111940,"heapFrag":8,"core":"4.4.6","canUpdate":"false","page":1,"numPages":12,"tftDriver":"Other","tftWidth":480,"tftHeight":480}
#[20:09:38.498][102388/111940  8][25364/26012  3] MQTT PUB: sensors => {"time":"2024-04-21T20:09:38","uptimeSec":37576,"uptime":"0T10:26:16"}
#[20:09:47.082][102388/111940  8][25364/26012  3] HASP: First touch Disabled
#[20:09:47.096][98292/108772  9][25364/26012  3] MQTT PUB: backlight => {"state":"on","brightness":255}
#[20:09:47.108][94196/108768 13][25364/26012  3] MQTT PUB: idle => off
#[20:09:47.121][94196/107184 12][25364/26012  3] MSGR: File not found: L:/idle_off.cmd
#[20:09:47.150][102388/111816  8][25364/26012  3] MQTT RCV:  = backlight {"state": "on", "brightness": 255}
#[20:09:47.163][98292/110228 10][25364/26012  3] MSGR: backlight={"state": "on", "brightness": 255}
#[20:09:47.178][98292/110228 10][25364/26012  3] MQTT PUB: backlight => {"state":"on","brightness":255}
#[20:09:59.989][102388/111876  8][25364/26012  3] MQTT RCV: p0b3.text = 20:10
#[20:10:04.515][98292/110356 10][25364/26012  3] MQTT PUB: discovery => {"node":"nappali_uj","mdl":"1-S3 4848S040","mf":"openHASP","hwid":"806599c88d40","pages":12,"sw":"0.7.0-rc12","node_t":"hasp/nappali_uj/","uri":"http://192.168.31.162","input":{},"power":[],"light":[40],"dim":[]}
#
```

smcgann99, 2024-04-26:
Well it doesn't look like anything is waking it via command, so it may just be a faulty screen / local interference ?

I haven't seen this issue with my screen.

fvanroie, 2024-05-13:
This all sounds to me like electromagnetic interference on the device/microcontroller. Make sure all the needed signals have an appropriate pull-up/pull-down resistor. Fixing an issue that only occurs randomly is nearly impossible without knowing the root cause...

fvanroie, 2024-05-28:
Is this problem resolved now?

username-AL, 2024-06-03:
I have the same problem. 
And also with the display 1-S3 4848S040. Sometimes it wakes up randomly and (it seems to me) it clicking in a random place.
[image]

vampywiz17, 2024-06-03:
> Is this problem resolved now?

not solved, unfortunately...

smcgann99, 2024-06-04:
This sounds more like a hardware issue than a bug. e.g. the screen is registering a touch when there is none.

fvanroie, 2024-06-05:
More troubleshooting is needed to determine the root-cause.
- Does it happen when you detach the screen from the PSU. Power it with a stable 5V power-supply and see what happens.
- Does this also happen with other firmware?
- Do you have multiple screens? Do they all have the same symptoms?

vampywiz17, 2024-06-11:
> More troubleshooting is needed to determine the root-cause.
> 
> * Does it happen when you detach the screen from the PSU? Power it with a stable 5V power-supply and see what happens.
> * Does this also happen with other firmware?
> * Do you have multiple screens? Do they all have the same symptoms?

@fvanroie 

- Does it happen when you detach the screen from the PSU? Power it with a stable 5V power-supply and see what happens.
- Try with USB port, it also happen

- Does this also happen with other firmware?
- I try more RC firmware, it happen all of them

- Do you have multiple screens? Do they all have the same symptoms?
- Only one screen, but others are also affect it.

fvanroie, 2024-06-11:
Thanks for your feedback. It really sounds like a hardware issue to me.
Maybe you can try other software like Tasmota, ESPhome, ... and see if that also have the problem.

Like stated before, without a way to replicate the *random* problem, it will be very hard to fix it.

FreeBear-nc, 2024-06-15:
It _might_ be that the touch driver chip is reacting to electrical noise and generating an interrupt. Been digging in to the GT911 programming notes, and I see a couple of config registers that can be set to filter out noise (and adjust the touch sensitivity). There are a couple of functions within the Arduino Goodix driver for the GT911 that read the config registers. Whilst there is a function call to set the resolution, there isn't one to adjust sensitivity.
It would only take a few lines of code to craft a suitable function, but the question is where within openHASP to put it (and when to call). Perhaps add something to the Configuration->Display screen in the web interface ?

fvanroie, 2024-06-16:
The goodix repo is a fork which already has some openHASP tweaks.
So it is fine to add the settings function to the goodix driver.
Then, openHASP can simple call that function when initializing the driver.
At first, a simple test is needed to see if it actually works before thinking about UI changes.

vampywiz17, 2024-06-16:
If you do this I glad to test it!

FreeBear-nc, 2024-06-17:
Hacked a few lines of code to read the config registers inside the GT911 chip. In the process, learned a little... My trivial little hack causes openHASP to crash & burn with a panic. Not had time to work out why.

In the meantime, I do have a standalone utility that could be used to tweak the relevant register in the GT911 chip.

smcgann99, 2024-06-28:
OK so it seems that my screen has developed this problem now ;-(
Recently noticed that devices were switching on randomly, and traced it to the screen registering a button press when there was none.

FreeBear-nc, 2024-06-29:
Tried adding a short bit of code to openHASP to modify the relevant register in the GT911 chip during setup. Unfortunately, this addition triggered a panic and infinite reboots. As life is short, and using the JTAG interface to debug is impossible, I've put together a standalone "project" - https://github.com/FreeBear-nc/HA-openHASP/tree/main/GT911
Compile, upload, and then upload the openHASP firmware.

To adjust the noise sensitivity, edit the NR_LEVEL flag in the platformio.ini and recompile/upload - I would suggest 6 or 7 to start with.

fvanroie, 2024-08-15:
Does this register fix the ghost touches?

vampywiz17, 2024-08-15:
@fvanroie 

I can not able to test it...

smcgann99, 2024-08-15:
If @FreeBear-nc could provide a (tested) pre built binary or 2 with different values, I'm willing to try it (not wishing to brick my screen) ;-)

In my case I seem to get a bunch of false touches, a power cycle seems to stop them. It can then go for days without problem, so will not be easy to be sure its helping for some time.

FreeBear-nc, 2024-08-15:
Had a spell of phantom touches on my Guition display. Didn't get round to uploading my code, and the events haven't reoccurred. Intermittent ghost events are a pig to diagnose and to say with any certainty that they have been fixed.

@smcgann99 Do you not have platformio set up on your computer ?

smcgann99, 2024-08-15:
> Had a spell of phantom touches on my Guition display. Didn't get round to uploading my code, and the events haven't reoccurred. Intermittent ghost events are a pig to diagnose and to say with any certainty that they have been fixed.
> 
> @smcgann99 Do you not have platformio set up on your computer ?

Sounds like you are seeing similar issues to me then, I had it on my desk for weeks before I noticed the problem. 
I suspect that it may be present on all Guition screens, but others may not have noticed. 
It would be great to get it fixed as these are really nice screens for the price.

I don't have it setup, I have managed to fumble my way to using gitpod in the past. But if you can do a couple of builds with different values then you can test one value and I can test another. I can leave it running for a week or so and see if I get any events. Would also need to see if it effects usability of the screen.

FreeBear-nc, 2024-08-15:
@smcgann99  Catch me on discord over the weekend to discuss options.

FreeBear-nc, 2024-08-18:
Pushed three firmware images to my repository. You'll need to use esptool to upload them to your board. Once loaded, do a reset and let it run. Then upload the openHASP firmware.
If the nr-7 or nr-10 builds don't work, the nr-5 build should put everything back to stock.

smcgann99, 2024-08-18:
> Pushed three firmware images to my repository. You'll need to use esptool to upload them to your board. Once loaded, do a reset and let it run. Then upload the openHASP firmware. If the nr-7 or nr-10 builds don't work, the nr-5 build should put everything back to stock.

Have you installed one of the values to your screen, if so which ?

FreeBear-nc, 2024-08-18:
> Have you installed one of the values to your screen, if so which ?

Tried 6 and 7. But I've only had one evening of phantom touch events, and can't say for sure if it made any difference.

smcgann99, 2024-08-18:
> But I've only had one evening of phantom touch events

So with 6 and 7 you are still seeing phantom touch events ?

FreeBear-nc, 2024-08-18:
> > But I've only had one evening of phantom touch events
> 
> So with 6 and 7 you are still seeing phantom touch events ?
Not currently experiencing any phantom touch events even after reverting to 5 (the default setting).

smcgann99, 2024-08-18:
Is there any notable effect on touch usability at higher values ?, if not then why not just leave at highest usable setting ?

FreeBear-nc, 2024-08-18:
> Is there any notable effect on touch usability at higher values ?, if not then why not just leave at highest usable setting ?

Didn't notice any change in touch sensitivity at higher values - the max is 15 which I haven't tried (yet).

smcgann99, 2024-08-19:
not sure if you had seen this ?
[https://hackaday.io/project/177256-put-a-raspberry-pi-cm4-into-an-original-ipad/log/212604-attempting-to-tune-the-touchscreen-controller](url)

smcgann99, 2024-08-20:
> You'll need to use esptool to upload them to your board

Can detail exactly how to upload your firmware, and then revert back to openhasp please ?
Sorry for being overly cautious but your warnings are pretty stark ;-)

FreeBear-nc, 2024-08-20:
> > You'll need to use esptool to upload them to your board
> 
> Can detail exactly how to upload your firmware, and then revert back to openhasp please ? Sorry for being overly cautious but your warnings are pretty stark ;-)

Please see: https://www.openhasp.com/0.7.0/firmware/esp32/
Upload the nr-X firmware files. Reboot.
Then repeat with the openHASP firmware.

Note: I use platformio for compiling, uploading, and monitoring the output. A very useful suite of tools and worth installing.

smcgann99, 2024-08-20:
So we cannot use ota via gui ? If not then does it need to also be erased first

FreeBear-nc, 2024-08-20:
The firmware isn't built with OTA in mind.

TNTLarsn, 2024-08-22:
Hello and sorry for hoppin' in. Just wanted to thank @FreeBear-nc for developing a solution for this issue.
I'm also facing some unintended wake-ups from time to time.
I just tried the nr-X firmware [as described ](https://github.com/HASwitchPlate/openHASP/issues/706#issuecomment-2297745756). Will see and and report back if this worked.

Just a little note on this: A little confirmation when flashing the nr-X firmware would be nice. The display stood black all the time and I'm not quite sure if the change on the "NR_LEVEL flag in the GT911 chip" took place.
Please also don't give up on implementing this into the openhasp-firmware. An option to change the NR-Level directly in the UI would be great.

FreeBear-nc, 2024-08-22:
> Just a little note on this: A little confirmation when flashing the nr-X firmware would be nice. The display stood black all the time and I'm not quite sure if the change on the "NR_LEVEL flag in the GT911 chip" took place. 

If you use the serial monitor (`pio device monitor`), the firmware dumps a bit of info to the serial port. Implementing printing to the display would use up a lot of time (something that I'm short on).

More information on pio here: https://docs.platformio.org/en/latest/core/userguide/device/cmd_monitor.html

fvanroie, 2024-08-22:
If this is the confirmed root cause for the Guition ESP32-S3-4848S040 with working *fix*, then it would indeed make sense to add it to the firmware.

smcgann99, 2024-08-23:
@fvanroie It's too early to say at the moment I think. I have the nr-7 set at the moment, but would need at least a week (based on my observations of how random the events are) to have some confidence that it helps. 

If other can share their findings here it would help.

smcgann99, 2024-08-29:
@FreeBear-nc I'm not sure if anyone else has been testing this patch ?
I installed the "7" setting 7 days ago, and haven't had any touch events so far.....
Probably too early to say in my case as ghost events were rare, but looks promising.

TNTLarsn, 2024-08-29:
I installed the "10" setting (just to be safe I guess)
Guess when!
[image]
Looking good sofar

vampywiz17, 2024-08-29:
If some time today, i also try to install it :)

vampywiz17, 2024-08-29:
I installed it. I hope good :)

```
.\esptool.exe --chip esp32-S3 --port COM3 write_flash -z 0x0 E:\nr-10\firmware.bin

.\esptool.exe --chip esp32-S3 --port COM3 write_flash -z 0x0 E:\esp32-s3-4848s040_full_16MB_v0.7.0-rc12_c27c8b6.bin
```

vampywiz17, 2024-08-30:
At these time, no random wake up. I use Nr-10 FW.

vampywiz17, 2024-08-30:
perfect :)

[image]

smcgann99, 2024-08-31:
> OK. Need to work out how to integrate into openHASP without the panic attacks.

Drink less coffee ? Nice find by the way ;-)

FreeBear-nc, 2024-09-01:
Has the potential to brick the GT911 chip if other registers are also modified (or the read gets corrupted).
So far, haven't managed to do any damage to my screen.

fvanroie, 2024-09-01:
You could test if the checksum of the read matches the expected checksum to make sure the integrity of the data is intact.
Also, id the noise level is already at the desired level, you can skip the write and reduce the chance of a failed write.

FreeBear-nc, 2024-09-01:
Increased error detection so that the function aborts before doing anything fatal.
If old and new values match, no point in writing data, so gracefully exit as per your suggestion.

smcgann99, 2024-09-03:
Just to let you know after testing on USB power for over a week with no issues, I swapped to Mains power and have had around a dozen ghost events today (using Nr-7 FW).
Can others confirm how they are powering the devices, and what level they used please ?

FreeBear-nc, 2024-09-04:
Pushed a couple of precompiled firmware builds to https://github.com/FreeBear-nc/HA-openHASP/tree/main/GT911/firmware
You have the option of an OTA build, and a full one - Both are built on top of openHASP: https://github.com/FreeBear-nc/openHASP/tree/noise_reduction

Take one for a test spin, and you can play with values between 0 and 15.

smcgann99, 2024-09-04:
Thanks, I will leave it running on 5 for comparison. Then work up from there

vampywiz17, 2024-09-04:
I use it with  Nr-10 FW, it work without issue on USB power. Today morning i switch it to Main power and the random wake up are come back... :(

[image]

vampywiz17, 2024-09-04:
I installed new firmware and set it up to value 5. But it not help at all... it need to increase or decrease the number to find out the correct setting?

smcgann99, 2024-09-04:
I have mine set to 8 at the moment, if I get more issues I will increase it 1 at a time to see if it stops

FreeBear-nc, 2024-09-04:
> I installed new firmware and set it up to value 5. But it not help at all... it need to increase or decrease the number to find out the correct setting?

Go to Configuration->Display Settings and increase Noise reduction to 12 or even 15.
You'll need to reboot the plate for changes to take effect.

[41 more replies]