HASwitchPlate/openHASP #710: opneHASP support for Pyportal

tome88, 2024-04-19:
Hi there, wanting to know if there is an easy way (or might be a way in the future) to have openHASP available for pyportal/pyportal Titano systems (they run on an ATSAMD51J20 and don't seem to be compatible at the moment)

tome88, 2024-04-20:
> openHASP does _not_ support the ATSAMD51J20 microcontroller. There are a few dozens of ESP32 devices that can run openHASP to choose from. If your really wanted to port the source code to a new MCU, I guess it _could_ eventually work... the question is if it's worth the time and effort, since most devices only cost a coupla bucks nowadays.

thanks for the quick reply @fvanroie

fvanroie, 2024-04-23:
Is this something you'll be pursuing and actively working towards? Or can this issue be closed?

closed 2024-04-29 by fvanroie