HASwitchPlate/openHASP #712: Wrong colors in screenshot

matsekberg, 2024-04-23:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

Screenshot renders wrong colors. See the top bar.

### To Reproduce

#### JSONL

```
{"page":0,"id":0,"comment":"---------- SALON ---------- 320x480", "swipe":1,"bg_color":"black", "bg_grad_color":"black"}
{"id":1,"obj":"label","click":false,"x":0,"y":0,"w":96,"h":40,"text":"-23.4 °","text_color":"White","bg_color":"#007070","bg_opa":255,"align":"center","radius":0,"border_side":0,"text_font":30, "pad_top":3}
{"id":3,"obj":"label","click":false,"x":224,"y":0,"w":96,"h":40,"text":"22:55","text_color":"White","bg_color":"#007070","bg_opa":255,"align":"center","radius":0,"border_side":0,"text_font":30, "pad_top":3}

{"id":4,"obj":"btn","action":"p1","x":0,"y":440,"w":79,"h":40,"text":"\uE2DC","bg_color00":"#007070","bg_color02":"Green","radius":0,"border_side":0,"text_font":32}
{"id":5,"obj":"btn","action":"p2","x":80,"y":440,"w":79,"h":40,"text":"\uE50F","bg_color00":"#007070","bg_color02":"Green","radius":0,"border_side":0,"text_font":32}
{"id":6,"obj":"btn","action":"p3","x":160,"y":440,"w":79,"h":40,"text":"\uF2A2","bg_color00":"#007070","bg_color02":"Green","radius":0,"border_side":0,"text_font":32}
{"id":7,"obj":"btn","action":"next","x":240,"y":440,"w":80,"h":40,"text":"\uE054","bg_color00":"#007070","bg_color02":"Green","radius":0,"border_side":0,"text_font":32}

{"page":1,"comment":"########## Page 1 ÖVERSIKT ########## 320x480","prev":4,"next":2,"bg_color":"black","bg_opa":255, "bg_grad_color":"black"}
{"id":254,"obj":"label","click":false,"x":97,"y":0,"w":126,"h":40,"text":"Översikt","text_color":"White","bg_color":"#007070", "align":"center", "bg_opa":255,"align":"center","text_font":30, "pad_top":3}

{"id":200,"obj":"label","click":false,"x":10,"y":46,"w":40,"h":30,"text":"\uE50F", "text_color":"red","bg_color":"black","text_font":24}
{"id":201,"obj":"label","click":false,"x":35,"y":46,"w":40,"h":30,"text":"\uF40B", "text_color":"red","bg_color":"black","text_font":24}
{"id":202,"obj":"label","click":false,"x":65,"y":46,"w":40,"h":30,"text":"\uE5A9", "text_color":"green","bg_color":"black","text_font":24}
{"id":203,"obj":"label","click":false,"x":100,"y":46,"w":40,"h":30,"text":"\uE1FA", "text_color":"green","bg_color":"black","text_font":24}
{"id":204,"obj":"label","click":false,"x":135,"y":46,"w":40,"h":30,"text":"\uE58F", "text_color":"red", "bg_color":"black","text_font":24}
{"id":205,"obj":"label","click":false,"x":160,"y":46,"w":130,"h":30,"text":"Uppstart...", "align":"right", "text_color":"grey", "bg_color":"black","text_font":24}

{"id":1,"obj":"linemeter","click":false,"x":14,"y":95,"w":138,"h":138,"value_str":"70 %","val":70,
"scale_end_color":"#444444","scale_grad_color":"#ffff00","line_color":"#ffff00",
"scale_width":7,"scale_end_line_width":4,"line_width":4,
"line_count":20,"value_font":30,"bg_color":"red","bg_opa":0}

{"id":2,"obj":"linemeter","click":false,"x":174,"y":95,"w":138,"h":138,"value_str":"70 %","val":70,
"scale_end_color":"#444444","scale_grad_color":"#ffff00","line_color":"#ffff00",
"scale_width":7,"scale_end_line_width":4,"line_width":4,
"line_count":20,"value_font":30,"bg_color":"red","bg_opa":0}

{"id":100,"obj":"label","click":false,"x":14,"y":190,"h":30,"w":138, "text_font":18,"text":"BATTERI","align":"center"}
{"id":101,"obj":"label","click":false,"x":174,"y":190,"h":30,"w":138, "text_font":18,"text":"VATTEN","align":"center"}
{"id":103,"obj":"label","click":false,"x":20,"y":260,"h":34,"w":160,"text":"\uE50FKylbox","text_font":24,"bg_color":"red","bg_opa":0}
{"id":3,"obj":"label","click":false,"x":180,"y":260,"h":34,"w":120,"text":"-__._ °","text_font":24,"align":"right","text_color":"ffff00","bg_color":"red","bg_opa":0}
{"id":104,"obj":"label","click":false,"x":20,"y":320,"h":34,"w":160,"text":"\uE50FGarage","text_font":24}
{"id":4,"obj":"label","click":false,"x":180,"y":320,"h":34,"w":120,"text":"-__._ °","text_font":24,"align":"right","value_color":"ffff00"}
{"id":105,"obj":"label","click":false,"x":20,"y":290,"h":34,"w":120,"text":"\uE50FSalong","text_font":24}
{"id":5,"obj":"label","click":false,"x":140,"y":290,"h":34,"w":160,"text":"-__._ ° .. -__._ °","text_font":24,"align":"right"}
{"id":106,"obj":"label","click":false,"x":20,"y":350,"h":34,"w":160,"text":"\uE50FKylskåp","text_font":24,"bg_color":"red","bg_opa":0}
{"id":6,"obj":"label","click":false,"x":180,"y":350,"h":34,"w":120,"text":"-__._ °","text_font":24,"align":"right","bg_color":"red","bg_opa":0}
{"id":107,"obj":"label","click":false,"x":20,"y":380,"h":34,"w":160,"text":"\uE599Solkraft","text_font":24}
{"id":7,"obj":"label","click":false,"x":180,"y":380,"h":34,"w":120,"text":"___ W","text_font":24,"align":"right"}
{"id":8,"obj":"label","click":false,"x":80,"y":380,"h":34,"w":120,"text":"(___)","text_font":24,"align":"right"}

```

#### Information page

openHASP
Version	0.7.0-rc10 fb2bebf
Build DateTime	Feb 2 2024 15:41:18 UTC
Environment	esp32-3248s035c_4MB
Uptime	48m 46s
Idle	long
Active Page	1
Device Memory
Free Heap	52.47 KiB
Free Block	45.98 KiB
Fragmentation	12%
LVGL Memory
Total	48.00 KiB
Free	23.19 KiB
Fragmentation	5%
Model	ESP32-D0WD-V3 rev3
Frequency	240MHz
Core Version	4.4.6
Reset Reason	CPU0: SW_CPU_RESET / CPU1: SW_CPU_RESET
Flash Size	4.00 MiB
Program Size Used	1.66 MiB
Program Size Free	1.68 MiB
Filesystem Size	560.00 KiB
Filesystem Used	32.00 KiB
Filesystem Free	528.00 KiB

### Expected behavior

Same colors.

### Screenshots or video

[image]

fvanroie, 2024-04-23:
The screenshot color looks to be the correct one: https://www.colorhexa.com/007070

marsman7, 2024-04-23:
I have uploaded your jsonl on my crowpanel. The top bar is dark cyan not blue. I think your display shows a wrong color.

matsekberg, 2024-04-23:
I think yo are correct, very strange.
I have tried to get close to the screenshot-green, see the secons image. This has hex value #103C1F

[image]

matsekberg, 2024-04-24:
Is there any color mapping taking plave in the LVGL-library that mismatch the Sunton display?

fvanroie, 2024-04-25:
LVGL doesn't map any colors, that is the display/driver's job.

smcgann99, 2024-04-26:
I have the 7" 480X800 sunton and find the colour reproduction is very poor, next to other screens I have, everything looks washed out with a strong blue hue.

fvanroie, 2024-04-29:
LCD displays usually have gamma correction settings, and other color parameters that can be set by the driver.
I haven't looked into it, but it should be possible to tweak the settings to match the wanted colors.
Check the documentation of your display to see what settings are available and how to implement them.

matsekberg, 2024-04-29:
Where would such code be?

marsman7, 2024-04-29:
https://www.openhasp.com/0.6.3/compiling/custom/

fvanroie, 2024-05-13:
Check the datasheet of the components, usually provided by the vendor.

closed 2024-06-15 by fvanroie