HASwitchPlate/openHASP #716: mqtt control

fifa255, 2024-04-27:
Does openhasp support publishing to mqtt topics?  

I would like to control some mqtt devices directly from the display rather than having to be dependent on homeassistant. Is this a possibility with openhasp currently?

marsman7, 2024-04-27:
openhasp now already sends mqtt commands when a control is activated.

https://www.openhasp.com/0.7.0/commands/mqtt/

fvanroie, 2024-04-27:
openHASP was concieved to be controlled by an Home Automation system. There isn't any intelligence or coding that happens on the device. So, you can add your [custom code](https://www.openhasp.com/0.7.0/firmware/compiling/custom/) to the firmware to make it do what you want...

A controller would need to know the topics but also the format of the messages to send and respond too. This is very specific for each device, so you need custom event handlers for that.

closed 2024-04-29 by fvanroie