HASwitchPlate/openHASP #717: Cannot style LED color

matsekberg, 2024-04-29:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug
I try to change the color of the LED object 

### To Reproduce
```json
{"page":1,"id":0,"comment":"---------- SALON ---------- 320x480", "swipe":1,"bg_color":"black", "bg_grad_color":"black"}

{"id":11,"obj":"led","x":10,"y":60,"w":20,"h":20,"color":"#FF0000"}
{"id":12,"obj":"led","x":60,"y":60,"w":20,"h":20,"color":"#FF0000","val":200}
{"id":13,"obj":"led","x":110,"y":60,"w":20,"h":20,"color":"#FF0000","val":10}

{"id":21,"obj":"led","x":10,"y":100,"w":20,"h":20,"color":"#00FF00"}
{"id":22,"obj":"led","x":60,"y":100,"w":20,"h":20,"color":"#00FF00","val":200}
{"id":23,"obj":"led","x":110,"y":100,"w":20,"h":20,"color":"#00FF00","val":10}

{"id":21,"obj":"led","x":10,"y":140,"w":20,"h":20,"color":"#FFFF88"}
{"id":22,"obj":"led","x":60,"y":140,"w":20,"h":20,"color":"#FFFF88","val":200}
{"id":23,"obj":"led","x":110,"y":140,"w":20,"h":20,"color":"#FFFF88","val":10}
```

### Expected behavior

[image]

### Screenshots or video

[image]

sfromis, 2024-04-29:
Looks like you copied an [example from HASPmota](https://tasmota.github.io/docs/HASPmota/#led), without being clear about this. While this is using a `jsonl` format "heavily inspired" by openHASP, it is not a guide about how to do things in openHASP.

matsekberg, 2024-04-29:
I agree, but its an example, one of many i have tried.

marsman7, 2024-04-29:
Take a look at the IDs of the objects, there are duplicates.