HASwitchPlate/openHASP #722: Add support for SD card e.g. Sunton 7"

Martinson50, 2024-05-12:
I have a board with SD card so I hacked this repository locally to add the SDCard as a source of images.
I dont feel to much comfortable to pull a request but to send the changed code in 3 main files:
seems to work without issues when you add src object with Z:/image.png

/src/main.cpp:
#if HASP_USE_SDCARD > 0
    sdcardSetup();
#endif

/src/hasp_filesystem.cpp:
bool sdcardSetup(void)

/src/hasp/hasp_attribute.cpp
static hasp_attribute_type_t special_attribute_src(lv_obj_t* obj, const char* payload, char** text, bool update)
....
          } else if (payload == strstr_P(payload, PSTR("Z:"))) { 
               char tempsrcf[64] = "";
                strncpy(tempsrcf , payload + 2, sizeof(tempsrcf));
               int readf = 0;
               File file;
               file = HASP_SD_FS.open(tempsrcf, "r");
               .....

If anyone it's interested in this funcionality I can send the rest of the code or give more info.

Martinson50, 2024-05-12:
By the way audio and i2c sensors added in custom code as proof of concept able to reproduce MP3 SDCard files in this sunton-8048s070c_16MB

fvanroie, 2024-05-12:
This is very interesting indeed. Please share your code in a fork or PR. It is hard to follow when the code is incomplete.
Ideally, all files should be able to use the SD card, like fonts, jsonl, etc...

fvanroie, 2024-05-22:
I found your fork at https://github.com/Martinson50/openHASP

Make sure not to touch code outside of the scope, or it can not be merged (like QR_CODE).
In any case it is very interesting to see the progress.

fvanroie, 2024-05-23:
Please also join this thread about SD card support #741 and file handling.