HASwitchPlate/openHASP #726: undefined reference to `mqttNodeTopic'

Martinson50, 2024-05-16:
undefined reference to `mqttNodeTopic' when #define HASP_USE_HA 

Just uncomment and compile  then... 
/home/bambu/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/lolin-d32-pro_ili9341/src/mqtt/hasp_mqtt_ha.cpp.o:(.literal._Z22mqtt_ha_add_device_idsRN11ArduinoJson8V6215PD212JsonDocumentE+0x24): undefined reference to `mqttNodeTopic'

Martinson50, 2024-05-16:
Dirty patch in line 88
void mqtt_ha_add_device_ids(JsonDocument& doc)
{
    JsonObject device = doc.createNestedObject(FPSTR(FP_MQTT_HA_DEVICE));
    JsonArray ids     = device.createNestedArray(FPSTR(FP_MQTT_HA_IDENTIFIERS));
    ids.add(haspDevice.get_hostname());
    ids.add(haspDevice.get_hardware_id());

    device[F("sw")]                        = haspDevice.get_version();
    device[FPSTR(FP_MQTT_HA_NAME)]         = haspDevice.get_hostname();
    device[FPSTR(FP_MQTT_HA_MODEL)]        = F(PIOENV);
    device[FPSTR(FP_MQTT_HA_MANUFACTURER)] = F(D_MANUFACTURER);

    doc[F("~")] = "hasp/plate/"; <---- doc[F("~")] = mqttNodeTopic; ORIGINAL
}

fvanroie, 2024-05-16:
`HASP_USE_HA` is no longer supported.

closed 2024-05-23 by fvanroie

corbolais, 2024-08-18:
`include/user_config_override-template.h` still references this `#define HASP_USE_HA`:

```
grep -nH HASP_USE_HA include/user_config_override-template.h
include/user_config_override-template.h:151://#define HASP_USE_HA                                // Enable Home Assistant auto-discovery
```