HASwitchPlate/openHASP #728: Makerfabs ESP32-S3 WROOM1 3,5 TFT SPI is not supported

dersch81, 2024-05-18:
https://www.openhasp.com/0.7.0/hardware/makerfabs/esp32-s3-spi-tft-touch/

[image]

[image]

[image]

marsman7, 2024-05-21:
The same board works 100 percent for me. Check your settings.

marsman7, 2024-05-21:
does the file exist in your system and is it compiled with ?

user_setups/esp32s3/makerfabs-s3-tft35-spi.ini

dersch81, 2024-05-21:
I did not try to complie it by myself. I tried the web esp flasher with
provided precomplied binaries.

marsman7, 2024-05-21:
Ich habe den ESP-Flasher noch nie verwendet. Das selbst Kompilieren läuft ohne Fehler durch.

fvanroie, 2024-05-21:
I added the Makerfabs S3 to https://nightly.openhasp.com/ please try it there.

dersch81, 2024-05-22:
> I added the Makerfabs S3 to https://nightly.openhasp.com/ please try it there.

Thanks for adding. But it fails with the same error:

[image]

fvanroie, 2024-05-24:
Thanks, I must have overlooked something.. please try again now.

dersch81, 2024-05-28:
> Is this problem resolved now?

Yes, it is working. Many thanks

closed 2024-05-28 by fvanroie