HASwitchPlate/openHASP #731: Unable to build full version 7.0

haaxema, 2024-05-19:
Hi,

I am trying to loclay build V7.0RC12 using VSCode/PIO for my WT32SC01
It works fine, but I only get the OTA bin, not the Full version.
As I am trying to upgrade from 6.3 I need the Full bin to do a serial upload.

Any thoughts?

Martin

fvanroie, 2024-05-21:
You can download it from https://nightly.openhasp.com/

fvanroie, 2024-05-24:
The issue is most likely that you are using Windows and pip is out-of-data or the PySerial library is missing...

If so, you can fix it by running these two commands in order:
```cmd
C:\Users\<username>\.platformio\python3\python.exe -m ensurepip --upgrade
C:\Users\<username>\.platformio\python3\python.exe -m pip install Pyserial
```

Make sure to replace `<username>` with your actual username. That fixed the issue for me.

fvanroie, 2024-05-28:
Is this problem resolved now?

closed 2024-06-05 by fvanroie