HASwitchPlate/openHASP #743: sunton 2432S028R displays artifacts

DaddoDoc92, 2024-05-25:
Good evening, I used the compiled version "esp32-2432s028r_full_4MB_v0.7.0-rc9_c200edf" correctly on my device, now I have updated to the rc12 version but unfortunately I have display problems on the tft screen which were not present in the rc9 version.
I'm compiling my firmware to increase the lvgl memory but unfortunately with the rc12 version I can't do this due to screen problems. Can you help me?
[image]

This photo is instead with the rc9 version

[image]

DaddoDoc92, 2024-05-25:
I managed to solve the problem independently by reading here and there, my model has the ili9342 tft, while on the bin it is set ili9341, of an old version of the PCB, furthermore I increased the lvgl memory to 64kbit and set height 240 width 320 , thus solving the screen problem.
I now have to create a sensor to have the battery voltage and to have audio feedback as this PCB also has an audio amplifier DAC.

For now, thanks to everyone. I attach screenshots of what I changed.

If you need to do other tests I'm available.

[image]

[image]

[image]

fvanroie, 2024-05-28:
Please create a Pull Request to incorporate the changes into the project.

Saharel001, 2024-08-07:
Any news about this bug?

Saharel001, 2024-08-08:
I found the device esp32-2432s028r-ili9342_4MB in esp32-2432s028.ini an add this in platformio_override.ini to compile a vaild firmware for this device. Works for the Display but touch are mirrored