HASwitchPlate/openHASP #751: Can't select correct backlight pin in Guition ESP32-S3-4848S040 firmware

tdvtdv, 2024-06-07:
In the firmware esp32-s3-4848s040_full_16MB_v0.7.0-rc12_cd34d89
it is not possible to select the correct pin. 
The wrong pins are offered for selection, and the correct pin 38 cannot be selected
[image]

 In the previous firmware version (0.7.0-rc12 82e4095) there was no such problem 
[image]

FreeBear-nc, 2024-06-12:
Backlight GPIO is set at compile time to GPIO38. If you need to change it, you'll need to edit TFT_BCKL in user_setups/esp32s3/esp32-s3-4848S040.ini

tdvtdv, 2024-06-12:
It is clear that you can compile anything, but I wrote about an error in the already created firmware for a particular model, which in fact does not work.

FreeBear-nc, 2024-06-12:
I have the ESP32-S3-4848S040 display myself. The backlight GPIO pin is hardwired in to the board, so there is nothing to set as such. I had submitted a change recently that hid most of the hard wired GPIO (mostly related to TFT control) from the user. See commits c70dfe6 and 7a3b8d0
Perhaps if the backlight GPIO is hard coded, the option to change it should be blanked out so as to avoid confusion.

tdvtdv, 2024-06-13:
If you leave the field empty, the brightness control does not work, i.e. everything less than 255 is perceived as 0.

To be more precise, there is probably no direct connection, but firmware esp32-s3-4848s040_full_16MB_v0.7.0-rc12_cd34d89 has exactly this behavior.

FreeBear-nc, 2024-06-13:
Will have to do a build with the latest code (I'm running a two week old build here).
Issuing a backlight("state":"on","brightness":25} dims the backlight as expected. Won't have time to dig any deeper until the weekend.

fvanroie, 2024-06-13:
The backlight pin should be changeable, `TFT_BL` is treated as a *preference* not hard-coded. Sp this shouldn't be checked here, as it prevents selecting the preferred pin:

```cpp
#ifdef TFT_BCKL
       || (pin == TFT_BCKL)
#endif
```

FreeBear-nc, 2024-06-15:
@tdvtdv Does the above commit solve your issue ?

tdvtdv, 2024-06-16:
I can't check it yet, because I took a ready-made build from nightly.openhasp.com, and there is an old version there.

FreeBear-nc, 2024-06-28:
Just checked out and compiled - Backlight defaults to GPIO 38, and can be selected via the web interface.
Version is displayed as: 	0.7.0-rc12 bbd96b2
Latest version available on nightly.openhasp.com is 0.7.0-rc12 cd34d89 which is over a month old - Time to trigger a new nightly build ?

fvanroie, 2024-06-28:
It's not automated to the point that it's just a "trigger" but still required manual work.
Do you have experience with GitHub automation and front-end building?
It would be *nice* to automate this indeed, but no time for that myself atm.

closed 2024-06-28 by fvanroie