HASwitchPlate/openHASP #753: Adafruit 3.5" 480*320 TFT FeatherWing V2 (TSC2007) not supported?

TheRealSimon42, 2024-06-16:
### Discussed in https://github.com/HASwitchPlate/openHASP/discussions/748

<div type='discussions-op-text'>

<sup>Originally posted by **TheRealSimon42** June  2, 2024</sup>
Hello everyone,
I am just trying to get started with this wonderful Project here, but run into the issue that my Touchscreen is not working. The screen itself is lighting up and showing everything correct, but the touch is not working (I am just able to navigate via the website, when enabling the cursor it stays where it is).

Hardware: 

- [Adafruit HUZZAH32 Assembled Plain Headers](https://exp-tech.de/en/products/adafruit-huzzah32-assembled-plain-headers?variant=47551050678599)
- [Adafruit TFT FeatherWing V2 - 3.5" 480x320 Touchscreen for Feathers - TSC2007 I2C 0x48 Touch Controller](https://exp-tech.de/en/products/adafruit-tft-featherwing-35-480x320-touchscreen-for-feathers?variant=47546750796103)

I suspect the reason for that (just my best guess) is the TSC2007 Touch controller because on the installer page (6.4 or so) it says STMPE610 under Touch Controller. I already found out that there is a nightly installer, but also the TSC2007 or v2 doesn't show up here. When using a FW-File from the Actions-Tab on Github, i.e. huzzah32-featherwing-35-v2_ota_v0.7.0-rc12_944f92e.bin, the screen won't show anything anymore.

But what confuses me, is that it shows up as supported inside the Documentation: https://www.openhasp.com/0.7.0/hardware/adafruit/featherwing/#supported-devices:~:text=(HX8357D%20with%20TSC2007%20resistive%20touch)

Is this maybe an Issue with the Web-Installer? </div>

fvanroie, 2024-06-24:
Hi, I don't own this hardware combination myself. This configuration is user-contributed, so if these binaries don't work it will require some tweaking and compilation of the firmware locally...

TheRealSimon42, 2024-06-26:
Okay, I understand. Maybe some Hint on the HW-Page would be great for new users 😉

By the way, if you want to get your hands on the HW and implement it into the installer, I would be happy to send it over to you as a loan 👌

LarsMichelsen, 2024-10-11:
> But what confuses me, is that it shows up as supported inside the Documentation

I was also mislead by the documentation and thought it was a well supported hardware combination. The website could be clearer on https://www.openhasp.com/0.7.0/hardware/#supported-devices to prevent that. Later on I found https://www.openhasp.com/0.7.0/firmware/#recommended-boards which is giving a clearer recommendation. A link to that might help.

Anyways, I have good news. I spent the past 3 evenings fiddling around with this and finally got it working, first the display and just now the touch functionality. I'll send a PR.

fvanroie, 2024-10-11:
> > But what confuses me, is that it shows up as supported inside the Documentation
> 
> I was also mislead by the documentation and thought it was a well supported hardware

I'm sorry you find the documentation misleading. The pages can be found in [this repository](https://github.com/HASwitchPlate/openHASP-docs) and you are free to add/correct/remove any information that is deemed incorrect. There is a [history ](https://github.com/HASwitchPlate/openHASP-docs/commits/0.7.0/docs/hardware/adafruit/featherwing.md) and git blame if you want to check the contributors... The project is open-source and all the information and code is best-effort and provided _"as-is"_.

I only have the original FeatherWings, and I now count at least 5 different display+touch models and 5 or more different ESP32 boards that can be paired with it. So, there are at least 25 possible configurations that could (should?) be created... You quickly see, this is not that easy, and that's just for AdaFruit FeatherWings!  Maybe we need to add a supported HW combination matrix to the docs?

This all leads to these questions: What's a *supported board*? Who tests and maintains it? Should we cut down on the number of HW? Is it reasonable to assume every page is 100% correct and up-to-date? Maybe post a disclaimer on every page? Or assign specific contributors to each page responsible to maintain it? I 'm not sure that will be feasible. Resources are spread quite thin already.

Please note this is a community effort and it is impossible for me to buy, configure, test and maintain all possible hardware configurations. We recently did [a poll](https://ko-fi.com/openhasp/goal?g=5) and there weren't any respondents using these boards. As an issue arises, thank you for contributing your solution. The next user will be thankful for it.

LarsMichelsen, 2024-10-12:
Please don't take my previous post the wrong way. I came across the project a few weeks ago, find openHASP a really interesting solution and am impressed by the project. Of course there is always room for improvement, but to be honest, it might be a bit boring if everything runs smoothly in such projects. In this respect, I have already had many interesting hours learning about the system and the components used. Thank you very much for the project!

> Or assign specific contributors to each page responsible to maintain it?

I think something like this would work well:

1. Write which boards are best supported. Whether it's 1 or 5 or something in between does not matter too much imho. Whatever you and or other core contributors are willing to maintain. These are then the goto choices for everyone who just wants something to get running and asks himself what to buy.
2. Keep support for other boards as community extensions. This is for all users who have or want some special hardware. It's up to the community to support them. Mark them to be contributions and EOL their support in case issues are reported repeatedly and no fixes are provided. This can be a multi stage process, e.g. when errors are reported repeatedly for an environment, then this is marked somewhere quickly and if no one cares to fix this for longer (months?), then the environment is dropped. In Home assistant I recognized that they are dropping support for not working and unmaintained extensions. While it may feel a bit strict in the first place, if you think further about it, I think it's great to have some "lifecycle management" in place for such things. It helps the project health and in the end also user experience. Not sure if something like this is viable for a project of this size.

noharminthat, 2025-02-13:
> Anyways, I have good news. I spent the past 3 evenings fiddling around with this and finally got it working, first the display and just now the touch functionality.
> 
> See here: [#812](https://github.com/HASwitchPlate/openHASP/pull/812)

I'm having trouble with this same combination. Can you share what you did to get it working? Thanks!

LarsMichelsen, 2025-02-27:
The changes I needed to make it work were merged with the #812 pull request.

What kind of issue do you experience?

I can not remember exactly how I made it work in the end. It was roughly like this:

1. clone git
2. Add `platformio_override.ini` with the following content
3. Run `pio run -e huzzah32-v2-featherwing-35-v2`

```
; USAGE: Copy this file and rename it to platformio_override.ini
;
; The platformio_override.ini file is not overwritten or monitored by git
; ONLY edit platformio_override.ini to make local changes to the parameters

[platformio]
extra_configs =
    ; Uncomment or edit the lines to show more User Setups in the PIO sidebar
    ; user_setups/darwin/*.ini
    user_setups/esp32/huzzah32-v2-featherwing-35-v2.ini
    ; user_setups/esp32s2/*.ini
    ; user_setups/esp32s3/*.ini
    ; user_setups/linux/*.ini
    ; user_setups/stm32f4xx/*.ini
    ; user_setups/win32/*.ini

[override]
; -- Hasp config options --------------------------------------
build_flags =
; -- Uncomment the next line to use the file include/user_config_override.h settings
    -DUSE_CONFIG_OVERRIDE

;region -- Default Build Environments : Used when Build All ---
extra_default_envs =
    ; Uncomment specific environments or create extra (copy names from square brackets in user_setups/*/*.ini):
    ; az-touch-mod-esp32_ili9341_4MB
    ; az-touch-mod-esp32_ili9341_8MB
    ; d1-mini-esp32_ili9341
    ; d1-r32-unoshield_ili9341_adc
    ; d1-r32-unoshield_ili9486_adc
    ; d1-r32-waveshare_ili9486
    ; esp32s2-featherwing-24-v2
    ; esp32s2-featherwing-35-v2
    ; esp32-2432s028r_4MB
    ; esp32-3248s035c_4MB
    ; esp32-3248s035c_4MB_lovyan
    ; esp32-3248s035r_4MB
    ; esp32-one_ili9486
    ; esp32-one_st7796
    ; esp32-s3-4827s043c_16MB
    ; esp32-s3-4827s043r_16MB
    ; esp32-s3-4848s040_16MB
    ; esp32-s3-8048s043c_16MB
    ; esp32-touchdown
    ; freetouchdeck_4MB
    ; freetouchdeck_8MB
    ; freetouchdeck-s3_8MB
    ; gs-t3e_16MB
    ; huzzah32-featherwing-24
    ; huzzah32-featherwing-24-v2
    ; huzzah32-v2-featherwing-24-v2
    ; huzzah32-featherwing-35
    ; huzzah32-featherwing-35-v2
    huzzah32-v2-featherwing-35-v2
    ; lanbon_l8
    ; lanbon_l8_eth
    ; lilygo-lily-pi_ili9481
    ; lilygo-lily-pi_st7796
    ; lolin-d32-pro_ili9341
    ; m5stack-core2
    ; m5stack-tough
    ; makerfabs-s3-tft35-spi
    ; makerfabs-tft35-cap_16MB
    ; makerfabs-tft35-cap_4MB
    ; makerfabs-tft-s2_ili9488
    ; nodemcu32s-raspi
    ; panlee-zw3d95ce01s-ar-4848_16MB
    ; panlee-zw3d95ce01s-tr-4848_16MB
    ; panlee-zw3d95ce01s-ur-4848_16MB
    ; s2-mini-esp32s2_ili9341
    ; ttgo_esp32_poe-ili9341
    ; lilygo-lily-pi_ili9481
    ; lilygo-lily-pi_st7796
    ; ttgo-t7-v1_5_ili9341_16MB
    ; ttgo-t7-v1_5_ili9341_4MB
    ; ttgo-esp32_tdisplay_v1
    ; wt32-sc01_16MB
    ; wt32-sc01_4MB
    ; wt32-sc01-plus_16MB
    ; wt32-sc01-plus_8MB
    ; wt-86-32-3zw1
    ; yeacreate-nscreen32
    ; wz2432r028
    ; ws_esp32_s3_touch_lcd_4p3
;endregion

;region -- Define your local COM ports for each environment ---
;[env:d1-mini-esp32_ili9341]
;monitor_port = COM6       ; Change to the correct port
;upload_port = ${env:d1-mini-esp32_ili9341.monitor_port}
;
;[env:esp32dev-mrb3511]
;monitor_port = COM3       ; Change to the correct port
;upload_port = ${env:esp32dev-mrb3511.monitor_port}
;
;[env:az-touch-mod-esp32_ili9341_4MB]
;upload_port = COM16 
;monitor_port = COM16
;
;[env:az-touch-mod-esp32_ili9341_8MB]
;upload_port = COM16 
;monitor_port = COM16
;
;[env:d1-r32-unoshield_ili9486_adc]
;monitor_port = COM5 
;upload_port = COM5  
;
;[env:d1-r32-unoshield_ili9341_adc]
;;upload_protocol = espota  ; Use ArduinoOTA after flashing over serial
;;upload_port = 192.168.0.4 ; IP of the ESP
;;upload_flags = --port=3232 ; --auth=haspadmin   ; OTA password
;monitor_port = COM5 
;upload_port = COM5
;
;[env:esp32-one_st7796]
;upload_port = COM6      ; Change to the correct port
;monitor_port = COM6     ; Change to the correct port
;
;[env:lolin-d32-pro_ili9341]
;monitor_port = COM5       ; Change to the correct port
;upload_port = ${env:lolin-d32-pro_ili9341.monitor_port}
;;endregion
;
;[env:ttgo_esp32_poe-ili9341]
;monitor_port = COM9       ; Change to the correct port
;;upload_port = ${env:ttgo_esp32_poe-ili9341.monitor_port}
;;upload_port = 192.168.4.5 ; IP of the ESP
;;upload_protocol = espota  ; Use ArduinoOTA after flashing over serial
;upload_flags = --port=3232 ; --auth=haspadmin   ; OTA password

;region -- Custom Environment configuration example -----------------
[env:my_custom_build]
extends = arduino_esp32_v2, flash_4mb
board = nodemcu-32s

upload_port = COM12     ; To change the port, use platform_override.ini
monitor_port = COM12    ; To change the port, use platform_override.ini

debug_tool = esp-prog
debug_init_break = tbreak setup
build_flags =
    ${env.build_flags}
    ${esp32.build_flags}
;region -- TFT_eSPI build options -----------------------------------
    ${lcd.lolin24}
    ${esp32.vspi}        ; Use VSPI hardware SPI bus
    -D TFT_DC=5
    -D TFT_CS=26 
    -D TFT_RST=-1        ; RST
    -D TFT_BCKL=22       ; None, configurable via web UI (e.g. 21)
    -D TOUCH_CS=17       ; (can also be 22 or 16)
; -- Options ----------------------------------------
    -D HASP_USE_TELNET=1
;endregion

;endregion
```