HASwitchPlate/openHASP #754: Infinite rebooting of esp32-s3-4848s040 for 0.7.0

StreakingJerry, 2024-06-19:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

Same as [https://github.com/HASwitchPlate/openHASP/discussions/738](url). But for board `esp32-s3-4848s040`. 

The logs shows that code can crash at the random point. The last line is always `ESP-ROM:esp32s3-20210327`. However the second one from the bottom could be:

`[1000D[0KPrompt > [1000D[9C[       1.367][131060/141444  7][58564/58568  1] HASP: Loading /pages.jsonl`
`[1000D[0KPrompt > [1000D[9C[       0.624][204788/216896  5][    0/    0  0] TFT : Started`
`[1000D[0KPrompt > [1000D[9C[       1.330][131060/141444  7][61576/61592  1] LVGL: Started`

or any line around those moment.

The hardware I used is a esp32-s3-4848s040 board with only ESP32-S3 WROOM-1-N16R8 and 480*480 LCD. No other components installed on the PCB. I use a TTL module with CH343 to upload code.

### To Reproduce

I tried both the firmware `https://nightly.openhasp.com/firmware/esp32-s3-4848s040_full_16MB_v0.7.0-rc12_cd34d89.bin` and firmware compiled locally with Commit Hash: bbd96b2. Both of them has the same issue.

I tried to touch the screen during the booting just as [https://github.com/HASwitchPlate/openHASP/discussions/738](url) suggest. But it doesn't not help. 

### Expected behavior

Start normally.

### Screenshots or video

None.

fvanroie, 2024-06-20:
I do not recognize that PCB as the Guition ESP32-S3-4848S040. So your device might be different or have other hardware etc...
Please share the link where you got this board. Thank you.

FreeBear-nc, 2024-06-21:
Agreed, that board is not a Guition device.
There are a few other display boards that look very similar, but come without relays or mains power. Even these have a built in USB port and can be powered that way.

I'm wondering if it is a conflict with the touch screen - Comment out all the touch settings in the user_setups/esp32s3/esp32-s3-4848S040.ini and see if that helps.

StreakingJerry, 2024-07-01:
I finally figured out the problem. It is because the power source is a 3.3V LDO. It works well with a single LCD or WiFi module. However, when they work together, the large current requirement drops the voltage lower than 2.8V, leading to a reboot. @fvanroie, you were right that this is a hardware problem. And I think this issue could be closed.

fvanroie, 2024-07-01:
Thanks for your feedback and posting the root case of the issue.
Just for future reference, do you have a link to the details for this board?

StreakingJerry, 2024-07-04:
[https://oshwhub.com/myzhazha/esp32s3_86-kai-fa-ban](url)

I have checked the pinout and schematic. For the LCD part it is same as Guition ESP32-S3-4848S040.

closed 2024-08-22 by fvanroie