HASwitchPlate/openHASP #764: sunton8048s070c_16MB: Anti

The-May, 2024-07-15:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

The-May, 2024-07-15:
For some Reason, my whole issue was omitted not saved:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

fvanroie, 2024-08-24:
This should now be fixed in the latest master: https://github.com/HASwitchPlate/openHASP/actions/runs/10540465094

The-May, 2024-09-11:
@fvanroie 

Cant confirm it. Issue seems to still persist. (Dont get irritiated by it - rotation is 90° here and was 270 before)

[image]

It looks like the nightly build for Sunton 8048S070C seems to not use the fix yet.

[image]

fvanroie, 2024-09-11:
That's an old nightly build date. You need to download the latest sunton.zip from the [GitHub Actions](https://github.com/HASwitchPlate/openHASP/actions/runs/10654822394).

The-May, 2024-09-11:
I downloaded the latest build from the Github Actions and it seems to work now.
[image]

[image]

The-May, 2024-09-11:
Works now, thanks.

closed 2024-09-11 by The-May

fvanroie, 2024-09-11:
Great! Thanks for testing and confirming