HASwitchPlate/openHASP #767: Sunton 2432S028R or similar to Touch not work at all

Szymon-ek, 2024-07-19:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

Blow developmen build esp32-2432s028r_full_4MB_v0.7.0-rc12_cd34d89.bin in.
Similar efect with Buided  from Source code via Visual Studio Code

### Describe the bug
New openHASP Nightly Development Build  Sunton 2432S028R, LCD work OK , www, all working fine, but Touch not work at all. With original Firmware LVGL demo touch work OK

### To Reproduce
Reprogram Board with: esp32-2432s028r_full_4MB_v0.7.0-rc12_cd34d89.bin

### Expected behavior
Workin touch

### Screenshots or video
[image]
[image]

fvanroie, 2024-07-19:
2432S028**R** is for resistive _touchreens_ only.

Szymon-ek, 2024-07-19:
Yes and this is resistive touchscreen, chip is xpt2046, is connected courtly to pins: 
TOUCH_CS=33
TOUCH_SCLK=25
TOUCH_MOSI=32
TOUCH_MISO=39
TOUCH_IRQ=36
Because i do not how to debug firmware yet, i do not know if this module was not compiled , or selected one is incorrect

Szymon-ek, 2024-08-03:
any idea whay it not work ?

Saharel001, 2024-08-07:
Hi there, in my case with the same display ID (but with usb c) and the same firmware it works. But the touch is mirrored. Maybe more as one versions of this display exist.

tremichl, 2024-08-09:
Have also an ESP32-2432S028 here with an additional USBC port. Apparently there are some more differences to the "original" ESP32-2432S028R
[image]
[image]
Would it be possible to create an image for upgraded HW version? Thanks!

Darthmonkey, 2024-08-19:
I'm having exactly the same issues as tremichl. Is it likely anyone will be looking at this?

RyanEwen, 2024-09-13:
Might be silly but did you initiate the Display Calibration via web UI?

I just set up 2 of these and touch works fine after doing the Calibration. Did not respond to touch before that.

pocketware, 2024-10-24:
> Might be silly but did you initiate the Display Calibration via web UI?
> 
> 
> 
> I just set up 2 of these and touch works fine after doing the Calibration. Did not respond to touch before that.

This fixes the no touch issue!

jf1452, 2024-11-09:
Same here, logged in by phone, have plate displayed on the screen; no web interface to calibrate.

evranch, 2025-01-03:
Same issue here. Device is a 2432S028R "version 3" with USB-C. Display works as esp32-2432s028r-ili9342_4MB but touch does not.

Touchscreen is confirmed working with https://github.com/rzeldent/esp32-smartdisplay-demo 

Same pinout as described above, working with smartdisplay using those pins. Calibration initiates but does not respond to touch, not even sure where to start debugging this.

Here are the definitions that make it work for smartdisplay:
`"'-D BOARD_HAS_TOUCH'",
      "'-D TOUCH_XPT2046_SPI'",
      "'-D XPT2046_SPI_HOST=SPI3_HOST'",
      "'-D XPT2046_SPI_DMA_CHANNEL=SPI_DMA_CH_AUTO'",      
      "'-D XPT2046_SPI_BUS_MOSI_IO_NUM=32'",
      "'-D XPT2046_SPI_BUS_MISO_IO_NUM=39'",
      "'-D XPT2046_SPI_BUS_SCLK_IO_NUM=25'",
      "'-D XPT2046_SPI_BUS_QUADWP_IO_NUM=GPIO_NUM_NC'",
      "'-D XPT2046_SPI_BUS_QUADHD_IO_NUM=GPIO_NUM_NC'",
      "'-D XPT2046_SPI_CONFIG_CS_GPIO_NUM=33'",
      "'-D XPT2046_SPI_CONFIG_DC_GPIO_NUM=GPIO_NUM_NC'",
      "'-D XPT2046_SPI_CONFIG_SPI_MODE=SPI_MODE0'",
      "'-D XPT2046_SPI_CONFIG_PCLK_HZ=2000000'",
      "'-D XPT2046_SPI_CONFIG_TRANS_QUEUE_DEPTH=3'",
      "'-D XPT2046_SPI_CONFIG_LCD_CMD_BITS=8'",
      "'-D XPT2046_SPI_CONFIG_LCD_PARAM_BITS=8'",
      "'-D XPT2046_SPI_CONFIG_FLAGS_DC_AS_CMD_PHASE=false'",
      "'-D XPT2046_SPI_CONFIG_FLAGS_DC_LOW_ON_DATA=false'",
      "'-D XPT2046_SPI_CONFIG_FLAGS_OCTAL_MODE=false'",
      "'-D XPT2046_SPI_CONFIG_FLAGS_LSB_FIRST=false'",
      "'-D XPT2046_TOUCH_CONFIG_X_MAX=LCD_WIDTH'",
      "'-D XPT2046_TOUCH_CONFIG_Y_MAX=LCD_HEIGHT'",
      "'-D XPT2046_TOUCH_CONFIG_RST_GPIO_NUM=GPIO_NUM_NC'",
      "'-D XPT2046_TOUCH_CONFIG_INT_GPIO_NUM=36'",
      "'-D XPT2046_TOUCH_CONFIG_LEVELS_RESET=0'",
      "'-D XPT2046_TOUCH_CONFIG_LEVELS_INTERRUPT=0'",
      "'-D TOUCH_SWAP_XY=false'",
      "'-D TOUCH_SWAP_X=true'",
      "'-D TOUCH_SWAP_Y=false'"`

fvanroie, 2025-01-03:
New hardware versions need a new configuration and documentation.
Please create a pull request for both.

Poussistar, 2025-06-01:
> C’est peut-être idiot, mais avez-vous lancé l’étalonnage de l’affichage via l’interface utilisateur Web ?
> 
> Je viens d’en configurer 2 et le toucher fonctionne bien après avoir effectué l’étalonnage. N’a pas répondu au toucher avant cela.

Hello, I have the same problem. I installed: esp32-2432s032c_full_4MB_v0.7.0-rc15_74063e2.bin. My touchscreen doesn't work anymore and I don't have the calibration option on my interface in the display settings.

pcbtborges, 2025-09-04:
Openhasp. 
Hy, I have the same board with USB-C. landscape works well but in portrait touch point does not correspond to screen buttoms.
If I keep touching the screen I eventually find where it is mapped to a certain visual button location.
Aparently rotating the visual screen is not rotating the touch-screen.
Any help on this one?
Thanks

woody4165, 2025-10-17:
Same here, here is my picture.

[image]

a quarter of the screen show always antiburn, while the rest is mirrored.

If I choose 270 degrees mirrored this is what I see

[image]

Anything I can do to have it works?

Thanks