HASwitchPlate/openHASP #781: Add retain flag

yorulez, 2024-08-13:
Good day.
Please add the ability to set a retain flag on the mqtt message.

fvanroie, 2024-08-15:
Which messages and why?

Elix-g, 2024-08-16:
Hi together, I've added some custom code to my project and implemented an automatic backlight dimming based on the environment light level. The value pairs I've defined for that, can be set from the plate. A function to publish retained messages would allow me to store those values easily on the MQTT broker. Something like dispatch_state_subtopic _retained() e.g. Not sure if that is what @yorulez meant, but definitely would be useful for me!

fvanroie, 2024-08-22:
I don't think a separate `dispatch_state_subtopic_retained` function is necessary. I've added a retain flag to `mqtt_send_state` instead. Retain is a mqtt flag, so I don't think it makes sense to add it to ``dispatch_state_subtopic` as well.

fvanroie, 2024-08-24:
@yorulez Could you please explain which messages you want to retain and why?

@Elix-g Does the new `mqtt_send_state` retain flag work for your use-case?

Elix-g, 2024-08-24:
@fvanroie Thanks, very appreciate your swift reaction! The new `mqtt_send_state` helps a lot for my use-case.

fvanroie, 2024-09-25:
Closed due to lack of rely from @yorulez

closed 2024-09-25 by fvanroie

adorobis, 2024-10-19:
> Could you please explain which messages you want to retain and why?

I have just configured a device with powerrelay, it nicely shows up in HA but after ha restart or config reload the switch is in unknown state until its state changes. This would be fixed if there is the retain flag for output state topic. Or is there other way this could be achieved?

TNTLarsn, 2025-03-14:
@yorulez Good day, could you please consider reopening this issue?  @adorobis  has already explained which message it concerns. I am also experiencing the problem that after every restart of Home Assistant, the power relays switch to the status "unknown."