HASwitchPlate/openHASP #783: ACE 1.35.4 on CDNJS missing themes

corbolais, 2024-08-19:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

Compiling latest openHASP (as of yesterday) includes ACE editor in version 1.35.4 which is loaded from CDNJS as well as corresponding themes.
ace.min.js v1.35.4 is loaded, whereas themes for v1.35.4 are 404s.

### To Reproduce

Working:
https://cdnjs.cloudflare.com/ajax/libs/ace/1.34.0/theme-monokai.js

Failing with 404:
https://cdnjs.cloudflare.com/ajax/libs/ace/1.35.4/theme-monokai.js

### Expected behavior

Accessible theme file:
https://cdnjs.cloudflare.com/ajax/libs/ace/1.35.4/theme-monokai.js

corbolais, 2024-08-19:
Actually, v1.35.3 themes are accessible:
https://cdnjs.cloudflare.com/ajax/libs/ace/1.35.3/theme-monokai.js

fvanroie, 2024-08-19:
It seems the CDN wasn't updated properly with all themes.

closed 2024-08-19 by fvanroie