HASwitchPlate/openHASP #788: Spaces replaced by TAB in file editor

matsekberg, 2024-08-27:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug
When I edit a file on an OpenHasp device using the file editor and enter multiple spaces, they are replaced by TAB character(s)

### To Reproduce
Just edit a file, enter  multiple spaces and the use the arrow keys to back go left. The cursor will jump rather than take one space-step at a time.

### Expected behavior
No TABs stored

### Screenshots or video

https://github.com/user-attachments/assets/a2da0864-df41-4afd-b914-14785aadb6b7

matsekberg, 2024-08-27:
Just noticed that I can type Oprion-Space on a Mac keyboard to enter hard spaces and they are marked with red background and dots.

[image]

fvanroie, 2024-08-27:
In the editor field, press F1.
Then select Show Settings Menu `ctrl-,`
[image]

Uncheck "Soft Tabs" and see if this fixes the tab problem.

matsekberg, 2024-08-27:
Editor field?

fvanroie, 2024-08-27:
The field where you type your jsonl code.

matsekberg, 2024-08-29:
Ok found it and problem solved. Is it possible to make soft tabs disabled by default?

closed 2024-09-25 by fvanroie