HASwitchPlate/openHASP #794: Backlight Pin results in blank screen after power cycle [esp32-terminal-rgb]

trip5, 2024-08-31:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug
Selecting "Backlight Pin" : "None" results in blank screen after power cycle.

### To Reproduce
- Factory reset
- Connect to Wi-fi
- Select Backlight pin none (default is blank, but none is only option)
- Restart to make option stick (display still functions)
- Pull USB plug, put back in
- Blank display but device is still connected and can be accessed via http

### Screenshots or video

Actually, there's simply no difference in logs from before selecting this and after but here:
````
Showing logs:
[21:53:10]ESP-ROM:esp32s3-20210327
[21:53:10]Build:Mar 27 2021
[21:53:10]rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
[21:53:10]SPIWP:0xee
[21:53:10]mode:DIO, clock div:1
[21:53:10]load:0x3fce3808,len:0x128
[21:53:10]load:0x403c9700,len:0xb90
[21:53:10]load:0x403cc700,len:0x28cc
[21:53:10]entry 0x403c98b8
[21:53:10]
[21:53:10]
[21:53:10]
[21:53:10]        open____ _____ _____ _____
[21:53:10]          |  |  |  _  |   __|  _  |
[21:53:10]          |     |     |__   |   __|
[21:53:10]          |__|__|__|__|_____|__|
[21:53:10]        Home Automation Switch Plate
[21:53:10]        Open Hardware edition v0.7.0-rc13
[21:53:10]
[21:53:10][       0.095][212980/245616 13][    0/    0  0] DBUG: Started @ 115200 bps
[21:53:10]#[       0.104][212980/245616 13][    0/    0  0] DBUG: Environment: esp32-terminal-rgb_16MB
[21:53:10]#[       0.114][212980/245616 13][    0/    0  0] UART: Started
[21:53:10]#[       0.122][212980/245616 13][    0/    0  0] UART: Client login from serial
[21:53:10]#[       0.147][212980/245516 13][    0/    0  0] CONF: SPI flash FS mounted
[21:53:10]Prompt > [       0.181][212980/245516 13][    0/    0  0] CONF: Loading /config.json
[21:53:10]Prompt > [       0.192][212980/244748 12][    0/    0  0] CONF: Loaded /config.json
[21:53:10]Prompt > [       0.201][212980/245580 13][    0/    0  0] DBUG: Loading debug settings
[21:53:10]Prompt > [       0.211][212980/245580 13][    0/    0  0] GPIO: Loading GUI settings
[21:53:10]Prompt > [       0.221][212980/245580 13][    0/    0  0] GUI : First Touch Calibration enabled
[21:53:10]Prompt > [       0.231][212980/245580 13][    0/    0  0] HASP: Loading HASP settings
[21:53:10]Prompt > [       0.241][212980/245580 13][    0/    0  0] WIFI: Loading WiFi settings
[21:53:11]Prompt > [       0.270][212980/243380 12][    0/    0  0] MQTT: Loading MQTT settings
[21:53:11]Prompt > [       0.287][212980/243380 12][    0/    0  0] TELN: Loading Telnet settings
[21:53:11]Prompt > [       0.297][212980/243380 12][    0/    0  0] MDNS: Loading MDNS settings
[21:53:11]Prompt > [       0.307][212980/243380 12][    0/    0  0] HTTP: Loading HTTP settings
[21:53:11]Prompt > [       0.317][212980/243380 12][    0/    0  0] GPIO: Loading GPIO settings
[21:53:11]Prompt > [       0.327][212980/243380 12][    0/    0  0] CONF: Settings loaded
[21:53:11]Prompt > [       0.388][204788/214576  4][    0/    0  0] WIFI: Connecting to RestrictedArea
[21:53:11]Prompt > [       0.399][204788/214536  4][    0/    0  0] MSGR: Starting...
[21:53:11]Prompt > [       0.408][204788/214536  4][    0/    0  0] MSGR: Started
[21:53:11]Prompt > [       0.416][204788/214536  4][    0/    0  0] TFT : Starting...
[21:53:11]Prompt > [       0.425][204788/214536  4][    0/    0  0] TFT : Starting...
[21:53:11]Prompt > [       0.434][204788/213724  4][    0/    0  0] TFT : Touch Started
[21:53:11]Prompt > [       0.737][196596/212600  7][    0/    0  0] TFT : Started
[21:53:11]Prompt > [       0.751][196596/212600  7][    0/    0  0] TFT : Started
[21:53:11]Prompt > [       0.827][196596/212600  7][    0/    0  0] LVGL: Starting...
[21:53:11]Prompt > [       0.840][147444/163232  9][62608/62832  1] LVGL: LV_FS_PC ready
[21:53:11]Prompt > [       0.850][126964/138060  8][61908/61956  1] GUI : Initialize Cursor
[21:53:11]Prompt > [       0.861][126964/137960  7][61576/61592  1] LVGL: Started
[21:53:11]Prompt > [       0.878][126964/137960  7][59116/59756  2] HASP: Custom theme loaded
[21:53:11]Prompt > [       0.907][126964/137960  7][58564/58568  1] HASP: Loading /pages.jsonl
[21:53:11]Prompt > [       0.941][126964/137960  7][58448/58452  1] HASP: Loaded /pages.jsonl
[21:53:11]Prompt > [       0.950][126964/137960  7][58448/58452  1] HASP: Changing page to 1
[21:53:11]Prompt > [       0.960][126964/137960  7][58288/58292  1] GPIO: Starting...
[21:53:11]Prompt > [       0.969][126964/137960  7][58288/58292  1] GPIO: Reboot counter 0
[21:53:11]Prompt > [       0.978][126964/137960  7][58288/58292  1] GPIO: Invalid pin 0
[21:53:11]Prompt > [       0.987][126964/137960  7][58288/58292  1] GPIO: Invalid pin 0
[21:53:11]Prompt > [       0.996][126964/137960  7][58288/58292  1] GPIO: Invalid pin 0
[21:53:11]Prompt > [       1.004][126964/137960  7][58288/58292  1] GPIO: Invalid pin 0
[21:53:11]Prompt > [       1.013][126964/137960  7][58288/58292  1] GPIO: Invalid pin 0
[21:53:11]Prompt > [       1.022][126964/137960  7][58288/58292  1] GPIO: Invalid pin 0
[21:53:11]Prompt > [       1.031][126964/137960  7][58288/58292  1] GPIO: Invalid pin 0
[21:53:11]Prompt > [       1.040][126964/137960  7][58288/58292  1] GPIO: Invalid pin 0
[21:53:11]Prompt > [       1.049][126964/137960  7][58288/58292  1] GPIO: Started
[21:53:11]Prompt > [       1.058][122868/133516  7][58288/58292  1] HTTP: Started
[21:53:11]Prompt > [       1.067][122868/132948  7][58288/58292  1] TELN: Telnet console started
[21:53:11]Prompt > [       1.077][122868/132948  7][58288/58292  1] FTP : Starting...
[21:53:11]Prompt > [       1.086][122868/132948  7][58288/58292  1] FTP : Disabled
[21:53:11]Prompt > [       1.116][122868/132948  7][58288/58292  1] MSGR: File not found: L:/boot.cmd
[21:53:11]Prompt > [       1.130][122868/131876  6][58288/58292  1] MQTT: hasp/plate/LWT
[21:53:11]Prompt > [       1.140][122868/131828  6][58288/58292  1] MQTT: hasp/plate/LWT
[21:53:11]Prompt > [       1.149][122868/132048  6][58288/58292  1] MQTT: Broker not configured
[21:53:14]Prompt > [       3.377][122868/131592  6][58124/58448  1] WIFI: Connected to RestrictedArea, requesting IP...
[21:53:14]Prompt > [       3.389][122868/131728  6][58124/58448  1] WIFI: Received IP address 192.168.1.111
[21:53:14]Prompt > [       4.185][122868/131900  6][58124/58448  1] MSGR: Loading L:/online.cmd
[21:53:14]Prompt > [       4.215][114676/125820  8][58124/58448  1] MSGR: jsonl={"page":0,"id":239,"obj":"msgbox","text":"%ip%","auto_close":20000}
[21:53:15]Prompt > [       4.262][118772/130828  9][57836/57876  1] MSGR: Loaded L:/online.cmd
[21:53:15]Prompt > [       4.272][122868/131432  6][57836/57876  1] HTTP: Started @ http://192.168.1.111
[21:53:15]Prompt > [       4.282][122868/131480  6][57836/57876  1] MDNS: Starting...
[21:53:15]Prompt > [       4.313][114676/124076  7][57836/57876  1] MDNS: Started
````

I assume that something is happening underneath - that the default option (blank, perhaps undefined) is fine but by selecting "none" the config is changed, the variable is defined but invalid?

trip5, 2024-08-31:
Oh and the only way to fix this after selected "none" is to perform a factory reset.

marsman7, 2024-12-06:
You can only select "none" in web interface display settings? You can't see GPIOs?

Bert-62, 2025-02-02:
I see the same issue as trip05.  After changing in "Display Settings" the "Backlight Pin" from empty/blank to "none" and reboot, the Display stays black. The Display is accessible via the IP address and I can see that the device is running normal (e.g. via Screenshot). 
For Backlight Pin there is a list of available entries (none, GPIO 0, GPIO 4, GPIO 10, ... GPIO 44) but it can't be deleted and changed back to blank. The only way to revert the Backlight Pin back to empty/blank entry is a factory reset.
 
The device is a Sunton ESP32-8048S043C, the installed firmware is  "sunton-8048s043c_full_16MB_v0.7.0-rc12_cd34d89"
[image]