HASwitchPlate/openHASP #811: Sunton 7" display use with debug port

szkezolt, 2024-10-07:
Hi!

Firstly i would like to thank this project for every author and programer. I very like it!

I'm not  programer, but i do some project with this displays with openhasp.
In my previous project i use openhasp firmware **0.7.0-rc2** on Sunton 7" display and i communicate with the display on serial debug port with another esp32board with inputs and outputs. 
It is working perfectly!

But, Yesterday i installed a new openhasp 0.7.0-rc14 Sunton 7" display. Now i cant communicate with the display. With usb connection from serial consol it working perfectly, but the debug port not receive my commands. i see the terminal messages, but cant do nothing if i text a command.

I can't decide if there was a design change on the panel or if something changed in the firmware.
Where can i find previous sunton binaries? Thanks for everyone.

szkezolt, 2024-10-07:
okay, i found a problem on the pcb. The rx pin is not working correctly on the four pin connector. I soldered a wire on the esp chip pin and it works now...
I have two of these 7" display and it is problem both of them.

fvanroie, 2024-10-07:
Thanks for reporting back with the fix. Maybe there was a bad batch of hardware, but certainly weird/inconvenient.

closed 2024-10-08 by fvanroie