HASwitchPlate/openHASP #814: jsonl parsing - duplicate objects

smcgann99, 2024-10-12:
### Perform all steps below and tick them with [x]
- [X ] Check the related part of the [Documentation](https://www.openhasp.com/)
- [X] Update openHASP to the latest version
- [X ] Reproduce the issue and describe all steps

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

marsman7, 2024-12-06:
I think this is not a mistake for two reasons:

- The type of the object cannot be changed afterwards. That makes no sense. Different objects in memory use the memory reserved for them differently. This is only possible if the object is deleted from the memory and another object is created.

- The ID is always valid for the parent object. For ID 1 and 2, it is the page. For label ID 3 it is label ID 1, but for button ID 3 it is label ID 2, which is why a new object is naturally created.

Simply test Parentid = 1 for both objects. You can see that no new object is created and all properties that both the label and the button have in common are adopted.

Translated with DeepL.com (free version)

smcgann99, 2025-01-25:
I understand what you are saying, but regardless of the parent id there are now 2 objects on page 2 with ID 3. 
So any interactions with them both report as ID 3, and any update to properties are applied to both. which is a bit messy and confusing.

Obviously this wasn't a design decision just a typo, but it would be better is this was handled when parsed, with a warning and no update to the page  ?

marsman7, 2025-01-25:
Yes, you could add a warning to the programme for this case, but I think it takes a lot of time. I'd rather have the designer pay attention and restart the device faster.