HASwitchPlate/openHASP #836: Updated settings for the Lanbon L9 commit issues

FreeBear-nc, 2024-12-14:
Updated local repository to [78a3b38](https://github.com/HASwitchPlate/openHASP/commit/78a3b38829710bde490c52a4c95ce6dc90852145) and compiled for a Guition ESP32-S3 480*480.
Getting double split image flickering (jittering) around on the LCD to the point of being unusable. Going back through recent commits, tracked the issue down to [c89d10d](https://github.com/HASwitchPlate/openHASP/commit/c89d10dd902907b4d2ee2965dd41728696d22792), specifically, this snippet:
```
@@ -142,7 +142,7 @@ void ArduinoGfx::init(int w, int h)

    /* TFT init */
    LOG_DEBUG(TAG_TFT, F("%s - %d"), __FILE__, __LINE__);
    tft->begin(GFX_NOT_DEFINED);
    tft->begin(SPI_FREQUENCY);
    LOG_DEBUG(TAG_TFT, F("%s - %d"), __FILE__, __LINE__);
    // tft.setSwapBytes(true); /* set endianness */
    LOG_INFO(TAG_TFT, F(D_SERVICE_STARTED));
```
Adding ```-D SPI_FREQUENCY=GFX_NOT_DEFINED``` to the build flags in esp32-s3-4848S040.ini fixes the display issue. But I suspect other displays may suffer from the same problem. Did try using 40000000, 80000000, and 26000000 to no effect.

fvanroie, 2024-12-14:
Updated in  e815e31 and  154a8ff
That should fix it, please test it. `SPI_FREQUENCY` should not be set for ST7701S or RGB displays.

FreeBear-nc, 2024-12-14:
Yup, that appears to have fixed it.
Thanks.

closed 2024-12-14 by FreeBear-nc