HASwitchPlate/openHASP #839: Turning off backlight on short idle disables long idle

vorbeiei, 2024-12-23:
### Perform all steps below and tick them with [x]
- [x ] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x ] Update openHASP to the latest version
- [ x] Reproduce the issue and describe all steps

### Describe the bug
Due to the line of code `if(hasp_first_touch_state) return; // don't update sleep when first touch is still active` in hasp.cpp long idle can never be reached if the backlight is turned completely off with short idle already.

### To Reproduce
turn off backlight in short_idle.cmd and watch it never reach long-idle

### Expected behavior
I guess the question is if that is expected behaviour, but I would think not? Long idle should still be reached if screen was turned off before.(In my peticular case I turn off backlight with short idle and trigger deep sleep on long idle in custom code)

fvanroie, 2024-12-23:
It looks like a performance enhancing early quick exit of that function. I can't say it's deliberate... that code hasn't been touched in a long time.

What happens when you comment out that code? Does it work as expected then?

vorbeiei, 2024-12-23:
When commenting it out long idle starts to work again, yes.

closed 2025-07-13 by fvanroie