HASwitchPlate/openHASP #846: Minimal firmware or factory reset to edit filesystem

hallard, 2025-01-04:
### Perform all steps below and tick them with [x]
- [X] Check the related part of the [Documentation](https://www.openhasp.com/)
- [X] Update openHASP to the latest version
- [X] Reproduce the issue and describe all steps

openHASP
Version | 0.7.0-rc13 042fe05
Build DateTime | Aug 26 2024 00:50:13 UTC
Environment | sunton-8048s070c_16MB

### Describe the bug
When I was changing my `pages.jsonl` I think I've reached the limits and then got a out of memory on reboot and loose access to filesystem editor and all web interface.

### To Reproduce
Increased `pages.jsonl` then on boot got 

```
Prompt > [       4.110][94196/110980 15][ 9228/ 9228  0] HASP: Changing page to 1
Prompt > [       4.120][94196/110980 15][ 9068/ 9068  0] GPIO: Starting...
Prompt > [       4.129][94196/110980 15][ 9068/ 9068  0] GPIO: Reboot counter 0
Prompt > [       4.138][94196/110980 15][ 9068/ 9068  0] GPIO: Started
Prompt > [       4.176][94196/106468 11][ 9068/ 9068  0] HTTP: Started
Prompt > [       4.178][94196/106104 11][ 9068/ 9068  0] WIFI: auth failed
Prompt > [       4.193][94196/105884 11][ 9068/ 9068  0] TELN: Telnet console started
Prompt > [       4.203][94196/105884 11][ 9068/ 9068  0] FTP : Starting...
Prompt > [       4.212][94196/105884 11][ 9068/ 9068  0] FTP : Disabled
Prompt > [       4.242][94196/105884 11][ 9068/ 9068  0] MSGR: File not found: L:/boot.cmd
Prompt > [       4.256][94196/104812 10][ 9068/ 9068  0] MQTT: hasp/makerfabs_70_dev/LWT
Prompt > [       4.267][94196/104764 10][ 9068/ 9068  0] MQTT: hasp/makerfabs_70_dev/LWT
Prompt > [       4.277][94196/104920 10][ 9068/ 9068  0] MQTT: makerfabs_70_dev_971130
Prompt > [       4.287][86004/93660 12][ 9068/ 9068  0] MQTT: Starting...
Prompt > [       4.288][81908/93924 12][ 9068/ 9068  0] MQTT: Transport error
Prompt > [       4.305][81908/93924 12][ 9228/ 9228  0] MQTT: Disconnected
Prompt > [       6.786][77812/88656 12][ 4004/ 8168 51] WIFI: Connected to CH2I-HOTSPOT, requesting IP...
Prompt > [       7.798][77812/87180 10][ 3400/ 8164 59] WIFI: Received IP address 192.168.1.188
Prompt > [       8.311][77812/88816 12][ 4004/ 8172 52] MSGR: Loading L:/online.cmd
Prompt > [       8.368][73716/82736 10][ 4004/ 8172 52] MSGR: jsonl={"page":0,"id":239,"obj":"msgbox","text":"%ip%","auto_close":2000}
Prompt > [       8.421][73716/82736 10][ 4004/ 7592 48] MSGR: json='backlight {"state":"ON","brightness":100}'
Prompt > [       8.433][73716/82616 10][ 4004/ 7592 48] MSGR: backlight={"state":"ON","brightness":100}
Prompt > [       8.445][73716/82616 10][ 4004/ 7592 48] MQTT: Not connected ??? backlight => {"state":"on","brightness":100}
Prompt > [       8.458][77812/87744 11][ 4004/ 7592 48] MSGR: Loaded L:/online.cmd
Prompt > [       8.468][77812/88444 12][ 4004/ 7592 48] HTTP: Started @ http://192.168.1.188
Prompt > [       8.479][77812/88492 12][ 4004/ 7592 48] MDNS: Starting...
Prompt > [       8.510][69620/81016 14][ 4004/ 7592 48] MDNS: Started
Prompt > [       8.680][69620/81016 14][ 1776/ 4488 61] LVGL: Couldn't allocate memory
Prompt > [       8.690][69620/80848 13][ 1776/ 4488 61] LVGL: Couldn't allocate memory
Prompt > [       8.699][69620/80848 13][ 1776/ 4488 61] LVGL: _lv_mem_buf_get
Prompt > [       8.708][69620/80848 13][ 1776/ 4488 61] LVGL: Out of memory, can't allocate a new buffer (increase your LV_MEM_SIZE/heap size) (0x00000000)
Prompt > [       9.323][69620/80216 13][ 1776/ 4488 61] MQTT: Started
Prompt > [       9.331][69620/80216 13][ 1776/ 4488 61] MQTT: Connected to broker 192.168.1.8 as clientID makerfabs_70_dev_971130
Prompt > [       9.348][65524/76528 14][ 1776/ 4488 61] MQTT PUB: page => 1
Prompt > [       9.358][65524/77592 15][ 1776/ 4488 61] MQTT PUB: idle => off
Prompt > [       9.368][65524/77020 14][ 1776/ 4488 61] MQTT PUB: antiburn => {"state":"off"}
Prompt > [      11.841][65524/77148 15][ 1776/ 4488 61] MQTT RCV: Out of memory
Prompt > [      12.380][65524/76688 14][ 1776/ 4488 61] MQTT RCV: Out of memory
Prompt > [      12.919][65524/76688 14][ 1776/ 4488 61] MQTT RCV: Out of memory
Prompt > [      13.458][65524/76688 14][ 1776/ 4488 61] MQTT RCV: Out of memory
Prompt > [      13.997][65524/76688 14][ 1776/ 4488 61] MQTT RCV: Out of memory
Prompt > [      14.536][65524/76688 14][ 1776/ 4488 61] MQTT RCV: Out of memory
Prompt > [      15.075][65524/76688 14][ 1776/ 4488 61] MQTT RCV: Out of memory
Prompt > [      15.614][65524/76688 14][ 1776/ 4488 61] MQTT RCV: Out of memory
Prompt > [      16.153][65524/76688 14][ 1776/ 4488 61] MQTT RCV: Out of memory
Prompt > [      16.692][65524/76688 14][ 1776/ 4488 61] MQTT RCV: Out of memory
Prompt > [      17.231][65524/76688 14][ 1776/ 4488 61] MQTT RCV: Out of memory
Prompt > [      17.772][65524/76688 14][ 1776/ 4488 61] MQTT RCV: Out of memory
Prompt > [      18.311][65524/76688 14][ 1776/ 4488 61] MQTT RCV: Out of memory
Prompt > [      18.850][65524/76688 14][ 1776/ 4488 61] MQTT RCV: Out of memory
Prompt > [      19.389][65524/76688 14][ 1776/ 4488 61] MQTT RCV: Out of memory
Prompt > [      19.928][65524/76688 14][ 1776/ 4488 61] MQTT RCV: Out of memory
```

### Expected behavior

if there any solution to have minimal firmware or serial command to be able to get out this memory error and discards changes I made to `pages.jsonl`, I got lot of config and custom graphics, would love not loosing them all since I know what to change in `pages.jsonl` to get things back to work

### Screenshots or video

fvanroie, 2025-01-04:
There is no minimal firmware version, but you can [disable the loading of the pages.json file](https://github.com/HASwitchPlate/openHASP/blob/48884ee98610afba6f71dc81742e247e08d48079/src/hasp/hasp.cpp#L639) and [compile locally](https://www.openhasp.com/0.7.0/firmware/compiling/local/).

hallard, 2025-01-04:
yeah thanks that was my though and wanted to avoid, building :-), anyway, will do that thanks.
BTW as lot of boards have push button may be prevent loading pages.json when button is pressed on boot could be a great workaround, just my 2 cents :-)

hallard, 2025-01-04:
hey just build with gitpod, awesome tool, anyway do you know the command to flash generated file (OTA or Full) without erasing flash? Just tried to find how but did not found. Thanks

fvanroie, 2025-01-04:
- Backup the full flash first, just to be sure!
- Flash the *full* binary to address 0x0 and it *should* preserve the filesystem if it already exists.

hallard, 2025-01-04:
> Backup the full flash first, just to be sure!

Thanks already done of course :-)

 > Flash the full binary to address 0x0 and it should preserve the filesystem if it already exists

That was the point was not sure of this one

fvanroie, 2025-01-04:
[Flash the ESP32](https://www.openhasp.com/0.7.0/firmware/esp32/#flash-esp32_1) as normal without erasing the flash first, of course.

hallard, 2025-01-04:
saved my day thanks

I let command used here as a reminder

```shell

esptool.py  -p /dev/tty.usbserial-1120 write_flash 0x0 /Users/charles/Downloads/sunton-8048s070c_full_16MB_v0.7.0-rc15_48884ee.bin
```
I also needed to add line `sunton-8048s070c_16MB` into `platformio_override.ini` because it was not into commented lines but indicated in comment `(copy names from square brackets in user_setups/*/*.ini)` 

```ini
;region -- Default Build Environments : Used when Build All ---
extra_default_envs =
    ; Uncomment specific environments or create extra (copy names from square brackets in user_setups/*/*.ini):
    sunton-8048s070c_16MB
    ; az-touch-mod-esp32_ili9341_4MB
    ; az-touch-mod-esp32_ili9341_8MB
    ; d1-mini-esp32_ili9341
    ; d1-r32-unoshield_ili9341_adc
```

closed 2025-07-13 by fvanroie