HASwitchPlate/openHASP #868: Sunton ESP32-8048S050 backlight ok but no display

plord12, 2025-02-23:
### Perform all steps below and tick them with [x]
- [X] Check the related part of the [Documentation](https://www.openhasp.com/)
- [X] Update openHASP to the latest version
- [X] Reproduce the issue and describe all steps

### Describe the bug
On ESP32-8048S050, although the backlight works there is nothing drawn on the display.  Sometimes you get slow flashing whole-screen colours.

See https://github.com/HASwitchPlate/openHASP/discussions/861 for chat.

### To Reproduce
Use https://nightly.openhasp.com/ to flash device - on startup, no graphics is seen.

### Expected behavior
On flashing, you should see the hostname & IP address

### Screenshots or video

plord12, 2025-02-23:
Note that I have a PR pending

plord12, 2025-02-23:
Note that high values of TFT_PREFER_SPEED is required to make the screen work but too high and the wifi slows considerably.

plord12, 2025-03-11:
> hello , can you send .bin file for this problem. i have the same problem but i don't known modify firmawre. thanks :)

This change is checked in so you can install from https://nightly.openhasp.com/

plord12, 2025-03-11:
Closing since this checked in

closed 2025-03-11 by plord12

jonancy54, 2025-03-11:
are you sure ? in website is RC12 and flash is already blank screen :(

Rolibaer, 2025-03-31:
Same problem here.
I updated me **ESP32-8048S043C_I** Display to RC12 and then to RC15.
now i have withe screen, WEB-Interface of the ESP is working.

Display spec says: 
- Driver Chip= **ILI9485**

Do you have any idea?

plord12, 2025-03-31:
> Same problem here. I updated me **ESP32-8048S043C_I** Display to RC12 and then to RC15. now i have withe screen, WEB-Interface of the ESP is working.
> 
> Display spec says:
> 
>     * Driver Chip= **ILI9485**
> 
> 
> Do you have any idea?

This issue was for the 8048S050C and the patch submitted only updated 8048S050C.

You might want to test a similar update for the 8048S043C.

Rolibaer, 2025-04-05:
Yes i would like to test it on the 8048S043C,  but unfortunately i had no success

smartroad, 2025-05-31:
I was poking around and found https://github.com/HASwitchPlate/openHASP/actions/runs/15084860385 and downloaded the Sunton files, uploaded the RC15 (i tried both OTA and FULL) but am still getting the slow rotating red, green, blue, black, white screen.