HASwitchPlate/openHASP #871: Add support for Guition JC2432W328

Monacoslo, 2025-02-25:
**Is your feature request related to a problem? Please describe.**
Popular cheap yelow display on AliExpress but existing configuration does not work

**Describe the solution you'd like**
Add support for this model

The whole documentation including Arduino code samples is on http://pan.jczn1688.com/s/6brhsi

fvanroie, 2025-02-27:
There is currently no support for the JC2432W328 device. I do not own this device, so it is up to the community.
I estimate the effort will be small to implement support for this new variant.
Are you able to contribute with development?

Modding, 2025-03-14:
I have 2 of these Devices sitting here for half a year, and i cant use them, because they are not compatible to OpenHASP and to ESP Home ...

I peridiodically search for a use in OpenHASP, and if this Monitor works i will Install OpenHASP on my Home Assistant, but till then i wait.

I have no idea how to programm them.