HASwitchPlate/openHASP #873: Add Support for GUITION JC1060P470

ahb-bs, 2025-02-26:
Would it be possible to get support for the ESP32-P4 + 1024*600 MIPI Display which uses the JD9165 display driver and the GT911 touchscreen - but it's been reported that this is only 800*500.

Device Name
JC1060P470

**URL of Device Specifications document**

Board
http://pan.jczn1688.com/directlink/1/HMI%20display/JC1060P470C_I_W.zip

Display driver
https://github.com/espressif/esp-iot-solution/tree/master/components/display/lcd/esp_lcd_jd9165

Touch
https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/device/usb_extend_screen/components/espressif__esp_lcd_touch_gt911

https://www.aliexpress.com/item/1005008328088576.html

fvanroie, 2025-02-27:
There is currently no support for ESP32P4 devices, and I have no plans to add this any time soon.
I estimate the effort will be substantial to implement support for this new MCU.
Are you able to contribute with development?

ahb-bs, 2025-02-27:
Firstly, the work you've done on this project is giving me a whole new avenue for making the home automation integrated into the house, so thank you! I looked to see if there was a Buy Me a Coffee link or something so I can show you my appreciation!

Secondly, I'm looking into (but don't have the skills to) make the changes as follows:
Revised Architecture
**Component    | Role**
ESP32-P4      |  Handles UI, Rendering, and Touch Input (LVGL)
ESP32-S3      |  Handles Wi-Fi, MQTT, Home Assistant, and Cloud Communication

**What This Means:**
ESP32-P4 runs openHASP and directly, renders everything on the display and manages touch actions
ESP32-S3 acts as a network controller and forwards Home Assistant events to the ESP32-P4.

I'm not sure who to split the code but I'm looking at it. It's a whole new world....

fvanroie, 2025-02-27:
Check out the [donation links](https://www.openhasp.com/0.7.0/#donate).

Unfortunately I don't have to time to investigate these new hardware changes, but I believe the P4 can use the C6 Wifi transparently with the correct networking stack/driver.

ahb-bs, 2025-02-28:
I just looked on the Github site! Thank you!

It can, I'm looking to see what I can do with getting basic comms in place and build from there.

EduardoPosada, 2025-11-13:
Hola,
Tengo algo parecido a lo que tu quieres.
Esp32p4 lcd MPI 1024x600.
Me funciona todo muy bien, lvgl, wifi, mqtt.
Desarrollado en platformio.
Para BT hay que desarrollar en esp-idf.
Si quieres podemos colaborar. Puedo compartir mi proyecto.
Saludos,Eduardo.
eduardo.posada@gmail.com

markusressel, 2026-02-15:
It seems like there has been a lot of development over at ESPHome for these devices:

https://community.home-assistant.io/t/jc8012p4a1-guition-esp32-p4-esp32-c6/939971

Maybe that can help here too?