HASwitchPlate/openHASP #876: value_align doesn't work on bar object

plord12, 2025-03-02:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug
value_align doesn't work on bar object

### To Reproduce

With the following design -

```json
{"id":1,"obj":"bar","x":50,"y":90,"w":160,"h":40,"val":50,"value_str":"center","value_align":"center"}                                                                                                                                                                                                  
{"id":2,"obj":"bar","x":50,"y":190,"w":160,"h":40,"val":50,"value_str":"left","value_align":"left"}                                                                                                                                                                                                     
{"id":3,"obj":"bar","x":50,"y":290,"w":160,"h":40,"val":50,"value_str":"right","value_align":"right"}    
```
The value in the bar is not aligned.

### Expected behavior

value should be aligned as per docs https://www.openhasp.com/0.7.0/design/styling/?h=value_align#value

### Screenshots or video

[image]

fvanroie, 2025-03-02:
Hi, the docs seem to be wrong about `value_align` possible options. Try this:

[image]

```json
{"id":1,"obj":"bar","x":50,"y":90,"w":160,"h":40,"val":50,"value_align":0,"value_str":"center"}                  
{"id":2,"obj":"bar","x":50,"y":190,"w":160,"h":40,"val":50,"value_align":7,"value_str":"left"}
{"id":3,"obj":"bar","x":50,"y":290,"w":160,"h":40,"val":50,"value_align":8,"value_str":"right"}
```

fvanroie, 2025-03-02:
The actual values range from 0 to 20 denoting these possible locations:

[image]

plord12, 2025-03-02:
Great, thanks, so just a doc issue.

ahb-bs, 2025-03-03:
> The actual values range from 0 to 20 denoting these possible locations:
> 
> [image]

Brilliant detail, thank you!

fvanroie, 2025-11-16:
Anyone want to help with updating the docs? Thanks.