HASwitchPlate/openHASP #896: Sunton ESP32-8048S043C wrong config

Rolibaer, 2025-04-26:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

zakkieboy44, 2025-06-04:
I experience the same issue on: [openHASP 0.7.0-rc12 

I can access the plate and see the display via the screenshot option, but the plate remains white.

I dont have the technical capability to recompile using the proposed ini - is there a d/l link anywhere?

Thanks

aristeu, 2025-06-09:
Same issue as Rolibaer, updated user_setups/esp32s3/sunton-esp32-s3-tft.ini with their changes and that fixed the issue for me.

aristeu, 2025-06-10:
Except for touch. I have a capacitive display, touch is not working properly, only top left is accurate.

aristeu, 2025-06-10:
I see e3ae1e6abfec13f98749c9f9ec1adfaa9bde64af already has Rolibaer's changes.
For fixing touch, the touch resolution needs to be changed (see https://github.com/limpens/esp32-8048S043/blob/main/main/lcd.h):
-    -D TOUCH_WIDTH=800
-    -D TOUCH_HEIGHT=480
+    -D TOUCH_WIDTH=477
+    -D TOUCH_HEIGHT=269

aristeu, 2025-06-10:
https://github.com/HASwitchPlate/openHASP/pull/916 created